pascallanger/DIY-Multiprotocol-TX-Module #606: Binding and software loading

scrc511, 2021-08-09:
After loading the new edgetx software 81449497, radiomaster Tx16s (new radio). Power comes on but the software wont load, had to removed the battery cycle the power then it will load. Another issue was after loading the new software had to rebind all the models. Reverted back to Opentx all models are bound and no loading software issue so far.

pascallanger, 2021-08-09:
I'm not sure why you open a case here and not on edgetx. As you can see OpenTX works flawlessly so the issue is omewhere on edgetx therefore you need to open your case on their github.

closed 2021-08-09 by pascallanger

scrc511, 2021-08-09:
My apology, thought that was on their GitHub. Pls disregard, thank you.