pascallanger/DIY-Multiprotocol-TX-Module #607: Poor range Spektrum AR630

oryfoxer, 2021-08-13:
Very poor range on Spektrum AR630 receiver. Range check took me 80 feet before I lost signal. Tried different Spektrum firmwares and all produced the same results.

Tried DSMX 2F and DSM2. Both produce the same issues.

OpenTX 2.3.14 
MPM 1.3.2.86

Tried disabling telemetry and channel map, nothing resolved the issue.

pascallanger, 2021-08-13:
The range check feature is common to all protocols and does not match the original manufacturer.

oryfoxer, 2021-08-13:
On a 636a and a 637T I can range check to 200 ft. So, there is still a performance issue with this receiver, despite not being set to manufacturer specifications.

oryfoxer, 2021-08-13:
If you live in the states, I can send you this receiver for testing. There seems to be a problem with the newer stuff.

I had a 637T that was giving me issues with range, then I upgraded it to the latest Spektrum firmware and the issue went away.

pascallanger, 2021-08-13:
I leave in France.
One thing you could test if you compile the code by yourself (explained on this github) is to enable frequency tunning on channel 15: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/1a921b1cdb9947707475211d53c324ee60cff75c/Multiprotocol/_Config.h#L129
By assigning a pot on channel 15 and turning it slowly, you could see if the range is improving.

oryfoxer, 2021-08-14:
No luck, not sure what the issue is. 

The only thing I can think is this is a bad RX or there is a compatibility issue with this RX. I replaced the RX in the airplane with a 636A and the issue is completely gone. I can range test to about 200ft before issues start to arise.

I gave it some thought and if you are comfortable with it, I will mail you the RX. Shipping to France is about $12 for the weight.

If no one else is reporting issues with this reciever, it may just be a bad one. 

Let me know how you want to proceed.

pascallanger, 2021-08-16:
@oryfoxer Can you confirm that you've tested the frequency tunning?
I'm fine if you want to send me the RX. I can have a look but I can't guarantee I can fix it. If you are fine sending it and want the RX back please consider the cost of shipping back.

oryfoxer, 2021-08-16:
> @oryfoxer Can you confirm that you've tested the frequency tunning?
> I'm fine if you want to send me the RX. I can have a look but I can't guarantee I can fix it. If you are fine sending it and want the RX back please consider the cost of shipping back.

I wasn't expecting it back. Just sending it to support the project. Like to see the Spektrum stuff supported and working as intended.

pascallanger, 2021-08-16:
> I wasn't expecting it back. Just sending it to support the project. Like to see the Spektrum stuff supported and working as intended.

Sounds good. Could you contact me on rcgroups user hpnuts? or using the email you see when sending a paypal donation.

oryfoxer, 2021-08-17:
The AR630 has been shipped via USPS. Not sure how long it will take to receive.

frankiearzu, 2023-10-30:
Just a comment on this ones... the AR630 don't have antenas..  so is very vulnerable to obstructions (like a battery, esc, etc). Has to be installed far away from those.   Since no antena, the range is lower to the equivalent AR631 (w antena). I had one and i reserved it for park flyer and short range  (definitly not motor-Gliders).