pascallanger/DIY-Multiprotocol-TX-Module #608: DIY module FS2A_RX dropouts

sasodoma, 2021-08-15:
Hello,

I wanted to have wireless trainer using my TX18S as the main radioand a FlySky FS-i6X as the slave radio. My radio came with a MPM and both FlSky2A and FS2A_RX protocols work on it without a problem. To do wireless trainer I needed another MPM that would receive the signal. So I bought and Arduino Pro Mini and an A7105 module and put together a simple MPM on a perfboard. Flashed it with 1.3.2.86 selecting only the A7105. The module works and binds sucessfully in FS2A_RX protocol and I can setup the trainer settings. But when using it the radio randomly says "Trainer signal lost" and the settings report "No MULTI_TELEMETRY detected". Sometimes the MPM is detected again on its own or I can turn the slave radio off and on, reset the arduino or disconnect and reconnect the serial connection between the radio and the arduino. All of these will make the radio detect the MPM again. Also all of this is with the internal MPM turned off, so no interference.

Is there a way to debug the firmware on an Atmega?

MPM hardware: Arduino Pro Mini 5V 16 MHz + XL7105 (A7105 module)
Multiprotocol FW version: 1.3.2.86
Radio: Radioking TX18S (Fake Jumper T18) with JP4IN1 (same as JP5IN1 without SX1276)

Schematic:
[image]

pascallanger, 2021-08-16:
What's the power supply voltage on the J1.3 pin? The issue I could see is that your 3.3V regulator (missing the input/output capacitors by the way) is probably not able to power the arduino+module and causing the resets or hangs up you see. Which is why on the original design you can see a first stage of bringing the voltage down to 5V then a second stage to 3.3V.

sasodoma, 2021-08-16:
Thank you for the quick response. I also suspected power problem, which is why I had already added capacitors to the regulator, but it didn't help. As per your suggestion I also tried adding a 7805 regulator before the LM1117 and just for good measure I soldered capacitors across the arduino and a7105 power pins. Still got dropouts. To completely rule out the regulator I powered the whole module from the 3.3V rail of an ATX power supply and still got dropouts. I did try switching to the FlSky2A protocol and just flying my whoop around the house and that seemed to work fine. Maybe something about the RX protocol is causing problems on the Atmega but not on an STM32 (the internal module). What else could I test?

closed 2021-09-25 by pascallanger