pascallanger/DIY-Multiprotocol-TX-Module #610: Add support for new f949g airplane boards, from cessna 182, wltoys airplane

LilTeo14, 2021-08-17:
Hello! i own a jumper t lite with a jp4in1 multiprotocol transmitter and open tx software, i already binded a lot of rc but in this one i tried all than i can found in internet and nothing worked.

Multiprotocol version: V1.3.1.78

First the pictures:

The airplane picture:

[image]

Airplane board pictures:

[image]
[image]

Receiver board pictures:

[image]
[image]

The airplane has not bind button, you turn on the radio, the airplane and they are connected.

Picture of the radio configuration:
[image]

Protocols i tried with that configuration:
FlySky-Std
FlySky-V9x9
FlySky-V6x6
FlySky-V912
FlySky-CX20
FlSky2A-Pwm,Ibus
FS2A_RX
Kn-WLtoys
MJXq-WLH08
MJXq-X600
MJXq-X800
MJXq-H26D
MJXq-WLH08
MJXq-E010(Very similar aspect board)
MJXq-H26WH
MJXq-Phoenix

And no one worked, thanks for your help, i dont know how to do reverse enginering but if is necessary i can learn.

LilTeo14, 2021-08-17:
In the manual the F949G board is also named P707G

LilTeo14, 2021-08-17:
Reading other issues, i found i need and logical analyzer, is this fine ? 
[image]

pascallanger, 2021-08-17:
It's a F949S. The S for stabilization indicated that you have a new board which is not Flysky compatible.
Lately the planes have been reusing an old protocol MT99xx : https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#MT99XX---17
Try this one too see if one of the sub protocols give you control. Multiple subprotocols could work but then you need to identify which one providers the right features for this plane (gyro could be on flip out something else so try everything).

LilTeo14, 2021-08-17:
Hello! Thank you for your answers, i tried all the MT99XX subtrocols and no one works, maybe it is necessary to have bind ch, no telem, or low power options turned on ?

pascallanger, 2021-08-17:
> bind ch, no telem, or low power options turned on ?
That won't do anything...

Hummm this is strange since the little RF board shown on your picture is usually doing the MT99xx protocol as I've seen the exact same in the multiple planes like the eachine mini wing dragon... Can you confirm that you've tried to bind the plane at the right time? Usually the board only listens for bind for a short period after being plugged. Look at the LED on the RX board, does the pattern changes when you try to bind it with MT99xx?
2 solutions after that:
 - see if your T Lite has some internal rx/tx pads on the multi module to connect a 3.3V serial to USB adapter so we can load a debug firmware and do a dump over the air
 - purchase the analyzer mentionned above and connect it on the original TX pins, RX+TX+GND

LilTeo14, 2021-08-17:
Nice, i will send a video about how i am doing it, if i am not wrong i can update the multi module firmware by the open tx software, i already did it with an external module but never tried with the internal module, and i will buy anyway the analyzer because i think it will be cool to learn how to use it jajajajaja.

LilTeo14, 2021-08-17:
https://drive.google.com/file/d/16ALs8gbjvGG3JPKwfR_AQpQoRXvNafRU/view?usp=sharing

pascallanger, 2021-08-17:
Is the plane doing this double flash as you as you plug it?

If you have an external multi module that would be the easiest. Load a debug fw, launch the xn297dump/auto protocol and use a terminal to see/record the output.

LilTeo14, 2021-08-17:
Yes, as soon as i plug it it start with the double flash, here is a video of binding with the original radio. 

https://drive.google.com/file/d/16JJREctMYJSFgzTa55iPu3jPTVW2byKe/view?usp=sharing

Where i can find tutorials to do what you are saing to me to do.

Thank you!!!

LilTeo14, 2021-08-17:
[image]

I think i found the debug fw for my module, plz tell me if it is.

LilTeo14, 2021-08-17:
I am reading all the documentation in the www.multi-module.org/ webpage and i am understanding a lot more, get me time and i will be able to do it jajajajaja.

LilTeo14, 2021-08-17:
O bad news "Protocol Dumping only works for protocols using an XN297L RF module, which has become common in many small models, and is only practical with external MULTI-Modules."
Only practical with external MULTI-Modules, let me check if i can get an external MULTI-Module!

LilTeo14, 2021-08-18:
I can get the tbs mpm, i am going thuesday for it.

LilTeo14, 2021-08-20:
I could not get the tbs mpm, can i do dump over the air with an a internal module ?

pascallanger, 2021-08-20:
As I said earlier you need to open your TX, locate the TX/RX pins of the module and connect a USB to Serial adapter on it. Share pictures of the internal of the TLite where the multi module is located and I'll guide you.

LilTeo14, 2021-08-21:
Here are the module pictures.
[image]
[image]

pascallanger, 2021-08-21:
The 5 pads above the STM32 is what we are looking for. Sadly the picture with your finger on the side is preventing to see what's ocnnected where...
Can you take a better picture?
Do you have a USB to serial dapter?

LilTeo14, 2021-08-21:
[image]

i already bought usb to serial adapter, it is going to arrive tomorrow.

pascallanger, 2021-08-22:
Ok you need to solder 2 wires between the multi module and the serial adapter:
[image]

- Plug the USB to Serial adapter in your PC, install the drivers which comes with it and find out the com port number it has been given. Don"t forget to set it to 3.3V before plugging it to the radio+module!
- Install the application Flash Multi: https://github.com/benlye/flash-multi. In the application set the correct com port and click on Serial Monitor
- Load a debug firmware on the module using OpenTX as you usually do.
- You should now see on the serial monitor the module booting.
- Create on the radio a new model and select the XN297Dump protocol and select Auto for the sub protocol. The module will start to scan packets over the air.
- Turn on the plane radio and let the module do a full scan. Hopefully after some time it will start to receive some packets and will display them on the screen. Once you think you have something, save the terminal output to a file and attach it here.

LilTeo14, 2021-08-24:
Hello! Open tx did not let me flash the internal debug firmware, because the hadware comptability check dont work with the dump version, can you give me plz the jp4in1 pinout as you give me above with the to rx and ground pins but with all the other pins to update the internal module with the ftdi, thank you!

pascallanger, 2021-08-25:
The order of the connector from the picture looks to be GND, Boot0, 3.3V, RX and TX.
To program the module from the outside you have to connect Boot0 to 3.3V when applying power (using a strap) which puts the STM32 in rom bootloader mode. Then connect your ftdi to GND, RX<-TX and TX->RX and use Flash Multi to send the firmware.