pascallanger/DIY-Multiprotocol-TX-Module #613: Radiomaster binding

ssdb1980, 2021-08-22:
I can't bind to my receivers.. Optima7 or Fs-iA6 receiver. Update firmware and protocols tried fine tune the frequency 
Tried every number and I still can't bind.
Any ideas that could help?

pascallanger, 2021-08-22:
Which radio master radio?
Which version of the module firmware?
Do you see the module version nearby the protocol selection?

ssdb1980, 2021-08-22:
Radio master tx12  firmware is 2.3.14
Receiver fs-iA6 protocol Flsky2A and I have tried all subprotocols
Receiver optima 7 protocol hitec subprotocol optima.

pascallanger, 2021-08-22:
You are not answering my questions...

ssdb1980, 2021-08-22:
Am sorry am new to opentx so am trying here. I have radio master tx12
module firmware is 2.3.11
This is what I see on the setup page
Internal rf
Mode muti
Type hitec
Subtype optima
No I can't bind to any receiver I only have 2

pascallanger, 2021-08-22:
Module firmware can't be 2.3.11... The module firmware is something like v1.3.2.86...
Can you do pictures of where you set the protocol?

HazzaHFPV, 2021-08-22:
2.3.11 is the version of OpenTX. Module firmware should be shown in the Modell-Setup page where you set the protocol under "status" as Pascal suggests.

pascallanger, 2021-08-22:
@HazzaHFPV  And on the second post he said that his radio is running OpenTX 2.3.14 so the information provided so far are not correct.

@ssdb1980 Please provide pictures of what you are doing.
Have you tried to upgrade the multi module firmware?

ssdb1980, 2021-08-22:
@pascallanger yes 
[image]
[image]
[image]

pascallanger, 2021-08-22:
Ok it's broken since you have flashed the firmware correct?

ssdb1980, 2021-08-22:
@pascallanger how did I fix it?

pascallanger, 2021-08-22:
By telling the truth and what you've really done would help...

ssdb1980, 2021-08-22:
I donwload all the updater files. I updated the bootloader will the radio was off and plug in to my computer though opentx companion 2.3  then turn it on to write firmware 2.3.14 to the radio then updated mult module to v 1.3.2.86

pascallanger, 2021-08-22:
But before all that was the module working?

benlye, 2021-08-22:
Could it be a 64Kb STM32?

pascallanger, 2021-08-22:
That's what I think too. But I would like to know if the radio was working before the updates.

ssdb1980, 2021-08-22:
@pascallanger no that is why I was doing the updating.

pascallanger, 2021-08-22:
When was the radio purchased? It's it brand new out of the box?

ssdb1980, 2021-08-22:
Yes it was band new

pascallanger, 2021-08-22:
Hummmm brand new, ok good to know.
Try and old release like this one: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.1.59
After flashing it check the status line under the protocol.

ssdb1980, 2021-08-22:
@pascallanger tried a few v1.3.230,v1.3.2.77, v1.3.1.77 v1.3.2.6 and v1.3.2.86 and still got nothing.

pascallanger, 2021-08-22:
Please listen, I've said and have you the link to the 1.3.1.59 version. I expect that no build would work above this version.

pascallanger, 2021-08-22:
By the way the TX12 only has a cc2500 RF chip so Flysky is not supported on this radio.
Only the protocols shown under this section are available: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#cc2500-rf-module

ssdb1980, 2021-08-22:
@pascallanger there are 34 on the link. So am not sure which one I need to download

pascallanger, 2021-08-22:
I said earlier chose opentx noinv after it's up to you.
Here is the aetr: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/download/v1.3.1.59/multi-stm-opentx-aetr-noinv-v1.3.1.59.bin

ssdb1980, 2021-08-22:
@pascallanger  
[image]
This came when it tried it

pascallanger, 2021-08-22:
Reduce the file name size after downloading the bin.

ssdb1980, 2021-08-22:
I did to 1.31.59.bin and to 1.5.9.bin
Same error messages

Miwer, 2021-08-23:
Had this happen to my TX12 bought off Banggood. Came with a C8 (64K) chip. After flashing the latest official image, same symptoms with message "no telemetry", and when I opened my radio, i could see the TX module blinking the SOS pattern.
Had to compile the firmware myself and was able to get it working again using normal flash procedure from SD card.
Thank you for the excellent guides on this issue. ;)

pascallanger, 2021-08-23:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md
And select the 64KB board.

We will add CC2500 64KB builds soon since both the TX12 and T8 are shipped with 64KB Flash chips...

Miwer, 2021-08-23:
@ssdb1980 before you go through the hoops of compiling the firmware, please do yourself a favor, and CHECK that you have the wrong chip in your radio - use the information here: https://www.multi-module.org/basics/module-hardware/unsupported-modules
You need to make sure you have the C8 chip and not the CB chip. 
This is how mine looked inside:
[image]

ssdb1980, 2021-08-23:
@Miwer this is mine

ssdb1980, 2021-08-23:
[image]
@Miwer this is mine

Miwer, 2021-08-23:
This is the one I compiled for mine. See if this will work for you. Use at your own risk (extract the zip file first)
[tx12-64k-aetr-1.3.2.88.bin.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7035253/tx12-64k-aetr-1.3.2.88.bin.zip)
MD5 checksum: cd7af135206e63cba5aea1cdd539fd3d

ssdb1980, 2021-08-24:
@Miwer hey what is MDS checksum

Miwer, 2021-08-24:
@ssdb1980 did you extract the zip file first?
Don't worry about MD5, you don't need it.

ssdb1980, 2021-08-24:
@Miwer yes

benlye, 2021-08-24:
Try the firmware file in this zip.  Make sure you extract the bin file from the zip before you copy it to your radio.  You may have to shorten the file name of the bin file.

This file was built using our standard GitHub release process, so should work.
[mm-cc2500-64-v1.3.2.89.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7041362/mm-cc2500-64-v1.3.2.89.zip)

ssdb1980, 2021-08-24:
@benlye I get these messages when I try flashing
[image]

[image]

pascallanger, 2021-08-24:
You can't flash the module through companion... You must cut the file to the SD card and flash it to the internal module using OpenTX under the SD card explorer.

Miwer, 2021-08-24:
@ssdb1980 that is not how you flash the multi-module. 
Have a look at this video, and try again - skip the download portion, and use the file as recommended by benlye
https://www.youtube.com/watch?v=lg9zF92mapM

ssdb1980, 2021-08-24:
Hey still getting error message. I   Want to thank all you guys for taking the time to help.
I have put in a ticket with radio master and order the 4in external module.
@Miwer @pascallanger @benlye

Miwer, 2021-08-25:
@ssdb1980 what is the error you get in the screen of the radio when trying to flash the firmware?

Miwer, 2021-08-25:
Well for what it's worth, I can confirm the file posted by benlye is working on my TX12 - also binding to receiver works.
@ssdb1980 I hope it works out for you somehow.
[image]
[image]

closed 2021-09-01 by pascallanger

krisdover, 2022-01-05:
Had the same issue with my RadioMaster TX12 using the **STM32 CC2500 only** download option (i.e. `mm-stm-cc2500-aetr-v1.3.3.0.bin`) from the Multi-module selection [download page](https://downloads.multi-module.org/). Even after finding this post and switching to the **STM32 CC2500 only - 64K** download option (i.e. `mm-stm-cc2500-64-aetr-v1.3.3.0.bin`) the module still didn't work (i.e. Status: No telemetry). The only thing that fixed it was the `mm-cc2500-64-v1.3.2.89.zip` file shared here by @benlye. 😅