pascallanger/DIY-Multiprotocol-TX-Module #616: Again a question about using a PC (with Python)

wvthoog, 2021-08-30:
Hi,

i would like to inform on how to use this project in combination with a PC (Linux based) I realize that others also inquired about this subject and tried to acquire as much information i could using this (issues) page and the thread on rcgroups.com

First off, what i would like to achieve is to control a mini drone (currently) an Eachine E016Hv2 with Python. As i already own a T18 Pro I've flashed it with the current multi-t18in-aetr firmware and can confirm that i can control the drone successfully.

So now i have two choices (i assume) 

1. to control it directly using a USB/SPI adapter icm with a CC2500 module and rewrite the E016Hv2 code somehow
2. to acquire a 4in1 module and interact with it using serial

Since i'm not too familiar with C++ I've gone ahead and acquired the JP4in1-SE module and was hoping it would just accept serial. But apparently it takes in SBUS (Baud 100000). So i guess i need another Arduino to convert the serial input to SBUS for the JP4in1.

Holding off on buying any more hardware and opted to ask which path to take here first. Would prefer to use the JP4in1-SE module since it allow me to control more than one type of drone.

So my question really drills down to this. Is there a possibility in DIY-Multiprotocol-TX-Module that allow me to sent (basic) serial commands to the module ? If not, what needs to be rewritten in order to allow that ? Alternatively, what path would you suggest 

Regards,
Wim

pascallanger, 2021-08-30:
Just check the closed issues on this github. People before you have done it successfully and shared some code.
Pascal

wvthoog, 2021-08-30:
Thanks Pascal. Didn't find it before but will go through it again.

wvthoog, 2021-08-31:
It's like finding in needle in a haystack. Went through the issues page completely, viewing any possible interesting title. Searched for , serial, uart, pc, computer but can't seem to find any relevant post.  Do you perhaps recall the title or date of one of these posts ?

pascallanger, 2021-08-31:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/discussions/501

dataphile-jeff, 2021-08-31:
> #501

Many thanks!

closed 2021-09-01 by pascallanger

wvthoog, 2021-09-01:
Awesome !

Got it up and running already with the JP4in1 and an E016Hv2

Very nice work. Would have taken me much longer to figure that out on my own