pascallanger/DIY-Multiprotocol-TX-Module #619: Radiomaster TX16s and WFLY WFR07S receivers disconnect them self at random

M-age-P, 2021-09-12:
Hi,

I have many of those "WFLY WFR07s" receivers left from my old radio and i wanna use them for my DIY Quadrocopters.

But when i bind them to my "Radiomaster TX16s" they disconnect them self at random even when the drone is next mo me.
Sometimes after a few seconds and sometimes after some minutes, then i need to restart the radio or a rebinding.

 i already crashed a drone and cut my thumb badly because i needed to disconnect the battery with a wild spinning prop...

I use the internal multi-module with the version "1.3.2.86 AETR".
Do somebody have the same receivers and have the same issues or know what to do?

Sorry for the bad English...

Greets Mario

pascallanger, 2021-09-12:
Couple of things:
 - Have you done the RF frequency tuning as indicated under the [RadioLink protocol](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md) details? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Frequency_Tuning.md
 - Have you setup your failsafe as indicated to prevent things like motors spinning when losing connectivity? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md

Pascal

M-age-P, 2021-09-13:
Hi again,

The WFLY receivers didn't have telemetry and there is no RF Finetune avaible so i cant do that.

I have tested all my different WFLY receivers i have 6x WFLY WFT07s and 3x WFLY RD201w....
ALL! of them keep disconnecting at random (after 2sec to 10min or so)... 

my flysky/frsky receivers working flawless.

Im pretty sure there is an error in the multi-module firmware. 
All the receivers working perfekt on my old WFLY Radio (also Tested).

Mario

pascallanger, 2021-09-13:
The WFLY protocol is using the CYRF6936 RF component which is different from Flysky (A7105) and Frsky (CC2500) so this does not rule out a module issue.
 - Have you set the right amount of channels in the option field? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#wfly---40
 - Have you setup your failsafe (not using "None" or "Hold") as indicated to prevent things like motors spinning when losing connectivity? https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#wfly---40 Is failsafe not working?
 - Have you tried different RX numbers?

OpenUAS, 2022-01-17:
Thx @M-age-P for the response, appreciated. @pascallanger Close issue maybe?