pascallanger/DIY-Multiprotocol-TX-Module #620: Devo telemetry GPS support for RX705

kristjanbjarni, 2021-09-13:
I have been testing and using the new Devo Telemetry changes added in this issue https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/324

I am using the Walkera QR X350 Pro with the Devo RX705 receiver and using the Jumper T16 transmitter.

This seems to work and telemetry values are being received however there is one issue and that there is a bug it seems to be in the RX705 receiver itself, so that it sends the GPS coordinates in decimal degrees (DD.DDDDDD) instead of the correct degrees minutes (DDMM.MMMM), when this value is then converted to decimal degress in OpenTX the GPS coordinates are incorrect. This issue is also a problem with the Tali H500 it seems, which probably uses the same receiver.

I reported this same issue originally for the DevationTX firmware:
https://www.deviationtx.com/forum/6-general-discussions/8209-problems-with-devo-f12e-nightly-builds#72467
and the DeviationTX team fixed it by adding a workaround by specifying the GPS format used.

Is there a possbility to do a similar workaround in the Multi Module? I am not sure how that would be implemented since there is no way to specify the format for the GPS coordinates. Perhaps there could be a secondary GPS telemetry sensor that is just the direct value from the sensor? Then at least it would be possible to display the value in a widget.

pascallanger, 2021-09-14:
I don't see how to solve this issue:
 - from a sensor perspective there can be only one GPS
 - from an OpenTX GUI protocol perspective, there is no field I can use to make it user selectable
I'm sending the GPS value received directly to OpenTX untouched (no calculation) using the FrSkyD telemetry. Maybe you can ask the OpenTX team about the possibility to choose the format on their end?

pascallanger, 2021-09-14:
Another way would be to write a lua script which does the translation. The OpenTX team should be able to help with that too.

kristjanbjarni, 2021-09-14:
I was afraid that maybe this was not possible to solve. One of the options would be to always recalculate the Devo GPS from decimal degrees to decimal minutes before sending it to OpenTX, however this could possibly mess up other Devo telemetry modules? For example the WK-CTL01-D which I don’t have. So this option is probably not a good one.

I have considered writing a lua script to reverse the calculation but the original calculation is destructive unless you know the original major decimal degree. I could recalculate four different positions since there is at most a single degree difference, and then one of these four positions should be the correct one.

I think maybe we should just close this one and I will either contact OpenTX team or find some kind of workaround.

pascallanger, 2021-09-14:
I can add a compilation option to switch from one to the other but you would need to compile by yourself and can't choose in the GUI.
Really need to see what OpenTX thinks about it.

pascallanger, 2021-09-14:
Is the R705 more common than the WK-CTL01-D?

kristjanbjarni, 2021-09-14:
> I can add a compilation option to switch from one to the other but you would need to compile by yourself and can't choose in the GUI.
Yes that could be an option, although I am not really very familiar with the Arduino IDE, but I think I could manage to build it.

> Is the R705 more common than the WK-CTL01-D?
I have no idea. It would be great if any other users that have Devo GPS telemetry could test how this behaves. This seems to a be a quirk with the RX705.

Have you ever thought about adding a generic parameter option to the GUI? Like having some generic "Options" field to be an integer value that you could set to some specific value/bitfield and the code could interpret that as some specific option for that protocol? Wouldn't that minimize requiring simultaneous UI changes with OpenTX for some very obscure options? I don't know, just an idea, but then I really don't know the specifics of how the multimodule communicates with OpenTX, so maybe not realistic.

pascallanger, 2021-09-15:
There is already a generic option value -127..127. But in the case of devo it's used to switch between auto and fixed id...
I was thinking that we've implemented a check box to switch between auto and fixed id on opentx but in fact the Option field is there just renamed Fixed ID. I should have checked... So I can use it to switch 0,1 will be one convention, 2,3 will be the other.
Ok now I need to find out how to convert from one format to the other.

kristjanbjarni, 2021-09-15:
This is how I think it is.

OpenTX expects to receive the GPS coordinates in NMEA format (Degrees decimal minutes):
DDDMM.NNNNNN,S

Example:
11706.956850,W

Which would translate to -117.1159475 in decimal degress

But it looks like RX705 is sending the value as degrees decimal but in the NMEA structure format like this:

11711.594750,W

So the fix would be to calculate

0.MMNNNNNN * 60 = EE.EEEEEE

Which result in this value: 6.95685

And the combined fixed value would be in this format:
DDDEE.EEEEEE,S

Which results in the correct value:
11706.956850,W

pascallanger, 2021-09-15:
Can you test v1.3.3.1 available here: https://downloads.multi-module.org/latest-test/
First step, make sure that the GPS indicates the same bad coordinates as before. They should be exactly the same.
Second step, power off your model and add 2 to the Fixed ID field, 0 becomes 2, 1 becomes 3. Power on your model and check if the GPS coordinates are ok.
Let me know how it goes.

pascallanger, 2021-09-16:
Can you test v1.3.3.2 (which is currently building and available in a few minutes)? There was a typo on latitude... I've done it late yesterday with no way to test  ;-)

kristjanbjarni, 2021-09-16:
Yes this works now :smile:

The GPS value shows the correct coordinates when Fixed ID=2

Thank you very much for this :+1:

pascallanger, 2021-09-17:
Great, I've updated the doc.

closed 2021-09-17 by pascallanger