pascallanger/DIY-Multiprotocol-TX-Module #621: Multi-Protocol Module JP4IN1-SE with FrSky Horus X10S Express and OpenTX version 2.3.14 doesn't work

AlphaCharlyX, 2021-09-25:
General information
OpenTX version 2.3.14 or 1e09791)
Hardware Frsky X10S Express
FW: opentx-x10express

Hardware Frsky Taranis X-Lite Pro

OpenTX Companion 2.3.14 on Win10

Hello,
I have the problem, that my new MultiProtocol-Module JP4IN1-SE doesn't work with my FrSky X10S Express and OpenTX version 2.3.14.

As you can see on the picture on the radio and opentx the status of the Multimodule is "No Multi-Telemetry detected". Binding models and Flashing the Multimodule is so impossible.

On my Taranis X9d and on my X-lite Pro with JR-Adapter the Multimodule works perfectly (binding models and flashing the module with the newest firmware is no problem here.

On the Multimodule itself is the newest Firmware 1.3.3.0 successfully installed.

[image]

I am pretty sure that I have the right firmware on the Multimodule because the same module works - as mentioned - perfectly on my FrSky X-lite Pro and on my Taranis X9d 2019. This is the back of my multimodule.

 
[image]

Thanks for your project and your help.

Greetings Alpha

pascallanger, 2021-09-25:
This is a known issue on the Horus X10S and not specific to the Jumper 4in1 but all multi modules.
A wire needs to be soldered to solve the issue. I'll try to find the post I made on RCG.

pascallanger, 2021-09-25:
Solder a wire as shown in red:
[image]

AlphaCharlyX, 2021-09-26:
Hello Pascall!
Thank you very much, for your great help. I did the soldering as you showed in the picture above. Now it is working perfectly.

Does this modification influence the function in my other radios? I haven't tested yet the soldered MPM in my X-lite Pro and X9d.

Greetings Alpha

pascallanger, 2021-09-26:
It shouldn't affect anything on the other radios.

enzo-molinari, 2021-10-01:
Hello

I have the same problem with my Frsky Taranis X9 Lite S and Irange iRX4 lite module. Could it be a fundamental problem for all access transmitters?

Many thanks and kind regards

Enzo

pascallanger, 2021-10-01:
May be... I don't know...

AlphaCharlyX, 2021-10-01:
Are somewhere pictures how to solder the IRangeX4 modules?

pascallanger, 2021-10-01:
It should be pretty similar. If you can't find out post a closeup picture of the PCB and I'll indicate where the wire needs to be soldered.

enzo-molinari, 2021-10-01:
[image]

AlphaCharlyX, 2021-10-02:
Hello Pascall,
thank you! Herr is a picture of the back of my IRangeX4 Module.

[image]

pascallanger, 2021-10-02:
@AlphaCharlyX 
Solder a wire as indicated by the red line:
[image]

pascallanger, 2021-10-02:
@enzo-molinari
Solder a wire as indicated by the red line:
[image]

enzo-molinari, 2021-10-02:
Thank you very much. As soon as I have soldered and tested it I will give feedback.
With kind regards
Enzo

AlphaCharlyX, 2021-10-02:
Everything works except flashing the Multiprotokoll-Modul through the X10S Express. The radio always shows "Sync error". With the other radios flashing the Modul is no Problem.

pascallanger, 2021-10-02:
@AlphaCharlyX if the telemetry works fine (the module version is displayed under the protocol), then the flashing problem is an opentx issue.

enzo-molinari, 2021-10-04:
Hello Pascal

I just found out that the MPM works on my X9 Lite S and my x10s Express without soldering a bridge. You only have to switch off the internal module. This means that only one module can be in operation at a time. Either the internal or the external.
Many thanks and kind regards

Enzo

AlphaCharlyX, 2021-10-05:
> Hello Pascal
> 
> I just found out that the MPM works on my X9 Lite S and my x10s Express without soldering a bridge. You only have to switch off the internal module. This means that only one module can be in operation at a time. Either the internal or the external. Many thanks and kind regards
> 
> Enzo

In my case and with my X10s Express, that did´t work.

HazzaHFPV, 2022-02-01:
Is the 10S running the Opentx Version from the FrSky website? If yes, frsky removed support for third-party modules in their own version. Official version loaded via the Companion should show the module as option.

wollenagold, 2022-02-06:
Hello together, 
now it's running, i think the problem was sitting in front of the PC ;-). 
Because loading the 2.3.14 with companion i forgot to mark the position "external Modul with unsupported Firmwares" After reloading the 2.3.14 with this option all is working well and i could connect my different RX'es with SFHSS. But the External Modul isn't visible when looking in the Groundsettings of OpenTX. Telemetrie with my FrSky-RX in my testmodel is also running so that's ok for me.
Wish you all a good time. Stay without Corona and have good flying times.

yura869, 2022-08-28:
Здравствуйте. У меня horus x12s с модулем irange irx4+.  В horus прошивка opentx 2.3.15, в модуле 1.3.14 (последняя). Провод падшая как показано на рисунке выше в обсуждениях, но всё равно пишет, что модуль не подключён. В что может быть причина? 
[image]
[image]

yura869, 2022-08-28:
Провод припаян. Причëм такая ситуация с двумя модулями irx4+.
[image]

AlphaCharlyX, 2022-09-27:
I think your soldering is not good.

Mars2588, 2022-10-02:
Hey, i have the same issue as some others in this thread. I attached the setting of my Horus x10s. The Module is a Irangex plus with the yellow rotary switch. I soldered the two pins. But didn´t change anything. I have no idea what can i do to get a solution. 

This module worked fine for me till I updated the internal IXJT Module. Since I updated there is no way to get it work. 

What i already tried to get it back to run:
-flash different versions of the firmware version of the 4in1 module.
-Did flash two times the latest OTX firmware via Companion.
-Soldered the two pins as shown in this thread.

It would be great if anyone of you had an idea what can i try to get the module back to work on my horus x10s.

[image]

[image]

pascallanger, 2022-10-02:
Even if the internal module is off?

Mars2588, 2022-10-02:
The internal module is off. Or is there a other way to turn it off. Because the attached picture show exactly a turned off internal modul!

Mars2588, 2022-10-02:
For connecting the two Pins in the irangex Board i used a self bondering wire. This one ist quiet thin. I can try to solder a thicker one and try again. Or isn't that a problem. 
Any other ideas?

Mars2588, 2022-10-03:
I changed the soldered cable didn't Change anything. 
The Modul itself seems to Work because with a radiomaster there isn't a Problem. Did anybody of you have an Idea what can i do that the horus x10s works with the multimodule?

alm-nl, 2022-10-07:
Are you sure it's an official OpenTX 2.3.15 version that you used? The date is kind of weird, October 2nd 2022? Official release was from 2022-04-22, therefore 2022-10-02 is weird. If you used binary from FrSky itself, please try with official version from open-tx.org (download and install Companion, configure the radio settings to reflect your radio and download the firmware, then flash the radio).

Mars2588, 2022-10-08:
Yes. The date is the date of the Flash of otx 2.3.15. I downgrade to 2.3.7 since the downgrade the mm irangex works very well.

jimmy6616, 2023-09-02:
@pascallanger i have the same issue with an X12S running a Radiomaster RM4in1, no telemetry, can you help please.

alm-nl, 2023-09-02:
> @pascallanger i have the same issue with an X12S running a Radiomaster RM4in1, no telemetry, can you help please.

It helps if you include clear (high resolution and readable) pictures of both sides of the PCB. I don't know if Pascal knows the exact board type you have.

pascallanger, 2023-09-02:
Do exactly has shown in this thread. I don't know how to help you more than that, this is not software but hardware related.

Unb0rn, 2023-09-30:
I can report the similar problem.
I have an X12S with several mods - external ACCESS and inverter replaced. Haven't used MPM for a while.

I have an iRangeX 4+ module with mod from this thread, it reports no telemetry. I also borrowed another IRX4 with quite an old FW. Without mod it reported no telemetry, with mod it reports "wrong protocol" and after updating it to the latest version it has the same no telemetry message. So two IRX4/IRX4+ modules with exactly the same faulty behavior in X12S, both work fine in X9D+ 2019, both can be updated in it.

This update thing that changed the behavior from "no telemetry" to "wrong protocol" indicates that it may be a software issue as well. I've also updated several things on X12S since I last used MPM - both internal ISRM module firmware and transmitter's fw (runs EdgeTX 2.9.1 now).

alm-nl, 2023-09-30:
Just to make sure, you have disabled the internal module while using the (modded) IRX4/IRX4+ module? Also, did it work when you did not have the external ACCESS modification done to the X12S?

Unb0rn, 2023-09-30:
Okay, I've run several tests and these are my findings:
- Downgrading EdgeTX to 2.8.0 didn't help (only firmware, not sdcard contents or radio.yaml). Maybe I need to try 2.7...
- Turning internal module on and then back off changes the status message from "no telemetry" to "no serial input"
- Other external modules work fine (tried R9M 2019 with ACCESS mod and a couple of ExpressLRS modules(in CRSF proto) - Thor 2.4 and DIY 433MHz, both work fine, both are able to communicate with LUA script and models, Thor even works on F1000 mode, so probably the inverter is exceeding 400k baudrate without any problems - could the problem be in this port baudrate negotiation?
- Haven't tried downgrading ISRM firmware yet

@alm-nl yep, tried with both isrm enabled and disabled. Regarding your second question - if only I could remember =) I had the same problem as this topic describes, probably back in 2021 or 2022, then I made this solder bridge and it worked for me. After that I haven't used the MPM for a while... It could be 2.7.x branch of EdgeTX and probably I already had this ACCESS mod, but I'm not 100% sure. EdgeTX now allows to use this "externalaccessmod" thing without rebuilding the firmware, so I just disabled the ACCESS port. It didn't help.

Maybe I should open the same issue in EdgeTX too

kevinlieb21, 2024-03-07:
I've got the same problem as @jimmy6616 - Radiomaster RM4IN1 external module and a Horus X10S radio. Does anyone know which pins are supposed to be joined? There are some good pictures of the PCB in [here](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/880) I can provide more detail as needed.  Thanks!

Merga189, 2024-11-13:
Same issue here as experienced by @kevinlieb21 kevinlieb21and @jimmy6616

And it does not seem to be a general problem, but only affects here the combination FrSky X10(S) Express and the RADIOMASTER RM 4IN1.

My FrSky X10S Express with edgeTX 2.10.5 / OTX 2.3.15 works with no problem with those modules

[image]

but there is an issue with that module on the right here

[image]

and I have two! of those IRangeX modules here. And NO soldering on both IRangeX board was necessary.

All I see when trying the RM 4IN1 on the X10S Express either with edgeTX or OTX ist that:

[image]
[image]

and flashing the module from the radio

[image]
[image]

And the funny thing is, that putting that RM 4IN1 in one of my FrSky Taranis X9D/X9D+ radios (OTX 2.3.15)   ... no problem!

Neither flashing the RM 4IN1 from the Taranis with firmware nor using the RM 4IN1.

br Kalle

oSPANNERo, 2025-01-17:
It looks like I am hitting this same issue with my X10S Express running 2.10.5 with the Radiomaster RM4IN1. Unfortunately, neither of the PCBs that have been addressed look like mine. @pascallanger when you have a moment, could you please tell me where I need to solder the jumper on this board?

[image]

[image]

Merga189, 2025-01-17:
Pictures of the RM board were published in  Sep 2, 2023 here https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/880

There was no answer then and I fear, you will get no answer now.

br KH

alm-nl, 2025-01-17:
If you have a multimeter you can probably find where the wire needs to go if you measure from the other side of the diode (anode side) that is below (right one on the photo) and connected to the C86 chip (pin 7). It should connect to a resistor, measuring zero ohm between the resistor and the diode (anode side).
Then I think you have to connect a wire between the top-side of the diode (cathode/marked side, which is also connected to the C86 chip on pin 7) to the other side of the resistor that is directly connected to the anode of the diode.
This is based on what I see in the picture of the first Radiomaster 4 in 1 module in this topic (it has the same C86 chip), although the resistor sizes are different.
For safety, check if you're not creating a short between ground and the pins where you put the wire because that can damage the module.

Merga189, 2025-01-17:
As I do have 2 of those RM modules here, I am willing to do a test, but it needs a sketch of the required wiring.

oSPANNERo, 2025-01-17:
> If you have a multimeter you can probably find where the wire needs to go if you measure from the other side of the diode (anode side) that is below (right one on the photo) and connected to the C86 [...]

Thanks for the instruction @alm-nl, I pulled the board back out and after looking at it up close and validating with my multimeter it appears there is no resistor connected to the anode of the diode. Instead it goes straight to the pin of the JST connector which just goes to the external connector to the radio. (See attached photo.) Assuming I understood you correctly and what I am observing is correct, seems like I would just need to delete the diode and jumper the pads. Does that sound correct?

[image]

alm-nl, 2025-01-17:
> > If you have a multimeter you can probably find where the wire needs to go if you measure from the other side of the diode (anode side) that is below (right one on the photo) and connected to the C86 chip (pin 7). It should connect to a resistor, measuring zero ohm between the resistor and the diode (anode side). Then I think you have to connect a wire between the top-side of the diode (cathode/marked side, which is also connected to the C86 chip on pin 7) to the other side of the resistor that is directly connected to the anode of the diode. This is based on what I see in the picture of the first Radiomaster 4 in 1 module in this topic (it has the same C86 chip), although the resistor sizes are different. For safety, check if you're not creating a short between ground and the pins where you put the wire because that can damage the module.
> 
> Thanks for the instruction [@alm-nl](https://github.com/alm-nl), I pulled the board back out and after looking at it up close and validating with my multimeter it appears there is no resistor connected to the anode of the diode. Instead there is only a trace that goes straight from the diode to the pin of the JST connector which then just goes to the module's external connector to the radio. (See attached photo.) Assuming I understood you correctly and what I am observing is correct, seems like I would just need to delete the diode and jumper the pads. Does that sound correct?

No, I wouldn't do that unless you want to risk breaking things. Have you measured from the right side of the diode (as in your picture) to all sides of the resistors on both sides of the board (what is reachable at least)?

oSPANNERo, 2025-01-17:
> No, I wouldn't do that unless you want to risk breaking things. Have you measured from the right side of the diode (as in your picture) to all sides of the resistors on both sides of the board (what is reachable at least)?

I just re-checked and the only two points I am finding continuity with the anode of the diode in question is the pin of the JST and the cathode of the diode next to it. (See attached picture.) I will say I am definitely just a "hobbiest" so its entirely possible I am screwing something up but I would still be willing to bet a taco lunch on it.

[image]

oSPANNERo, 2025-01-17:
That said... if trace through the second diode it comes to a 10K resistor ("103") which then leads to pin 48(?) of the STM31F1 microcontroller. That pin looks to be [VDD_3](https://www.utmel.com/components/stm32f103c8t6-microcontroller-pinout-datasheet-and-circuit?id=194).

[image]

alm-nl, 2025-01-17:
This is the original schematic I found in the github repository: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/STM32%20PCB/Schematic_Multiprotocol_STM32_MB_v1.0_t.jpg

[image]

The yellow line is what was missing in the earlier photo's. What it does is add a resistor of 470 ohm in parallel over the upper diode. Maybe this resistor was left out of the board of RadioMaster because it didn't connect anywhere (only a jumper which was empty).

oSPANNERo, 2025-01-17:
So, again, assuming I am not missing something it sounds like what I might need to do is attach a 470ohm resistor to the same pads as the diode as that would effectively give me the same parallel path of diode and resistor?

I wonder if the diode serves a purpose other than protection? Cause if that is its sole job, I might take the chance of just swapping it out with the resistor and live "dangerously".

alm-nl, 2025-01-17:
I think you will be safe to go, it's to limit the voltage/current that flows when pin 1 (output) of the SL74LVC2G86 is pulled down. Adding a resistor will also limit the current that can flow but will be more detectable by the radio I guess.

Merga189, 2025-01-17:
.... and that quite simple solution is unknown to RM?

I opened an issue with RM on that non working with my X10 last year.

This are the signals CPPM and SPORT recorded when powering the transmitter on

X9D+ and the RM4IN1

[image]

Horus and an IRangeX MPM module

[image]

and Horus and RM4IN1

[image]

RM was not able to present such a simple (one wire to get the SPORT going) solution, but send me another of those modules.

Good luck!

oSPANNERo, 2025-01-17:
Guess I need to go find a 470 ohm resistor and see how bad my microsoldering skills are.

alm-nl, 2025-01-17:
> and Horus and RM4IN1
> 
> [image]
> 
> RM was not able to present such a simple (one wire to get the SPORT going) solution, but send me another of those modules.

Clearly shows something ain't right. 👍

alm-nl, 2025-01-17:
> Guess I need to go find a 470 ohm resistor and see how bad my microsoldering skills are.

A normal 470 ohm 1/4 Watt or less resistor might probably fit just fine when you bend the legs. But you'll definately need a fine tip soldering iron and some electronics flux can be helpful as well. And some fine-tip tweezers ofcourse. 😉

Merga189, 2025-01-18:
@alm-nl 

fingers crossed! And please post a pic of your soldering. Might help others. 

br KH

alm-nl, 2025-01-18:
> [@alm-nl](https://github.com/alm-nl)
> 
> fingers crossed! And please post a pic of your soldering. Might help others.
> 
> br KH

You meant @oSPANNERo 😉

Merga189, 2025-01-18:
Cab happen. Sorry. Then good luck to @oSPANNERo ;-)

oSPANNERo, 2025-01-18:
> [@alm-nl](https://github.com/alm-nl)
> 
> fingers crossed! And please post a pic of your soldering. Might help others.
> 
> br KH

Hopefully components should arrive the beginning of the week.

Merga189, 2025-01-18:
WORKS!!

[image]

Just a normal resistor, don't have a 470 SMD at hand.

Not modified RM

[image]

and modified

[image]

YOU ARE THE BEST @alm-nl

oSPANNERo, 2025-01-18:
Nice work @Merga189

oSPANNERo, 2025-01-22:
Confirming @Merga189's confirmation, replacing the diode with a resistor allowed the unit to be seen by X10S in both EthOS and EdgeTX. (Though the "Low RSSI" warnings in EthOS when connecting to a AR6610T while standing 12 inches away sent me down a rabbit hole for a while. ) Thanks again @alm-nl!

[image]

Merga189, 2025-01-23:
Why remove the 1N4148? I think, I understood the advice, that the resistor should be parallel to the diode.

[image]

> What it does is add a resistor of 470 ohm **in parallel over the upper diode** (in your photo). 

I ordered 1/8 W resistors, which should fit nicely next to the diode. The resistors, I have here are a bit to large ....

[9 more replies]