pascallanger/DIY-Multiprotocol-TX-Module #622: No possibility to work with Internal XJT HORUS S12 and external Multimodul together ???? #8693 opened 28 days ago by Githubbachelor

Githubbachelor, 2021-10-01:


Githubbachelor, 2021-10-01:
Hallo dear Specialists
I have another time a Firmware latest which does not support longer my 6 Airplanes with installed 2 Receiver!

Before, time ago you fixed it to use both receiver, the internal and external MULTI Modul. I want to use my planes in Holiday now and again that desaster that the use of Intern and extern Multimodul in my Horus is not possible??? If I want to use the extern MULTI M. the internal will be disabled??? Did we go backwards or forwards? I am sitting now with my big planes at a lake and can do nothing!! Why can I not use longer my both receiver Internal D8 with channel 1-8 and external over my Multi Modul latest firmware another D8 with channel 9-16? That was working all the years! Maybe you can tell me with which firmware for Horus S12 it does work.
Hope for a quick reply to use my planes at the lake now and not next year. And I think that it is not a problem with the MULTI MODUL, I try also older firmware for Multi M. with same problems.

Githubbachelor, 2021-10-01:
I use two Receiver D8 with DSMX or D8 with D8 and now all not possible again after the last bug I reclamed - there was the problem that the the binding was losing after a new switch on of the Horus. And you fixed that problem. But not we cannot bind two receiver anymore!
I have all latest firmwares installed

Githubbachelor, 2021-10-01:
There was a version which work with both tx moduls, internal XJT and Extern Muli all protokols with X12S - now again not working! I use many models with d8 at internal and different receiver for external Modul - please fix that bug

pascallanger, 2021-10-02:
This in an OpenTX issue. Nothing to do with Multi.

closed 2021-10-02 by pascallanger

pascallanger, 2021-10-02:
OpenTX 2.3.14 has a bug with the external modules not being handled properly. The latest nightlies have a fix. Try this if you don't want to downgrade.

Githubbachelor, 2021-10-06:
Hallo Pascal, The nightly 2.3.15 581N has the same problem. If 2 Receiver the internal will be disabled