pascallanger/DIY-Multiprotocol-TX-Module #643: 4DRC V8 (Tx: RW1608P, Rx: CP298L)

PinkLolicorn, 2021-12-09:
Any hope for this drone? I'm currently in no possession of multiprotocol radio, so I was unable to tinker much. Anyway, here is the data I was able to collect from the drone hardware itself.

### Tx
**RW1608P**
T377DSY42036

Same as Eachine e015? https://www.deviationtx.com/forum/protocol-development/8099-eachine-e015-flight-boat-car

[image]

[image]

[image]

[image]

### Rx
**CP298L**

It seems that chipsky does not provide full documentation on their website, here is basic info:

http://chipsky.com/view/product/p5.html
http://www.chipsky.com/data/upload/files/20170516_800.pdf

```
Chip Introduction: 
CP298L is a single chip 2.4GHz transceiver chip embedded in baseband communication protocol 
The single chip 2.4GHz transceiver chip works in 2.4GHz~2.483GHz ISM band, 
Suitable for ultra-low power wireless applications. 
The design of CP298L wireless communication system only needs 
The design of CP298L wireless communication system requires only one MCU and few peripheral devices, and it is a Low-cost wireless system solution. 
The modulation method adopts GFSK modulation, the maximum transmitting 
Transmitting power can reach 4dBm, receiving sensitivity in 1Mbps rate can reach -88dBm.
On-chip integrated voltage regulator ensures high power
The integrated on-chip voltage regulator ensures high power source rejection ratio (PSRR) and wide supply voltage range. 
The minimum operating voltage is 1.8V and the maximum operating voltage is 3.6V.

Chip Features:
◆Operating frequency band: 2400~2483MHz
◆Low operating voltage: 1.8~3.6V
◆Null port rate: 500Kps, 1 Mbps, 2Mbps
◆SPI interface: 4-wire system, maximum rate is 10Mbps
◆Modulation method: GFSK
◆available low-cost crystal: ±60ppm
◆Ultra low power consumption
◆Hardware auto-answer and auto-retransmission
◆Fast channel switching, can be applied to frequency hopping algorithm
◆Operating frequency band: 2400~2483MHz

Typical applications:
◆Wireless keyboard and mouse
◆Wireless avionics
◆Wireless industrial control

Translated with www.DeepL.com/Translator (free version)
```

[image]

pascallanger, 2021-12-09:
Without having access to the TX and RX there is little I can do...
You can either send me the hardware to France or sponsor me so I can buy it.

PinkLolicorn, 2021-12-09:
Ok, so this is, as I understand, confirmed - these chips are currently not supported. I wasn't really sure about that.

I feel like investing some time into learning a new skill. Furthermore, I actually have a couple of similar cheap drones and would like to contribute so all of them would be supported, including new models I will come by in the future. How would you approach the reverse-engineering of these? I found the `multi-module.org` note about that quite incomplete.

I will be honest, I haven't done a lot with that type of hardware. First and last time, was tinkering with the proprietary Zigbee gateway UART, so nothing special besides of soldering.

I think the software side (being proficient in more high-level languages) will be not much of a problem, besides of understanding the protocol and ecosystem a bit more. As you may tell, I'm currently missing the required hardware.

I'm planning on acquiring Eachine TX16S radio as a first step. Is this enough to test the new protocols, or this is would be mainly a final testing platform? Do I actually need SDR like HackRF or the Eachine radio can be used in similar fashion?

pascallanger, 2021-12-09:
This chip is fully emulated by the CYRF6936. The E015 is flying great with Multi.
You would need a SDR to find the RF frequencies being used, their order and make a dump of a couple of packets to be decoded.

PinkLolicorn, 2021-12-09:
What SDR do you use/would recommend for these purposes? Is there anything other than HackRF One that would do the job and is worth considering?

pascallanger, 2021-12-09:
I'm using a PlutoSDR.

PinkLolicorn, 2021-12-09:
I will be able to start working on this somewhere next week when the hardware arrives. Couple of questions for now:

> their order

Do you mean the frequency hopping pattern or something else?

> make a dump of a couple of packets

Wouldn't it be required to dump packets for all actions? Like the auto-start/auto-land button or fine-tuning pad.

pascallanger, 2021-12-12:
Frequency hopping pattern

A few packets already to see the structure of what's sent. Reversing a protocol is taking a lot of time and effort. Just with a few packets we will know already if it's doable or not...

PinkLolicorn, 2021-12-14:
My HackRF finally arrived. After turning on, the tx is broadcasting on 2460mhz (and something is visible on 2480mhz too?):

[image]

Not sure how to approach it now, but I got these from [jopohl/urh](https://github.com/jopohl/urh):

[image]

[image]

Do you have any software recommendations? I would also appreciate some additional guidance with the recording and decoding process. I have the samples, but I'm not sure if this is the format you would like.

[image]

Anyway, here are recorded signals (.complex16s):

**2.46GHz**
- [HackRF-20211214_165346-2_46GHz-2MSps-2MHz--5.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7713010/HackRF-20211214_165346-2_46GHz-2MSps-2MHz--5.zip)
```
- drone is off
- remote on
- pairing
```

- [HackRF-20211214_170712-2_46GHz-2MSps-2MHz-pair-takeoff-landing.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7713067/HackRF-20211214_170712-2_46GHz-2MSps-2MHz-pair-takeoff-landing.zip) 
```
- drone on
- remote on
- pairing (throttle up then down)
- takeoff (throttle up)
- bringing drone down (throttle down)
- auto-landing button
```

**2.48Ghz**
- [HackRF-20211214_171438-2_48GHz-2MSps-2MHz.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7713130/HackRF-20211214_171438-2_48GHz-2MSps-2MHz.zip)
```
similar to the second 2.46GHz signal but from the different take
```

[image]

PinkLolicorn, 2021-12-14:
### Frequencies

After turning on the remote, there is only one frequency active:

- 2459,9 MHz

[image]

After pairing with no receiver (throttle up then down):

- 2450,9 MHz
- 2455,9 MHz
- 2473,9 MHz
- 2477,9 MHz

[image]

### 2459,9 MHz

Remote before pairing seems to be producing repeating 5 fragments with no real data (?). These are with 1.5 ms delay between 5-fragment segments, 0.38 ms between each fragment and with fragment length of 0.22 ms. Each segment is 2,64 ms:

[image]

[image]

Then the signal changes with no action at the end of the tenth segment:

[image]

```
10001110001111100000111110000011111110000001111111111100000000000001111111111111111111111111111111111111000000000000000000000001111111111111111 [Pause: 767 samples]
01100001100000111100000011111111110000000000000000010001111100011100011100011100011110001110001111000111000111000111100011110001111000011110001 [Pause: 2935 samples]
110011001100110011001100110011001110011001110001100011000111000110011100111000110001110001110001110001110001110001110001110001110001110011100011100011 [Pause: 767 samples]
```

[image]

```
00000111111111111111111110010000001111100000011100000111110000011100011100000111000001000111110100011100011101110111000101110001011101110111010 [Pause: 767 samples]
11000000000001111110000011100011100010001110111110111000100010001011100010001000101110111010111010111111010101000100001010101010101010 [Pause: 772 samples]
00011100011100011101110111000101110101110101010111111010101010111101010101010101010101010110101010010101101010010100101101 [Pause: 771 samples]
10001000100010101000101010101010101010101010011111110101010101101010101101011010110101101001010011010110110110100110100 [Pause: 769 samples]
011101010100010111111111101010100101010010101101010010010110110100100101101101011001011011011011001001001001101100110110 [Pause: 3027 samples]

1111100011111011100010001011101000101000101011101010101010101010111101010101010101010101011010101010010100101011010100101 [Pause: 770 samples]
1011100010101000101010101010110101010101010110101001010010100101001011011010010010110110110110110110110010010011001101 [Pause: 770 samples]
01010101010101011010110101001011010010010110101101101100101100100100100110110010011011101101100110011011001100110110011 [Pause: 771 samples]
01010101010110100100101101101101101101101100110110011011001100110011001100110011001100110011001100011001100111001100111001 [Pause: 770 samples]
11111010101001011011010010010011011011001101100100110011011101110110011001100011001100110011100111001110011100011100111000111 [Pause: 2938 samples]

000111011101000101010101010101010101010010101011010101101011010010100100101101101101101101101101100100110110010011011000 [Pause: 768 samples]
10111010101101010101101011011010110110110100110100100100110110010110010011011001001101100110110011001100110011001110011000 [Pause: 767 samples]
01010101010110101101001101101101100100100110110011001001110011001100110001100110011001110001000111001110011100011000111000111 [Pause: 769 samples]
10101101010010110100100110110010011001101110110011001100111011100110011101110011000110011100111000110001100011100111000110001 [Pause: 767 samples]
111101010110110110010010010011011001100110011001110111001110011001110011100111001110011100011100111000111100111100001110000111110 [Pause: 2945 samples]

011101110101010101010101010010101101011010100101101101101001101101011001011001011011011001001100100110111011001100110011 [Pause: 768 samples]
10101010010101101001011011011011011001001001100110011001100110011001100110011001100110011001100110011001100110011001100110011001 [Pause: 767 samples]
1001100110011001100110011001100110011001100110011001100110011001100110011001100110011100111001100111001100110001100011001100011100110001100 [Pause: 766 samples]
10011001100110011001100110011001100110011001100110011001100110011001100110011001110011001110011100110001100111001100011100111001110011100111 [Pause: 767 samples]
10011001100110011001100110011001100110011001100110011100111001110011100111001110011100110001100011100111000110001100011100011100111001110001110011 [Pause: 2926 samples]

100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100111001100110 [Pause: 767 samples]
01100110011001100110011001100110011001100110011001100110011100110011001100011001110011000110001100110011000110011000110011100110011100111001110 [Pause: 767 samples]
1001100110011001100110011001100110011001100110011001100111001100110011000110011001110011000110011000110001100111000110011100111000110001100 [Pause: 766 samples]
100110011001100110011001100110011100110011000110011100110011100111001110001110011100111001110001110001110011100111000111000111000111000111000111000 [Pause: 766 samples]
01100110011001110011001100111001100111000110011000111001100011100011100111001110001110001110001110011100011100011100011100011100011100011100011100011 [Pause: 2943 samples]

101100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011001100110011100110011100111001 [Pause: 767 samples]
1001100110011001100110011001100110011001100110011001100110011100110001100110011100110011000110001100011000111000110001100011100011100011100011 [Pause: 768 samples]
[... 108 lines cut ...]
000000000000010010101010011001000111111111111111101011001111110110101010100001001011010101010101010001011111100110000010110111111011001011100011001001 [Pause: 738 samples]
[... cut ...]
I cannot really figure out how to get Universal Radio Hacker to produce same binary output for these two. I noticed the center moved quite a lot in the second signal and in general this sample is a lot more "dirty". Here is the capture of the first one:

[HackRF-20211214_223638-2_46GHz-2MSps-2MHz-pairing.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7715167/HackRF-20211214_223638-2_46GHz-2MSps-2MHz-pairing.zip)

And this is the part from the analysis above, the signal drops after pairing:

[image]

pascallanger, 2021-12-15:
Sorry but I'm really busy lately with familly issues and will come back to you I hope in not too long.

ergpopler, 2023-05-28:
Hello, is there any new updates? I am looking to possibly use a teensy 4.1 to amplify the signal and extend range and control from my linux device.