pascallanger/DIY-Multiprotocol-TX-Module #645: unknown receiver in wltoys cessna can't find a protocol for binding

ogarridoa, 2021-12-14:
After reading many information about this plane and how to do the binding I've found that the one that I have received did not work with any of the suggested protocols.
I've tried MT99xx/a180 and all typos of esky/flysky protocols but no luck.
[image]
[image]
any chances that anyone know about this card?

pascallanger, 2021-12-15:
Which Cessna is it? F949G or F949S or ???

ogarridoa, 2021-12-15:
It is a F949, no gyro, brushed motor. just this one

https://es.aliexpress.com/item/32872588719.html?spm=a2g0s.9042311.0.0.274263c0X3oDwQ

r-foerster, 2021-12-15:
I used to fly Wltoys F949 with multi protocol, but this one is different.
Transmitter looks different and the one from Wltoys has servos directly on board.

pascallanger, 2021-12-16:
The description from your url states that it is a F949S:

> WLtoys-avión teledirigido F949, juguete de avión de ala fija, versión actualizada RTF, Servo Digital **F949S** con giroscopio, 3 canales, 2,4G, 3D6G

Can you update the module to the latest version 1.3.3.4 available here: https://downloads.multi-module.org/latest-test/
And then use this protocol/sub protocol: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-a180---5

ogarridoa, 2021-12-16:
OK, I'll give it a go and let you know.

thanks

Bill12223, 2025-03-14:
wltoys f949s cessna-182, protocol

pascallanger, 2025-03-14:
The F949S is in the documentation...
Check here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-a180---5

CNCDAN, 2026-10-22:
Hello all.  I've been wrestling with this wltoys hj-wl949r 2024.12.15 for longer than I'd like to admit.  I've been able to bind everything under the sun to my Radiomaster tx16 running the latest firmware but nothing will connect to this wltoys green board.  I've tried everything recommended that I can find but hopefully I'm just missing something.  I'd very much welcome any suggestions.  I have some little planes to work on and this tiny board would be perfect for them.  Many thanks.