pascallanger/DIY-Multiprotocol-TX-Module #646: ZHCHORC TOYS / ZC Z50 CESSNA (Tx: XNS1042CV, Rx: XNS1042CV)

PinkLolicorn, 2021-12-14:
This is the current data gathered on the ZHCHORC TOYS Z50 plane. I will be extending this issue further when my multiprotocol radio arrives.

### Tx

**XNS1042CV**
2114CTa

Mentioned in the Eachine E016H issue: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/469#issuecomment-727174903

[image]
[image]
[image]

**Tx awaiting pairing**

[image]

[image] [image]

### Rx

**XNS1042CV**
2114CTa

[image]

[image]

pascallanger, 2021-12-15:
The XNS is a XN297 compatible component. If you have an external multi module, it should be straight forward to reverse. On an internal module you need to soder 2 wires to a FTDI(USB to serial 3.3v).

pascallanger, 2021-12-15:
You want to make is something like this: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/610#issuecomment-902468945
You cn read the full thread to learn but here is a summary. Connect GND and TX from the internal multi module to the GND and RX of your FTDI. Then load a debug firmware and find the XN297DUMP protocol, sub proto auto, on the PC open a terminal or flash multi serial monitor and share the output file.

PinkLolicorn, 2021-12-15:
I would also like to ask: which external module is best suited for this application?

pascallanger, 2021-12-15:
On any external multi module, you would just plug the usb cable, install the debug firmware and dump. No modification/soldering/... required.

PinkLolicorn, 2021-12-15:
Ok, so this would be a completely separate setup from my current radio that runs on the USB power as the external device - just like HackRF, is that correct?

pascallanger, 2021-12-15:
The external multi module would be plugged on the back of the radio, connected to the PC using USB and configured through the radio UI under the protocols.

PinkLolicorn, 2021-12-15:
Oh, ok. I have found "IRangeX IRX6" modules and similar version with the right connector for my radio:

[image]

Is this one suitable for the dumping of XN297-based devices? It lists Eachine E010, but I don't see any obvious indication where to solder the FTDI.

I noticed this one is mentioned in the https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/157 so my suspicion that this is NRF24L01 module is right.

It seems like using the full module for the sole purpose of dumping is a bit of a waste, so this one would be nice with permanently soldered FTDI, and is even fairly cheap.

pascallanger, 2021-12-15:
No it doesn't run my code, not upgradable, short range,...

PinkLolicorn, 2021-12-15:
Thanks for the clarification. Are there any single-chip modules worth considering?

pascallanger, 2021-12-15:
no...

PinkLolicorn, 2021-12-15:
[image]

Is this the correct TX and GND?

PinkLolicorn, 2021-12-15:
[image]
[image]

I'm not sure what is the cause of this. Standard firmware flashing works fine. Is there a chance the downloads page has an incorrect version of the firmware for the external module?

[image]

pascallanger, 2021-12-16:
> [image]
> 
> Is this the correct TX and GND?

Yes this is correct. (previous was off by one pin ;-) )

pascallanger, 2021-12-16:
> [image] [image]
> 
> I'm not sure what is the cause of this. Standard firmware flashing works fine, even usbdebug version works fine. Is there a chance the downloads page has an incorrect version of the firmware - for the external module instead of the internal one?
> 
> Edit: Seems like the same problem as in the example issue: [#610 (comment)](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/610#issuecomment-904998161)
> 
> [image]

The file is the right one... Which version of OpenTX are you running? Latest versions are supposed to not check inv noinv anymore since this is an autodetection in place instead.

PinkLolicorn, 2021-12-16:
> previous was off by one pin

Yes, I edited this post three times in one minute because the alignment of printing was so confusing that I accidentally marked the wrong ones in the screenshots two times in a row while first correctly telling their position physically (by order). 😆 

> The file is the right one... Which version of OpenTX are you running? Latest versions are supposed to not check inv noinv anymore since this is an autodetection in place instead.

It would be 2.3.11-otx (4328055e). My radio had it preinstalled. I thought of upgrading when this issue came to light, but it seemed a lot more hassle than updating multi alone, and I didn't see anything related in the changelogs. I guess there is no escape from that now. Likewise, I'm going to upgrade it and try again with the ftdidump.

PinkLolicorn, 2021-12-16:
So, I got 2.3.14-otx (1e09791a) installed, and it still does not allow me to flash with ftdidebug version.

pascallanger, 2021-12-16:
Same error message?

pascallanger, 2021-12-16:
Can you try the bin from the latest version: https://downloads.multi-module.org/latest-test/

PinkLolicorn, 2021-12-16:
Yes, it looks the same, at least to me. Using 1.3.3.6 also yields the same message.

[image]

[image]

[image]

[image]

[image]

pascallanger, 2021-12-16:
Hummm ok, we can give it a no inv file. I'll compile you something this evening after work.

benlye, 2021-12-16:
I don't think that will help.  The reason the radio won't flash the file is that the FTDI debug files are built without support for the bootloader.  All internal modules have the bootloader installed so even if you were able to flash this FW to the module from the radio, it would not work (the FW simply won't run).

Our Arduino board definitions don't actually have a way to build an FTDI debug FW with bootloader support.  The only way to get a debug build on the internal module would be to connect an external FTDI adapter and flash it from the Arduino IDE or using Flash Multi.  This would remove the bootloader so afterwards you would need to use the IDE or Flash Multi to re-install non-debug FW and the bootloader.

PinkLolicorn, 2021-12-16:
I don't really feel like messing that much with my internal module. Seems like I will be getting iRangeX IRX4 for this job, which should arrive at the end of the year (and plus version would be an additional month, so I'm going to ignore that it exists).

benlye, 2021-12-16:
It's not as bad as it sounds - you can solder wires to the pins and have a 5-pin dupont connector on the other end ready to attach to your FTDI adapter.  When you're not using it you can tuck it under the antenna/USB port cover and only pull it out when you need it.  You do have to deal with BOOT0 as well, which requires putting a diode in the wiring harness so that you can leave it attached all the time.  

But yeah, that is a lot of hassle compared to using an external module.  A JP4IN1 would work as well if you can get one of those more easily than the IRX4+.  The USB port is convenient to have on any external module.

PinkLolicorn, 2021-12-16:
I don't see JP4IN1 anywhere in stock. I'm ok with soldering the missing bits to the IRX4 later if needed. Not only that, but I don't really like the protruding rotary switch of the plus version anyway. Also, it was 40% less for the basic version, so there's that.

Lagarda, 2021-12-28:
Hi to Everyone!
what is XNS1042CV? Is there any analogue? Do you have a datasheet?

PinkLolicorn, 2022-01-12:
I got my iRangeX, but I was unable to connect it successfully with my PC. It works fine in the radio, pairs with the rx. I can flash different firmware from OpenTX etc.

I was, however, not able to do anything from the Flash Multi. It behaves the same as the module was never connected and for example displays "Failed to read the MULTI-Module".

I had no success with flashing debug firmware for ftdidebug from the OpenTX due to the same issue as before but opposite side (needs inv file) and seen no output with usbdebug. Also tried checking for output with PuTTY.

Connecting the USB yields a windows' notification with the contents of: `unknown usb device (device descriptor request failed)`. For a moment I thought this may be a bootloader issue but isn't it already available as I can flash the module from the OpenTX?

I tried installing the drivers as described here, made sure the module was connected with the PC: https://github.com/benlye/flash-multi/blob/master/doc/Troubleshooting.md#solution-1

Not sure how to proceed.

benlye, 2022-01-12:
When the module is plugged into the PC, what are the LEDs on it doing?

Also, does it have a physical bind button? If so, can you try plugging it in while holding the button down?

benlye, 2022-01-12:
Something else you can try...

It sounds like you're running recent firmware, which doesn't have support t for the USB port, but the old bootloader.

You can try flashing this FW to the module using the radio
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases/tag/v1.3.1.59

After that you may be able to use Flash Multi to install the newer bootloader, as described here
https://github.com/benlye/flash-multi/blob/master/doc/New_Bootloader.md

PinkLolicorn, 2022-01-12:
> When the module is plugged into the PC, what are the LEDs on it doing?

Should they light up when the module is connected to the USB, being outside the radio? Because they don't, only if in the radio. It wasn't entirely clear in the many places where the module should be and does it matter.

> Also, does it have a physical bind button? If so, can you try plugging it in while holding the button down?

Well, it does not, and I only have bigger buttons of that type, but I could probably add some jumper. Which pads need to be connected?

I added most of these connections myself and wonder if there may be something "broken" with the ftdi/usb in the nonplus version on the hardware level, but seems unlikely as LEDs work? I don't think I broke something, at least visibly, no bridges, no misplaced components, I even cleared the module with isopropyl alcohol at some point just to make sure there are no unexpected connections from residual flux etc.

[image]

> It sounds like you're running recent firmware, which doesn't have support t for the USB port, but the old bootloader.

Just flashed it and tried briefly with no changes. I'm still getting the same error notification and no COM port is registered. Should I try something more specific?

benlye, 2022-01-13:
When you plug the module in via USB it's important that it's not being powered by the radio - is your radio off?

When you first plug it in the red led should flash very quickly, then a little slower, then even slower.  If you watch the first 20s or so of this video you'll see what I mean - https://youtu.be/IvNgOQxHdD0

PinkLolicorn, 2022-01-20:
> If you watch the first 20s or so of this video you'll see what I mean

Well, my module does not light up when connecting just to the PC like in the video.

[image]

I really wonder if there is something going on with this nonplus version that I have that prevents both USB and FTDI from working. Everything seems to be there, but I'm not so familiar with the design.

I think I will try some other usb sockets or soldering directly to the breakout board to exclude something being funky about these I got currently. But FTDI is not working too, so it does not fill me with optimism. 😆

benlye, 2022-01-20:
Very odd. My module (the one in the video) is also the non plus with only the LEDs and USB port added.

I notice that you haven't put headers on the BOOT0 pins. You can connect to the FTDI pins with an external USB to serial adapter (like a CP2102), including 3V to power it, but you do need to bridge BOOT0 on order to flash firmware or the bootloader that way.

closed 2022-07-07 by pascallanger

filipkotoucek, 2022-12-26:
Hello. I just received ZC-Z50 Cessna without the main motor in front (don't know if that matters). I had touble to binding to it using irangex x4plus module with FW 1.3.3.20. TX is Taranix qx7 with edgeTX 2.7.1. I tried both subprotocols for GD00X even with fine tuning over the full spectrum with 40 increments.

This led me to thought if the protocol is not somehow different. I tried to dump the communication. Here is the log:

Only TX is on:
```
Protocol selected: 63, sub proto 3, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=6, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7

Packet detected: bitrate=Using NRF
250K C=7 S=Y A= E7 E7 E7 E7 E7 P(4)= AA 67 64 01
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7
RX on channel: 7, Time:     0us P: AA 67 64 01
RX on channel: 7, Time:  2994us P: AA 67 64 01
RX on channel: 7, Time:  2994us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2991us P: AA 67 64 01
RX on channel: 7, Time:  2992us P: AA 67 64 01
RX on channel: 7, Time:  3000us P: AA 67 64 01
RX on channel: 7, Time:  2986us P: AA 67 64 01
RX on channel: 7, Time:  2991us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2988us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2986us P: AA 67 64 01
RX on channel: 7, Time:  2996us P: AA 67 64 01
RX on channel: 7, Time:  2993us P: AA 67 64 01
RX on channel: 7, Time:  2995us P: AA 67 64 01
RX on channel: 7, Time:  2993us P: AA 67 64 01
RX on channel: 7, Time:  2997us P: AA 67 64 01
RX on channel: 7, Time:  2987us P: AA 67 64 01
RX on channel: 7, Time:  2994us P: AA 67 64 01

1 RF channels identified: 7
--------------------------------
Identifying RF channels order.
Time between CH:7 and CH:51

Channel order:
7:     0us

--------------------------------
Identifying Sticks and features.
```

The com between model and TX happens on ch 48 and 51. On identifying sticks I moved:
1. rudder stick (first byte changed)
2. trims (second byte changed 10 -> 11 etc)
3. RATE? ( second byte chanched 10 -> D0). This is quite interesting because the change does not have any impact on plane behaviour. There are 2 buttons on the TX front side. 
   a. Left button which changes the value in dump beeps once everytime I click it but does not have impact on planes behaviour.
   b. Right button does not change the values in dump. First click beep twice, second click beeps once, third click beeps twice. And there is audible plane behavior change between those two modes, so I assume this would be the RATE button.

```
Protocol selected: 63, sub proto 3, rxnum 0, option 0
XN297 dump, address length=5, bitrate=250K
Proto=XN297DP, nbr_sub=6, Sub=Auto  , Opt=8, FS=0, CHMap=0, rfSw=2
------------------------
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,461

Packet detected: bitrate=Using NRF
[... 166 lines cut ...]
P: 85 10
P: 84 10
P: 83 10
P: 82 10
P: 81 10
P: 80 10
P: 80 11
P: 80 12
P: 80 13
P: 80 12
P: 80 11
P: 80 10
P: 80 0F
P: 80 0E
P: 80 0D
P: 80 0E
P: 80 0F
P: 80 10
P: 80 D0
P: 80 10
```

Sadly I don't know how to catch the bind in the logs.

Is there anybody who would know how to help me find where is the bug or what am I doing wrong?