pascallanger/DIY-Multiprotocol-TX-Module #648: Trying to reduce number of protocols - no luck

nostromo123, 2021-12-26:
Radio FrSky X10 Express 
OTX 2.3.14
Multi FW 1.3.3.0
Arduino IDE 1.8.18 under Debian Bullseye
flash-multi-0.6.1 under WIN 10

MPM module used

[image]
[image]
[image]

In -Config.h changed AETR to TAER and disabled all protocols but MLINK and FRSKYx.
[_Config.h.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7777011/_Config.h.txt)

The Arduino IDE upload protocol tells me 'wrote and verified' for bootloader and firmware. 
[Neue Datei.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7777014/Neue.Datei.txt)

But I still got all 'disabled' protocols on the radio!?

I tried with 'usb support' enabled and disabled
[image]
I tried with/without a jumper on boot-3.3V
I copied 'Multi4in1/StmMulti4in1.bin' and flashed with flash-multi under WIN 10, then erased before flashing, then activated debug and could still see all protocols in terminal window.

I am stuck!

br KH

nostromo123, 2021-12-26:
This MPM ships with an CKS32, but is recognized as STM32 here
[image]
[image]

pascallanger, 2021-12-26:
Which software is running on your radio and version?

nostromo123, 2021-12-26:
[image]

pascallanger, 2021-12-26:
can you take a picture of the protocol section on opentx?
does the module fw version appears there?

nostromo123, 2021-12-26:
I have here two *.bin files generated under Arduino IDE with USB support set in IDE  disabled (-usb) and enabled (+usb)
With **reduced protocols** (only FrSky and M-Link) 34k and 39k size.
[image]

Under WIN 10:
**- read the module (-usb)**
[image]
**- erase the module**
[image]
**- write (+usb)**
[image]
**- read module (+usb)**
[image]

**Now pics from the radio (no rx bound!)**

switching between those two displays

[image]

[image]

No module Info, but I think this is issue 
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/621
Flashing from X10 Express also not possible!

[image]

My aim here was by disabling **all** protocols but FrSky and Mlink in -Config.h to reduce the scrolling between protocols on the radio, but I still seem to have **al**l protocols visible??? 

/****************************/
/*** PROTOCOLS TO INCLUDE ***/
/****************************/
//In this section select the protocols you want to be accessible when using the module.
//All the protocols will not fit in the Atmega328p module so you need to pick and choose.
//Comment the protocols you are not using with "//" to save Flash space.

I interpreted that comment, that by disabling, the protocols don't show up, but I might be wrong!?

PS. Cold you comment please on 'Arduino IDE with USB support set in IDE', as I did not find any information in the documentation.

nostromo123, 2021-12-29:
Sorry, I at least found the answer for my add. question in RCG post ;-)

> Cold you comment please on 'Arduino IDE with USB support set in IDE', as I did not find any information in the documentation.

https://www.rcgroups.com/forums/showpost.php?p=45466633&postcount=19826

**But back to my basic question, I disabled most protocols in _Config.h, reduced the *.bin file to 34k and still those disabled protocols show up, when scrolling the protocols on the radio.**

Is that behavior supposed to be like that?

br

closed 2022-07-07 by pascallanger