pascallanger/DIY-Multiprotocol-TX-Module #649: Bind with Turnigy Ia6c (Flysky2A) fails

tchleb, 2021-12-27:
Hi,
I have tried to bind to wo different Turnigy Ia6c RXs. Unfortunately the TX want bind. 

I use Radiomaster TX16s with internal 4 in 1 with the latest Firmware V1.3.3.0

Is there a solution for binding to Turnigy receivers?

pascallanger, 2021-12-27:
These RXs bind without an issue. You need to follow the std process:
 1. put the bind strap on the RX
 2. power on the RX
 3. launch a bind from the TX

https://youtu.be/7uroYR77cgs
 
Notes:
 - the TX and RX needs to hear each others so try either really close or at a distance.
 - you must select a uniq RX number by RX

tchleb, 2021-12-27:
Thank you for your fast reply.

I already tried this steps.
1. Press the bind button on RX
2. Power on the Ia6c
3. On my remote I selected Flysky2A, PWM_ibus, 50Hz and choosed a unique receiver ID and pressed the bind button.

The remote does not bind with the ia6c. I tested to different ia6c. 

I tested successfull a uz65 with building Flysky2A receiver.

closed 2022-07-07 by pascallanger