pascallanger/DIY-Multiprotocol-TX-Module #655: Add protocol for FMS A 18 "mini" 540mm

Zebble, 2022-01-17:
Just snagged this nice little model: https://www.aliexpress.com/item/1005003567728014.html

It's listed as FMS and seems to be a smaller version of their PA 18 1700mm.

I couldn't find a protocol that works so I did an XN297 sniff/dump. Seems similar to the MT99xx protocol so I tried all subprotocols in MT99xx and can't get it to bind. Also didn't see anything different in what should have been the bind process, it just starts spitting out the same channel data.

The board/brick doesn't look similar to anything else I can find, and there are no model numbers on the board. The controller looks identical to the Dragon Mini Wing except the wording on the front says "EAZ RC, ERC-1 2.4GHz".

I've attached the dump and here's a summary of the data:

byte[0]: Throttle - low=E1, high=00
byte[1]: Rudder - left=E1, right=00
byte[2]: Aileron - right=E1, left=00
byte[3]: Elevator - forward=E1, back=00
byte[4]: Elevator trim
byte[5]: Aileron trim added with Mode switch (beginner = 00, mid=20, expert=60)
byte[6]: 08 = RTH (stays for 2 seconds/80 packets after press), 80 = button press (aerobatics) - instantaneous, single packet, even if held down. Only works in beginner and mid modes.
byte[7]: Rudder trim plus seems to alternate adding 00/40 between packets.
byte[8]: CRC/Checksum?

Anything else I can help with to get this protocol added, just ask!

Thanks
[xn297-fms-pa18-540mm.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7883835/xn297-fms-pa18-540mm.zip)

[moved this report to github, as per hpnuts' request]

-zeb

Zebble, 2022-01-17:
To answer your questions from rcgroups:

"The checksum is different. They've added 0x89 to it. I don't know where this value is coming from...
Can you capture a bind packet? This might give a clue."

I can't seem to get it to go through the bind process.  No matter what I do, it immediately connects and starts sending the same packets in XN297, unless I'm missing something?

"The byte[5] description looks a bit weird to me. Can you explain how the trim behave? Since on the other channels trim 20 means centered but for this one and only this one it is different?"

byte[5] seems to mathematically add 00, 20 or 60 to the value of the aileron trim, depending on the position of the mode switch (beginner, mid, expert, respectively).  Does this make it clearer?

Not sure why they did this either, or the rudder trim having 00/40 added in alternating packets.

pascallanger, 2022-01-17:
To receive bind packets, set the sub protocol to 1Mb and the RF channel to 2. The module will try to receive packets from there.
Then power on the RX only and don't touch anything on it. If you still get nothing try to move slowly the throttle up and then down.

Zebble, 2022-01-18:
Very cool!  I did what you've indicaated and do see something happen when the TX is turned on.  I've attached the output.  Note that I tried it 3 times during the session so have separated the 3 attempts in the txt file with "----------------------".

I didn't touch the throttle.  It just started spitting stuff out.

I hope this helps
[fms-bind.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7885608/fms-bind.zip)
!

pascallanger, 2022-01-18:
You haven't changed the RF channel to 2 (Channel=0,0x00) which is why there are so many bad readings. But anyway I have what I need. Let's see if I can figure it out now.

pascallanger, 2022-01-18:
The biggest problem I have at this stage is that all sub_protocols have been used for this protocol. I need to create a second entry...

pascallanger, 2022-01-18:
If you power on the TX first and do throttle up and down. Wait a minute then power the plane. Does the plane works right away or you need to power cycle the TX for it to work?

pascallanger, 2022-01-18:
```
"The byte[5] description looks a bit weird to me. Can you explain how the trim behave? Since on the other channels trim 20 means centered but for this one and only this one it is different?"

byte[5] seems to mathematically add 00, 20 or 60 to the value of the aileron trim, depending on the position of the mode switch (beginner, mid, expert, respectively). Does this make it clearer?
```
It does not. While dumping what values does the trim takes on Aileron when you go to min, mid, max? Do the test with the different modes and report.

pascallanger, 2022-01-18:
I've got a code nearly ready which is waiting for some of your answers from above.
I've modified MT99xx so much that I'm crossing fingers that I haven't broken anything...

Zebble, 2022-01-18:
> If you power on the TX first and do throttle up and down. Wait a minute then power the plane. Does the plane works right away or you need to power cycle the TX for it to work?

It connects right away.

Gathering the aileron trim values now.

Zebble, 2022-01-18:
> ```
> "The byte[5] description looks a bit weird to me. Can you explain how the trim behave? Since on the other channels trim 20 means centered but for this one and only this one it is different?" [...]

Here are the values for aileron trim/mode switch combo:

byte[5]			beginner	mid	expert

ail trim min (left) 	3F		7F	BF
ail trim mid		20		60	A0
ail trim max (right) 	01		41	81

Zebble, 2022-01-18:
> You haven't changed the RF channel to 2 (Channel=0,0x00) which is why there are so many bad readings. But anyway I have what I need. Let's see if I can figure it out now.

You have what you need but I'm wondering how to change channel?  I thought I did that by changing the Receiver #.  I'm using EdgeTX and don't see an "options" field.

pascallanger, 2022-01-18:
> You have what you need but I'm wondering how to change channel? I thought I did that by changing the Receiver #. I'm using EdgeTX and don't see an "options" field.

It should display RF channel with a value of which you can change from -127 to +127. But not an issue I have everything I need. I'll create a build later today.

pascallanger, 2022-01-18:
> Here are the values for aileron trim/mode switch combo:
> 
> byte[5] beginner mid expert
> 
> ail trim min (left) 3F 7F BF ail trim mid 20 60 A0 ail trim max (right) 01 41 81

I understand now.

Zebble, 2022-01-23:
FYI, have opened an issue with EdgeTX to get the "RF Channel" option added.
There's a lot of confusion around this XN297 dumping feature so I may document my experience in the Wiki at some point, in hopes it makes it easier to add future protocols.

pascallanger, 2022-01-23:
I haven't tested anything but you can try to install this firmware: https://downloads.multi-module.org/latest-test/
Then test with the protocol MT99xx2/PA18: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#MT99XX2---92
Please report.

Zebble, 2022-01-23:
Thanks Pascal!  Just did a quick test and doesn't seem to bind...  I can do a XN297 dump from another radio if you need?

pascallanger, 2022-01-23:
If you have a second radio that could help.

Zebble, 2022-01-23:
Yup, I'll get this test firmware installed on a Jumper T-lite and do the sniff from a TX16S with an external module.  I'll do that later today.

pascallanger, 2022-01-23:
Oh I thought you had another original radio... Doing a dump from what I'm doing won't help me...

I've pushed 1.3.3.9, the build is ongoing, wait 5 minutes before redownloading the test vesion. If you can see if it works.

Zebble, 2022-01-23:
I only have the 1 original controller.  Was just going to sniff the bind channel with the original and compare it to what was coming out of your test version if that helps.

Tried 1.3.3.9 and still won't bind.

Zebble, 2022-01-23:
If it helps, I just sniffed the original controller and your 1.3.3.9 version, with the plane off:

On the original controller, I get 138 packets like:  RX:  8423us C=2 S=Y A= CC CC CC CC CC P(9)= 20 14 03 25 C9 02 13 89 55

On 1.3.3.9, I get 54 packets like:  RX:  9280us C=2 S=Y A= CC CC CC CC CC P(9)= 20 14 03 25 4F 78 00 89 55

The multimodule only attempts this on first try.  I can't BIND to try again - it beeps but nothing happens on channel 2.  I have to turn off the radio and turn it back on to try again.  Even then it's not consistent.  Sometimes it tries to bind, sometimes it doesn't.  If I change receiver number, sometimes it works again.  Seems to be most consistent when Bind on Channel is enabled, but even then not 100% consistent.

Zebble, 2022-01-23:
I have a friend with the same plane.  Going to see if I can borrow his TX...

pascallanger, 2022-01-24:
Can you try v1.3.3.10?

Zebble, 2022-01-24:
Still no bind.  I'll work on getting the 2nd radio.

pascallanger, 2022-01-25:
Can you try v1.3.3.11?
After if it still doesn't work, I'll debug it on my radio since I haven't took the time to do it yet.

Zebble, 2022-01-25:
Thanks Pascal.
Still no bind with 1.3.3.11.
Take your time.  This is all just fun stuff.  Family always comes first.

Zebble, 2022-02-03:
I was able to grab a second controller (and plane) and have attached a captured XN297 dump and attached.  I've noted the controller changes in the file.  I tried:

controller 1
controller 2
controller 2 again
controller 1

I turned the plane off and on between attempts, of course.  Both controllers bound fine
[capture - 2 controllers - FMS.txt](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/7998924/capture.-.2.controllers.-.FMS.txt)
.

pascallanger, 2022-07-07:
Sorry I dropped the ball... Are you still interrested?

wiilover22, 2026-08-15:
UPDATE: Seems to be using CH12, 27, 42, and 57 at the same time while controlling the plane. The data is the same on all channels it seems. PID=0 all the time.

Below what it looks like when TX is bound and controlling plane.

RX: 48541us C=12 Enhanced pid=0 S=Y A= C7 95 3C BB A5 P(12)= 21 E2 0D B6 00 80 80 82 04 00 42 10

1  2  3  4  5  6  7  8  9  10 11 12

21 E2 0D B6 00 80 80 82 04 00 42 10

1. 21 always same

2. Hex continuously counting to max and restarting

3. 0D always same

4. B6 always same

5. 00 -> E6 Thr 0->100 (Doesn't go to FF for some reason)

6. 00 -> FF Rud L->R

7. 01 -> FF Ele  0->100

8. 00 -> FF Ail L->RF

UPDATE:
9. 04/84/44 MODE Down/Mid/Up Gyro Off/Optimized/Stabilized - These are default values, and the lower values reset on each power cycle.
	When pressing stunt button, the second value goes to 'C'
	When setting the RTH direction, first value goes up two digits (0->2, 8->A, 4->6)
	When activating the RTH feature, first value goes up one digit (0->1, 8->9, 4->5)
	With both RTH set, and activated, first value goes up three digits (0->3, 8->B, 4->7) 

10. 00 always same

11. 06/0A/0E/12/16/1A/1E/22/26/2A/2E/32/36/3A/3E (up to 7E) Rud Trim L -> R 42 is Middle

UPDATE:
12. 30/50/70/90/B0/D0/F0/10/30/50/70/90/B0/D0/F0/"10"/30/50/70/90/B0/D0/F0/10/30/50/70/90/B0/D0/F0 Left <-> Mid <-> Right Ail Trim (The 10 in the center is middle)
	When increasing/decreasing Ele Trim, the first value goes up/down one digit at a time (0/1/2/3/4/5/6/7/8/9/A/B/C/D/E/F)

I also figured out some of the bind mode stuff! When I manually set the MPM to 1Mpbs, and RF channel to -1 (so it scans), I found when the controller is in bind mode, it's only active on CH28, with different information:

RX: 20219us C=28 Enhanced pid=1 S=Y A= C7 95 3C BB A5 P(12)= 61 28 0D B6 00 81 80 82 00 00 AA 55

All values stay the same, HOWEVER, PID counts from 0->3 then resets over and over.

The light on the RX doesn't work at all with the TX16S. When bound to the original TX, the light blinks waiting for TX, solid when bound, and blinks when RX voltage is low.

The original TX beeps when plane RX voltage is low, so there's some kind of telemetry...but I can't figure out how any of that is being transmitted.

pascallanger, 2026-08-15:
Working on a new version with the above

pascallanger, 2026-08-16:
Please test v1.3.4.66, it should do what you have above. Channels assignment: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-t28

>What I meant about the throttle calibration...my planes throttle seems to go from 0%->100% within a few millimeters of my TX16 throttle moving. Wasn't sure how to go about calibrating that, or if that's done in the code you provide.

I think this is because the plane was in vertical mode and therefore trying to hold on the motor/propeller as soon as it could. The vertical mode is not accessible anymore so I'm guessing this is gone.

> The light on the RX doesn't work at all with the TX16S.

I think you need to assign a switch to the light channel CH7 to turn on the RX LED.

> The original TX beeps when plane RX voltage is low, so there's some kind of telemetry...but I can't figure out how any of that is being transmitted.

That should already work since they haven't touched the code too much. When you power up and bind the plane to the radio, you should see the signal bar on the radio and get a telemetry lost message when you power it off. For the battery status, it's in the doc link above.

wiilover22, 2026-08-16:
I haven't been able to fly the plane yet, windy rainy day =( However, some of the channels are a bit different from the wiki:

1. CH1-4 AETR (Wiki currently correct)
2. CH5 Mode (Wiki currently correct)
3. CH6 Unused? I can't seem to get CH6 to do anything on the plane currently (Different)
4. CH7 Receiver light turns on at +100%, off at 0/-100% (Wiki currently correct)
5. CH8 Flip toggle? This plane didn't have a dedicated "flip mode" before...however I've never successfully used this planes "stunt" mode on the original TX...(This plane has had a really rough life, with lots of crashes...but it's one of the reasons it's become a personal favorite lol) 
With CH8 at +100% AND CH5 0% (Stabilized mode), the FC/gyro puts the plane into a permanent inverted angle orientation...like it would do flips until you turned the mode off. (Different)
6. CH9 Circle toggle? Same as above, no dedicated "circle mode" before...however with CH9 at +100% AND CH5 at 0% (Stabilized mode), FC/gyro puts plane into permanent right turn utilizing the ailerons and elevator. (Different)

I'm personally perfectly happy with this current setup (so maybe just a wiki change is needed?), although it would have been neat if somehow that previous "vertical mode" could somehow also be added. I haven't been able to set the RTH function...but honestly it never worked well...and I never used it. It simply set whatever direction the plane was facing, as the "RTH direction".

I've tried the remaining channels, up to CH16...and couldn't find anything else. Can confirm I have full throw on my throttle, unlike before.

Edit: Also, wanted to say thank you so much for responding so quickly, and being so helpful with this. Greatly appreciated!

pascallanger, 2026-08-18:
Ch6 is doing exactly what you described when you press the stunt button on the original TX.
Not sure about what the other channels do... Could be some unused or debug features...
Not sure why I've sent 13.2V, whatever...

yawarts1, 2026-08-19:
SGF22 protocol T28 can bind, but all the servo motors rotate to their limit and lock up.

pascallanger, 2026-08-19:
>SGF22 protocol T28 can bind, but all the servo motors rotate to their limit and lock up.

Which version are you using? Which plane is it?
As you can see above @wiilover22 has no issues.

yawarts1, 2026-08-19:
FMS 410 mm MINI T-28 , version  v1.3.4.68

https://www.fmshobby.com/products/410mm-t-28-receiver-4-in-1-integrated?srsltid=AfmBOoqk6gsX00q8bczzbV-3JCk90PJhY46mGv2CsSmJMcZVcVb6hpZc

wiilover22, 2026-08-19:
@yawarts1 check if CH8 is +100%. That's not not simply a "flip" toggle, it's an "inverted flight" toggle. I was able to take my FMS 410 mm MINI T-28 out yesterday, bound to my TX16S MK3, and tested all features. It flew fantastically, and I don't think anything should be changed...other than the wiki definitions. 

> Ch6 is doing exactly what you described when you press the stunt button on the original TX.

So the instructions to do a stunt for the original TX require you to "Set RTH" on the ground, then once flying, "Activate RTH"...THEN press the stunt button (which starts a beep on the TX)...THEN giving aileron or elevator inputs will cause the plane to do "stunts". To exit, you'd have to reverse that process...all while flying.

Now I tried using their terrible RTH feature a few times, never used it long enough to test their "auto stunt" feature. I just flew the plane manually, and loved it. 

I'm guessing with a TX16S just sending raw commands, it's skipping some kind of internal check on the original TX's side...which honestly is a good thing. Now, when you enable CH8 +100%, the plane instantly inverts and HOLDS. Then you can just fly around upside down with the gyro self stabilizing the entire time =D

Everything works as I stated in my above post, with ONLY the realization the "flip" toggle is actually "inverted flight" toggle.

I was flying this on my own compiled  v1.3.4.66 version yesterday. I have yet to test the newest commits. I'll compile, upload, and make sure the plane is still working correctly.

pascallanger, 2026-08-19:
Ok let me add CH8 Invert in the doc and in the MultiChan.txt (used by the MultiChannelsUpdater.lua script, if you don't use it give it a try)

wiilover22, 2026-08-19:
This will be a new git issue entirely, but I wanted to reply here first since it's relevant - MultiChannelsUpdater.lua fails to load on an TX16S MK3.

`?:0: attempt to call a nil value (field 'drawScreeenTitle')`

MultiConfig.lua also appears to be setup for B/W screens...and doesn't display properly either.

Would you like me to make a new issue for this?

pascallanger, 2026-08-19:
Yes go ahead and create an issue. But I don't have a TX16S MK3 to troubleshoot...

pascallanger, 2026-08-19:
Doc: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#sub_protocol-t28

wiilover22, 2026-08-19:
Wiki looks good. I'd personally add a note below:

CH6/CH9/CH13 (Stunt, RTH, RTH Set) are NOT required to be bound, or used to fly the plane. 

CH9 (RTH) *WILL* function without CH13 (RTH Set) bound/used. 

Edit: These activate at +100%. I keep those channels at -100% normally.

pascallanger, 2026-08-19:
Added.
-100% (disabled) +100% (enabled) is the default behavior for all feature channels so I only add a comment when they are working differently