pascallanger/DIY-Multiprotocol-TX-Module #656: Tx 12 kann meine Hott nicht binden

Gunhallo, 2022-01-20:
Alle neue Software auf gespielt auf der tx 12 aber es ist kein binden mit hott grauper Empfänger möglich---
Habe jetzt tage damit verbracht aber kein Erfolg zu bekommen , Neuen Sender tx erworben bin fast so weit die anlage in die tonne zu tretten.....

mha1, 2022-01-30:
Which firmware versions for MPM and OpenTX?

mha1, 2022-01-30:
... and which Graupner receiver and which RX firmware version?

HazzaHFPV, 2022-01-30:
Bevor du sie in die Tonne tritts, nehme ich sie lieber ;)
Ich besitze diverse HoTT Empfänger die ich mit mehreren 4in1 Modulen betreibe. Um welche Empfänger geht es denn?
Auf Anhieb fällt mir nur die Frage ein: Mal das Frequenz-Tuning durchgeführt?
Beste Grüße
Harry

Gunhallo, 2022-02-01:
Hey

Ja es geht um den GR12 neu erworben ,,, Intern und extern und Frequenz-Tuning von 129 bis -129 keine bindung ....weiss mir kein Rad mehr ....

Bootloder neusten und Multi neuster aufgespielt und nix machter ,,,,man kann auch nicht erkenne warum nicht ....

Soviel nachgelesen aber maches verstehe ich auch nicht Bekriflichkeitn .....

 

Danke für deine Antwort

 

gruss

gunther

 
 

Gesendet: Sonntag, 30. Januar 2022 um 18:24 Uhr

Gunhallo, 2022-02-01:
Hey es geht um einen neuen Gr 12 .....

 

grusss 

gunther

 

Gesendet: Sonntag, 30. Januar 2022 um 16:22 Uhr

closed 2022-07-07 by pascallanger