pascallanger/DIY-Multiprotocol-TX-Module #657: Clone TX with EU LBT Protocol

crasy666, 2022-01-25:
Hello to the Dev Team,

first of all thank you for your outstanding work, that is a great project, I am using Radiomaster TX16S as main main TX for my helis and would like to use it on my RC Excavators as well.

And for that topic I would like to raise my request;:

I need cloneTX as we use multi TX at the same time and try to handover the models between the different TX. Until now I am using DeviationTX with the same TransmitterID and a fixed receiverID for each model, it works quite well.

I tried the CloneTX with two TX16S in FrskyX2 D16 protocol, it is working good, and I think that could be an replacement for my DeviationTX radios.

But I have to use D16, which is normally "not allowed" in Europe, EU LBT should be used.

**Now the question:
Can I use also CloneTX on EU LBT? or can you implement that in a future version of the firmware?**

Thank you in advance for your help

crasy666

pascallanger, 2022-01-25:
CloneTX on EU LBT is already supported by Multi since a long time. Not sure which version of firmware you run on your radio but it should just work like D16.

crasy666, 2022-01-25:
Hi Pascal, 

btw, I am from Germany, but since other are not German I used English. You are right, I tried over my lunch break EU-LBT and it was working, maybe it was my problem, I thought only D16 is working, and I had some issues with the RX6R Firmware, but that is not "your topic"... 

So we can close this topic, EU LBT is working on clonedTX, as a proposal, maybe you can spend one sentence in the description of the Radio cloning, for such "experts like me"

THX
crasy666

closed 2022-01-25 by crasy666