pascallanger/DIY-Multiprotocol-TX-Module #660: Problem with V911S/E119 on version 1.3.3.7

Alexander-C, 2022-02-02:
WL Tech V911S Helicopter won't bind on version 1.3.3.7.
It works fine on 1.3.3.0

pascallanger, 2022-07-07:
I don't think there is an issue with the latest code but let me know.

Alexander-C, 2022-07-09:
Hi Pascal, The V911s helicopter won't bind on the new release either. Only
works on V1.3.3.0

pascallanger, 2022-07-09:
Ok I'll check the difference.

pascallanger, 2022-07-09:
By the way why have you opened the issue with V911S/E119? The V911S heli is using V911S/V911S and not V911S/E119. I know that the V911S/E119 works since a friend flew his this afternoon running the latest release.

Alexander-C, 2022-07-10:
Hi Pascal,

I was trying to bind the Heli using V911S/V911S without success. Strangely
I did finally manage to bind the helicopter. I tried to bind using the
V911S/E119 just in case..No Luck. I then changed it back and without
powering down the receiver I tried the V911S/V911S protocol and lo and
behold it bound almost instantly. In case this was a lucky fluke I rebound
the heli with the supplied transmitter, powered it down and then tried the
bind again using V911S/V911S on my TX16. Worked first time.
So now it works.

Regards,

Alex.

pascallanger, 2022-07-10:
It always worked, just that you have to respect a timing to do the bind this heli which needs to be done only once.

closed 2022-07-10 by pascallanger