pascallanger/DIY-Multiprotocol-TX-Module #662: Add PhilG NRF protocol

diego0815, 2022-02-04:
Would it be an option adding the PhilG protocol: https://mode-zero.uk/viewtopic.php?f=42&t=971 seems they have very simple receivers based on RF-Nano boards that can be built easily. 
Another plus is no reverse engineering, as the INO sources are available attached to the first post.

diego0815, 2022-06-25:
As it seems currently rfnano boards are hard to get hold off i guess we can close this for now...

closed 2022-06-25 by diego0815

stanekTM, 2024-03-05:
Hello,
it seems like a good idea. I myself integrated similar nRF communication with telemetry into MULTI to use my DIY RX, but only with a fixed channel. I haven't considered FHSS yet due to reliability.
Looking at the .ino code, it doesn't seem to use telemetry, that's important to me.
It would be a good alternative for the CABELL protocol.
Due to the busy schedule of lead developer @pascallanger, this alternative for DIY enthusiasts seems to have been forgotten. Maybe I'll try to write something in time. Thank you for the tip.

stanekTM, 2025-03-03:
Hi @diego0815 ,
If you like to play with Arduino Nano and NRF24L01+ transceiver, you can use the ["Stanek"](https://github.com/stanekTM/TX_FW_Multi_Stanek/blob/master/Multiprotocol/STANEK_nrf24l01.ino) protocol from my [Multiprotocol](https://github.com/stanekTM/TX_FW_Multi_Stanek) fork for now. I have been using this setup for years on my RC models. No need for FHSS yet, in my country channel 76 is safe. The custom FHSS code is in testing. You can also use the "Cabell" protocol, I used it myself. This is a benefit for the DIY community.