pascallanger/DIY-Multiprotocol-TX-Module #666: Spektrum Capacity counting between -128 and 127 mAh

elekSENDER, 2022-02-11:
Hello @pascallanger,

I dont know if this is the right place for my issue, but please let me know if i have to ask anywhere else.

I´m trying to use Spektrum telemetry via a Spektrum TM1000 on a Horus X10s with an irange4in1 multimodule and on a Radiomaster TX16s.
So far so good, i can see all connected sensors and modified the sensor settings with ratio to see useable values.

Now i have an issue with the battery capacity counting: it starts with 0 up to 127, the next value after 127 is -128 and it counts negative up to 0. 
This repeats, i can see only values from -128 to 127 in the Radios, so its useless for a mAh-capacity warning. The log of this sensor looks like a sawtooth.
The Sensor ID is 3402, "mAh", ratio: 1.0

This occurs on OpenTX and EdgeTX on both Radios.
Firmware: 
OTX: 2.3.14
ETX: 2.6.0
Irange4in1: 1.3.3.7
Radiomaster intern MPM: 1.3.3.7

Is this an issue in the multimodule-firmware or perhaps in OTX/ETX?

Thx in advance
Alexander

pascallanger, 2022-02-11:
Multi is sending the telemetry data tax to the radio firmware to be decided there. So opentx, edgetx and erskytx are doing the decoding. Therefore if there's an issue on the values, that's where you need to open an issue.
Pascal

closed 2022-02-11 by pascallanger

elekSENDER, 2022-02-11:
Added a screenshot of the log:

[image]

elekSENDER, 2022-02-11:
ups, was not refreshing the side before uploading the log.
thx a lot!