pascallanger/DIY-Multiprotocol-TX-Module #671: Support for FS Cev04?

Riffer, 2022-03-24:
Is it possible to connect and setup the extension module with OpenTX?

If yes, how?

diego0815, 2022-04-10:
Hmmm strange way of assigning the channels to the outputs...
It actually needs support by the TX: https://www.rcgroups.com/forums/showpost.php?p=44091981&postcount=227

I guess someone with a working flysky setup would have to capture the ibus.

Unfortunately I have neither any flysky gear nor the skills to implement it.

diego0815, 2022-06-25:
As it seems purely ibus related it might more be an issue to best be tackled by edgeTX/openTX maybe even a LUA script?

Riffer, 2022-06-27:
My best guess is, this at least needs a small kind of support by 4in1 for the setup. There is a stub already for fail-safe and I think this setup needs something similar.

diego0815, 2022-06-28:
Do you have the equipment? Then you could spy at ibus while binding the cev04. 
It seems ibus is a noninverted 115k2baud serial bus: https://www.arduino.cc/reference/en/libraries/ibusbm/