pascallanger/DIY-Multiprotocol-TX-Module #673: Add FX601

albatorsssx, 2022-03-25:
Hello,
Can you add the FX 601 model?
Here is a link to the product:
https://www.banggood.com/FX601-2_4Ghz-2CH-235mm-Wingspan-Delta-Wing-EPP-RC-Airplane-RTF-with-3-Axis-Gyro-System-p-1687052.html?rmmds=myorder&cur_warehouse=ES&ID=531270
And attached is a picture of the radio.
[image]

Thank you in advance.

pascallanger, 2022-07-18:
You can try the new FX 620 subprotocol using the latest test version.
Please report if it works.

albatorsssx, 2022-07-19:
Hello,

Thank you very much for the implementation. I put the latest test version but I don't see the new protocol.

Is there a problem ?

Thanks,

Jean Louis

albatorsssx, 2022-07-19:
Yes I have access but it's not for my module, I have a iRangeX IRX4 (mm-stm-serial-aetr). Sorry.

pascallanger, 2022-07-21:
Test builds are now available. Please try and report.

albatorsssx, 2022-07-21:
Yes He is present. Thanks. Unfortunately no binding. Should I configure the channels in a certain way?

pascallanger, 2022-07-21:
Have you tried other protocols like kf606 and such since looking at the pictures the board seems to be similar.

albatorsssx, 2022-08-05:
Sorry for the late response. I was on vacation. I tested but it doesn't work. No bind. It doesn't matter if it doesn't work, the device is just a toy. Thank you anyway for the time spent on this.

pascallanger, 2022-08-05:
Yeah without the hardware there is nothing I can do. And frankly, FX "toys" airplanes are quite bad since they have no proportionality on ailerons just throttle.

closed 2022-08-05 by pascallanger