pascallanger/DIY-Multiprotocol-TX-Module #674: RB R/C F1/F2

r-foerster, 2022-03-27:
I would like to use my surface multiprotocol radio to control the above model.
Although not sure if they all share the same radio system, F1/F2 has alot of brothers and sisters e.g.
JY66
MN-90 MN-99 MN-91 FJ-45
JJRC Q60 Q61 Q63
WPL B1 B16 B24 B36 C14 C24 C34

Transmitter site:
Chip: RF2518
Board name: LN1953T2

Receiver site (2 in 1 Receiver and ESC):
Chip: RF25155
Board name: LN1953R2-T

Link to product page: https://www.aliexpress.com/item/1005001565120968.html

Do you have any protocol suggestions based on transmitter and board information?

Thats alot in advance.

diego0815, 2022-04-02:
The closest I could find is this here: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/453

r-foerster, 2022-04-04:
Thanks a lot! Will try and report if it binds.

closed 2022-07-07 by pascallanger