pascallanger/DIY-Multiprotocol-TX-Module #675: TX16s + Hobbyzone Aeroscout

RealDoctorMix, 2022-04-02:
I'm a beginner 🍼! I recently purchased a Radiomaster TX16s and a BNF Hobbyzone AeroScout, which has a Spektrum AR631 Receiver. I managed to bind it however I'm not getting more than a few degrees of elevator travel. I tried setting up flight modes on CH5 but with little success. I also tried to set up panic mode on CH6, with no success either.

I'm a bit frustrated having lost 2 weekends trying to resolve this, instead of flying on the field... to the disappointment of my 9 Y.O. son :( 
Can anyone help, please?

pascallanger, 2022-04-02:
Check the doc on how to increase the outputs range (table).
Note that a few deg of elevator might be enough unless it's an aerobatic/3D airplane that's usually more than enough.
Finally, you should ask help on rcgroups since someone else might have the same plane as you: https://www.rcgroups.com/forums/showthread.php?2165676-DIY-Multiprotocol-TX-Module/page1566

closed 2022-04-02 by pascallanger

RealDoctorMix, 2022-04-02:
Thank you for your quick answer [pascallanger] !  Will check that link now.

RealDoctorMix, 2022-04-02:
> Check the doc on how to increase the outputs range (table). Note that a few deg of elevator might be enough unless it's an aerobatic/3D airplane that's usually more than enough. Finally, you should [...]

Unfortunately that thread you linked is not helping much pascallanger. Can you please reopen this ticket? BTW, I'm completely out of my water here, apologies for being such a nwebie!  Thank you

kilrah, 2022-04-02:
What he was saying is "ask your question on that thread", not "your answer's there already".

RealDoctorMix, 2022-04-02:
Thank you so much ❤️ So that’s how little I know about this 🤣 🙏

topybear, 2022-04-02:
Is it possible that the plane is in beginner mode? 
https://forum.flitetest.com/index.php?threads/opentx-and-spektrum-rx.64858/