pascallanger/DIY-Multiprotocol-TX-Module #676: DSM Programming Spektrum Avian

chucky-whiskers, 2022-04-04:
Someone recently asked about DSM Forward Programming a Spektrum Avian ESC for "reverse throttle" functions etc... on a TX16S. This is a plausibly worthwhile update if it is not already working on the current version. I pointed this person to this lua script. It is worth looking into.

pascallanger, 2022-07-07:
Not sure what you are requesting... This someone should come here and explain the issue.

pitts-mo, 2022-08-06:
I recently stumbled onto this while attempting to add a head tracking pan/tilt system for a friend insists on the Avian feature of reversing via an additional AUX switch :-/.

An Avian ESC utilized with a Spektrum receiver allows for enhanced ESC configuration via a telemetry related extension from a Spektrum radio.

observations/understanding:
A Spektrum radio running "Avian Prog" appears to display a menu that can be navigated via stick commands. The result feels very similar to OSD control via MSP DisplayPort telemetry. It is my current belief that the Spektrum RX is generating this interface OR the Spektrum RX must be passing all the RX channels into the ESC pin in order for the AVIAN ESC to process the stick commands and generate the menu to send back over telemetry.

Spektrum Quick Hit Tech Tip - How to setup Thrust Reversing on Avian ESCs and BNF Aircraft
https://www.youtube.com/watch?v=_ZHWzBWe1e0

chucky-whiskers, 2022-08-06:
This problem is still not solved. I want a lua script capable of doing this programming on a RadioMaster TX16S (or any open-source, multi-protocol radio). Maybe this isn't the right place to suggest this, but I think it is within the realms of possible.

pitts-mo, 2022-08-07:
@pascallanger can this issue be re-opened, Should I attempt to enhance my understanding and start a new issue to build a LUA for this OR can you provide a better suggestion?
Thank you, -p

chucky-whiskers, 2022-08-07:
From what I have seen, the DSM protocol in both OpenTX and EdgeTX for the TX16S do not operate stick telemetry correctly. For instance when using stick inputs to access receiver programming functions the same way you would using a DSM transmitter, the receiver/onboard equipment does not respond. Spektrum stick "command functions" are somehow issuing a different signal than stick "inputs" used for controlling the craft, or the open-source software is sending different types of signals or something. The serial code sent to the receiver may have different priority levels per channel or something. Anyway, the TX16S will not do these setup functions on the Spektrum gear with the current programming. If that means updating the firmware for the 4-in-1 module, that could be a fix. A lua script is another possibility to address the issue. I know the problem and the function of the fixes, but I do not write code much. I'd feel more confident pushing this to someone with more experience.

chucky-whiskers, 2022-08-07:
Actually, it is a telemetry issue that probably requires a lua script either way.

pascallanger, 2022-08-08:
Without the hardware there is little I can do.
Can you ship a RX/model?

chucky-whiskers, 2022-08-08:
I cannot at this point in time.

pitts-mo, 2022-08-09:
I am checking with my friend and will try to talk him into sending gear.
-p

The cheapest Avian ESC and smart compatible receiver appears to be:
Avian 15 amp ESC: https://www.spektrumrc.com/Products/Default.aspx?ProdID=SPMXAE1015
AR410: https://spektrumrc.com/Products/Default.aspx?ProdId=SPMAR410#prod_section_support

pascallanger, 2022-08-09:
Ship or a donation to purchase the hardware. I think I have a receiver that should work but I'm in vacation and cannot check. I'll check how much such esc cost in France.

pitts-mo, 2022-09-04:
Hello @pascallanger,
Have you returned from your vacation? Do have time to evaluate the hardware you would need to attempt a build of lua replacement for Spektrum's "Avian Prog"?

The friend I am inquiring for has indicated he is willing to ship/pay for an ESC. 

Although, I am just now realizing the Spektrum smart RX and a Spektrum TX may also be needed in order to see what their "Avian Prog" does to work through RX and configure the Avian ESC.

-p

frankiearzu, 2023-01-20:
Im starting to understand the internals of how the Telemetry based screens works (TextGen). 
Have been able to do some experimentations with the Blade Heli AR636 to help on the configuration.

How TextGen works:
You enter the menu using a comvination of stick and switch movements. 
Each menu line is respresented by a telemetry sensor. So if you have a menu with 5 lines, it will come from 5 sensors. 
Another telemetry sensor has the # of the active menu option so that it can be highlighted. 
The menu line sensor holds the the current value/option (a message ID/number). That ID/Number needs to be translated to English.

Once i get my hands on an Avian Receiver, i think i could get the code done fairly quickly.

Frankie

BuddhaSource, 2023-04-06:
@pascallanger hope you found the hardware needed to test it.
I would love to see ESC reversal capabilities.

frankiearzu, 2023-04-06:
@BuddhaSource   The AVIAN ESCs uses TextGen to configure via the TX.
The problem is that the current EdgeTx firmware was not processing this TextGen messages correctly, but i did the fix. it will be available in Edge 2.9.0.     
For the ones who cannot wait, i created some firmware build who has the official EdgeTX 2.8.0 + some Spektrum telemetry fixes (TextGen, GPS sensor, ESC data) here:   https://github.com/frankiearzu/DSMTools
Has firmware for RadioMaster TX16, Zorro.   What Radio are you using.
After the firmware is installed. use the new Telemetry scripts to access TextGen. (DSM_Smart_Tel.lua)

frankiearzu, 2023-04-06:
Here is a video configuring the AVIAN ESC from a RadioMaster TX16
https://www.youtube.com/watch?v=yrAqV4b9Li4

BuddhaSource, 2023-04-06:
Excellent, thank you @frankiearzu 
I have both Tx16s and Zorro both. I will try this on TX16.

frankiearzu, 2023-05-02:
UPDATE: TextGen Support included in EdgeTX 2.8.3.  No need for custom firmware any more.

BuddhaSource, 2023-06-02:
I confirmed it worked. Thank you for supporting it.

frankiearzu, 2023-10-30:
@pascallanger this can be closed.   the TextGen is part of the DSM_Smart_RX.lua

closed 2023-10-30 by pascallanger