pascallanger/DIY-Multiprotocol-TX-Module #677: setting FORCE_GLOBAL_ID doenst break bind (FrSkyX2)

jasc76, 2022-04-05:
I set FORCE_GLOBAL_ID and compiled 1.3.3.11.
After setting this option an existing bind was not broken, is this correct?

pascallanger, 2022-04-05:
That's more than strange.
What happens when you use the LUA app which is made for changing the ID called MultiConfig: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts#multiconfig

jasc76, 2022-04-05:
I also tried with 1.3.3.7 (standard release from site) and MultiConfig says the same "No Config Telemetry"

jasc76, 2022-04-08:
@pascallanger  
Hi
Any hint on this?

pascallanger, 2022-04-08:
I'm thinking you are not using the latest lua script...
With the old one, just create a blank model and select multi on the module you want to configure.

jasc76, 2022-04-08:
I used the scripts provided in the 1.3.3.7 Release on github. I even coped them again to be sure..

Ok, then I will copy the luas directly from master branch and try again and report back

pascallanger, 2022-04-08:
The script as been modified since 1.3.3.7. This is why I pointed you to the page in a message above.

jasc76, 2022-04-08:
Ah, ok
but then the script versions from the releases page are not up to date.
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/releases

pascallanger, 2022-04-08:
No since this is a release. They will be in the next one...

jasc76, 2022-04-11:
Ok, not using the release version it works

but the ID is not changed although I set it in the build config.h and of course delete the comment

pascallanger, 2022-04-11:
You must be doing something wrong. I'm using FORCE_GLOBAL_ID on all my modules.
Copy and paste the line you've changed here to see.

jasc76, 2022-04-12:
#define FORCE_GLOBAL_ID	0xAFFE1234

jasc76, 2022-04-12:
> Did you change it in _config.h or _myconfig.h?

In config.h

Should it be named _config.h ?

pascallanger, 2022-04-12:
The config file is called _config.h

jasc76, 2022-04-13:
Ok, I tried again. fresh clone and this time I did it in the _MyConfig.h 
but no luck .... I effing dont know what I'm doing wrong. 
for compile do I need to open the myconfig or the config ?

I guess I'm too dumb for this :-(

pascallanger, 2022-04-13:
If you do it in _MyConfig.h then you have to enable myconfig in _config.h

closed 2022-04-19 by pascallanger

jasc76, 2022-04-21:
Ok, with myconfig it works ! 
thnks !