pascallanger/DIY-Multiprotocol-TX-Module #678: Endpoints and channel order behave strange.

Kenaiwolf, 2022-04-13:
Hi,
iRangeX IRX4+ TX
Multiplex Cockpit SX
Using PPM mode
protocol Futaba/SFHSS
Fine tuning done

Bind with Eachine E180 heli. Problem is mainly throttle. TX is -100 +100 version. Moving throttle from -100 to -28 nothing moves. Once it hits -27 I get full throttle response. At least it looks like that by sound and looks. (no blades mounted, heli on the table situation)
Moving throttle from -27 up is no change at a glance.
I'm sure channels are mapped correctly (triple checked) Roll, Pitch axis, Collective pitch, Tail, Gyro, they all work properly Endpoints are probably not correct and center position may be wrong too.
I suspect endpoint seting of transmitter is to blame here. But I can't figure out what are endpoints of Cockpit SX in microseconds . Right now they are set as /Multiplex MC2020 (1250<->1950 microseconds)

24 hours later.
Did some more testing and find some strange things, at least for me.
Managed to test endpoints by experimentation and finish on strange
#if defined(TX_CUSTOM)
#define PPM_MAX_100 3000 // 100%
#define PPM_MIN_100 0 // 100%
#endif
In this setting aileron, elevator have correct range of movement and midstick is OK too. End of stick movement is in sync with end of servo movement. However throttle stopped working completely. Reversed throttle tested, no change noticed.
Even more strange is swapping channel order seems to not working.  I was treeing to set correct channel order to get throttle function, whole day. In some point i found out some channel orders hes same results as other.  Finally I swapped what was working CH1 Ai, CH2 El, CH3 rudder from 0x12345678 to 0x31245678 and i get:
EL on CH3
AI on CH2
CH1 seemed to not doing anything. (can't test rudder without throttle - tail motor heli)
So i get 0x23?45678
If I understated how it works, I should have gotten Ai on CH3, El on CH1 and rudder on CH2 

I belie there must be some overlap in channels reading, because of wrong endpoints setting. 2 channels on outputs controls by 1 PMM CH on impute or something like that. The only second explanation will be channel order settings are buggy. 
I'm starting to feel desperate.

pascallanger, 2022-04-13:
I don't know the E180 heli but you might have like on most helis of that size a channel for 6G/3D and may be other stuff. It looks to me that your throtle issue is linked to one of these additional channels are not set correctly.
Setting MAX to 3000 and MIN to 0 is decreasing a lot the end value which will be used. So that should explain why throttle doesn't work anymore since you can't get to -100% and init the ESC.

Do you have a Multiplex RX? If yes what you can do is to bind this RX with the Cockpit SX normally, connect servos on the outputs, make sure that the endpoints of the radio are set to normal, and draw the mid, max and min positions for Ail, Thr, Ele, Rud and also on which order they are on the RX.  Then bind the same RX, same servos to multi using the MLink protocol. Now, you need to change PPM_MAX and MIN to get the exact same ouputs (mid, max, min), also check the channel order. From there you should have the channel order of the radio, and the MIN, MAX endpoints pretty closed. Change to the SFHSS protocol, reassign the channels accordingly and give it a go.

Kenaiwolf, 2022-04-13:
Thank you for reply. 
E180  expects channels:
1: Aileron
2: Elevator
3: Throttle
4: Rudder
5:  switching 3D and 6G modes 
6:  Collective pitch 
7: Aux 1 don't care
8: Aux 2 don't care

In mean time I manged to get throttle on stick CH4 on my TX. So i redefine endpoints using throttle and digital tachometer (RPM meter). now i am on 
#if defined(TX_CUSTOM)
	#define PPM_MAX_100	2062	//	100%
	#define PPM_MIN_100	938	//	100%
#endif
 -94 start spinning rotor +100 is max RPM
As long as i know E180 should not have any throttle range internal configuration. So hopefully I'm, spot on with endpoints.

My TX is AERT so i set it like that. It have fix CH order in Heli mode. It is like this:
CH1 Aileron
CH2 Elevator
CH3 Rudder
CH4 Colective PITCH
CH5 Throttle
CH6 Gyro
CH7 AUX1
CH8 N/A it is 7 channel TX

channel order 0x**0000000** gives me CH1=Ai, CH2=El, CH3=Ru, **CH4=Throttle**  CH5=Gyro CH6=Pitch CH7=free CH8=N/A
I change channel order to 0x00054000 it worked i got CH1=Ai, CH2=El, CH3=Ru, **CH4=Gyro**  CH5=Throttle CH6=Pitch  CH7=free CH8=N/A
Finally i wanted to swap CH4 and CH6 so I set 0x00056400 and all hell break loose.
I got CH1=Ai, CH2=El, CH3=Ru, **CH4=Gyro**  **CH5=Pitch**  **CH6=Throttle**  CH7=free CH8=N/A
What am I doing wrong ?

Yes, I have Multiplex RX in my Extra 330 but it is 35Mhz RX as is my TX. It work perfectly for years now. But it can't be connected to computer or anything else  except 35Hz TX or by cable to PPM signal directly. :-( I know my equipment is little outdated, but so am I. And i hate to trow away what is not broken.

Kenaiwolf, 2022-04-13:
Ooh, I found my mistake it should be 00064500. Eureka it  is working all now.

closed 2022-04-13 by Kenaiwolf