pascallanger/DIY-Multiprotocol-TX-Module #683: Propel Remote control uses XN297lbw chip

bangdc90, 2022-04-26:
Dear all,
I have some propel TX, which use XN297lbw. I tried to use  these TXs to control esp32+nrf24l01 but not success.
My invesigation result:
- Propel TX address: {0x12, 0xd6, 0x74, 0xbd, 0x71}
- Propel RX address: { 0x15, 0x59, 0x23, 0xc6, 0x29 }
- Bind Channel: 0x0B
- Payload length: 18

Could you help me check the PulseView data:
[bind_sequence.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/8564737/bind_sequence.zip)
Source code (impl on Platform IO base on source code in multiprotol) to know where was I mitake?
[rx_nrf24_esp32.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/8564745/rx_nrf24_esp32.zip)
I am studying in Promisc.cpp to catch payload packet from Propel TX.

Thank you very much for any help.

pascallanger, 2022-04-27:
I'm not sure what I should look at. Are you receiving any packets from the TX?

bangdc90, 2022-04-27:
> I'm not sure what I should look at. Are you receiving any packets from the TX?

Dear @pascallanger ,
Thank you very much for your attention.
Today I use XN279 dump and get the following information:
Detecting XN297 packets.
XN297 dump, address length=5, bitrate=250K
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11

Packet detected: bitrate=Using NRF
250K C=11 S=Y A= 12 D6 74 BD 71 P(16)= 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
--------------------------------
Identifying all RF channels in use.
Trying RF channel: 0,1,2,3,4,5,6,7,8,9,10,11
RX on channel: 11, Time:     0us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 14995us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3266us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11718us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3263us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11736us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3264us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11712us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3281us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11713us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3267us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11722us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3266us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11730us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3264us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11715us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3278us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11713us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3268us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time: 11717us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
RX on channel: 11, Time:  3270us P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80
Trying RF channel: ,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,61,62,63,64,65,66,67,68,69,70,71,72,73,74,75,76,77,78,79,80,81,82,83,84,85,86,87,88,89,90,91,92,93,94,95,96,97,98,99,100,101,102,103,104,105,106,107,108,109,110,111,112,113,114,115,116,117,118,119,120,121,122,123,124,125,126,127

1 RF channels identified: 11
--------------------------------
Identifying RF channels order.
Time between CH:11 and CH:0

Channel order:
11:     0us

--------------------------------
Identifying Sticks and features.
P: 10 39 9F 00 00 80 08 10 00 00 00 00 00 00 00 80

With these information, I configured for my rx board :
    const uint8_t tx_address[] = { 0x15, 0x59, 0x23, 0xc6, 0x29 };
    const uint8_t bind_address[] = {0x12, 0xd6, 0x74, 0xbd, 0x71};

    NRF24L01_Initialize();
    NRF24L01_SetTxRxMode(TX_EN);
    NRF24L01_SetTxRxMode(RX_EN);
    XN297_SetTXAddr(tx_address, BAYANG_ADDRESS_LENGTH);
    XN297_SetRXAddr(bind_address, BAYANG_ADDRESS_LENGTH);
    NRF24L01_FlushTx();
    NRF24L01_FlushRx();
    NRF24L01_WriteReg(NRF24L01_01_EN_AA, 0x00);      // No Auto Acknowldgement on all data pipes
    NRF24L01_WriteReg(NRF24L01_02_EN_RXADDR, 0x01);
    NRF24L01_WriteReg(NRF24L01_11_RX_PW_P0, BAYANG_PACKET_SIZE); // rx pipe 0

    NRF24L01_WriteReg(NRF24L01_03_SETUP_AW, 0x03);   // address size
    NRF24L01_WriteReg(NRF24L01_04_SETUP_RETR, 0x00); // no retransmits
    NRF24L01_SetBitrate(NRF24L01_BR_250K);             // 1Mbps
    NRF24L01_SetPower(RF_POWER);
    NRF24L01_Activate(0x73);                         // Activate feature register
    NRF24L01_WriteReg(NRF24L01_1C_DYNPD, 0x00);      // Disable dynamic payload length on all pipes
    NRF24L01_WriteReg(NRF24L01_1D_FEATURE, 0x01);
    NRF24L01_Activate(0x73);
    delay(150);

    XN297_Configure(_BV(NRF24L01_00_EN_CRC) | _BV(NRF24L01_00_CRCO) | _BV(NRF24L01_00_PWR_UP) | _BV(NRF24L01_00_PRIM_RX));
    NRF24L01_WriteReg(NRF24L01_07_STATUS, 0x70);
    NRF24L01_FlushRx();

Now I get the payload from TX the same as XN279 dump.
But because I don't have a drone, there is only TX, so I can not understand the meaning of received package. When I trie to move sticks, the received package did not change.
Do you have any suggestion for me to can continue reverse engineering?

bangdc90, 2022-04-27:
As I guess, maybe my RX need to respond something to TX, then the TX will can exit bind mode.
Because when I test with another TX+drone of PROPEL (this one is different with the TX I am reverse engineering), I saw that if I turn on the drone, then turn on TX and push throttle stick up and down, the TX will have beep sound, the led stop blink. Without drone turns on, TX has nothing change.

closed 2022-07-07 by pascallanger