pascallanger/DIY-Multiprotocol-TX-Module #688: Extended limits limited to 150. Brain FBL no longer configurable. Radiomaster TX16S

ReneHitz, 2022-05-25:
In the current firmware for the radiomaster TX16S, the servo travel can be set to a maximum of 150%. This is not sufficient to control a Brain FBL controller. The internal value in the brain then only just goes up to 86% instead of 100%. OpenTX 2.3.15 with internal multi module 1.3.3.7 are installed. The same problem also exists in Edge TX 2.7.1.

pascallanger, 2022-05-25:
I'm guessing you have a DSM receiver correct?

ReneHitz, 2022-05-25:
It's a Radiomaster R81. Settings in the transmitter: "Multi FrSky D D8". The receiver connects to the FBL Brain via S-Bus.

pascallanger, 2022-05-26:
You have to set extended limits for the outputs to really go to -125%..0..+125%. Have you checked your radio outputs?
The frsky protocol is already going way beyond what a standard setup do. It could also be that the RX is not able to go that far.

ReneHitz, 2022-05-26:
Yes, the xtended limits are set to - 150%...0..+150% The maximum that can be set. How can I determine that the receiver cannot receive this range? I found out from a friend that he also had the problem with a multiplex transmitter. I haven't heard anything helpful from MSH Electronics so far.

closed 2022-06-16 by pascallanger