pascallanger/DIY-Multiprotocol-TX-Module #691: XK K127 Protocol - Pascallanger, I will send you a K127 if you need one

okevin54, 2022-05-30:
Tried every protocol and scoured the forums to find the protocol for the K127. Pascallanger, give me an address and I will mail you my XK K127 if that will help with the protocol. My K127 XK transmitter will run my C186, and all the other versions of it (C400, BO-105, many companies putting their label on it. So if you could get the K127, it would also get all the newer C186 helis.
Please let me know if this would help. 
Thanks
Kevin

pascallanger, 2022-05-31:
Hi Kevin,
I would need the TX and RX to look at the protocol. If you ship the TX, do you know if they sell one of the heli as a BNF?
- Pascal

pascallanger, 2022-05-31:
Hi Kevin,
I haven't been able to find the heli in BNF... I've found it at 77€ here: https://fr.aliexpress.com/item/1005004234784325.html?spm=a2g0o.ppclist.product.10.47fd8mGS8mGSvm&pdp_npi=2%40dis%21EUR%21%E2%82%AC%2095%2C88%21%E2%82%AC%2076%2C70%21%21%21%21%21%400b0a24a616539784762676633e6413%2112000028632933544%21btf&_t=pvid%3Abd0d8f91-2c68-48f2-9c8c-a18e76e70700&afTraceInfo=1005004234784325__pc__pcBridgePPC__xxxxxx__1653978476&gatewayAdapt=glo2fra
What about you do a donation of half the heli price (which might be the shipping cost depending of where you leave). If that's ok then I can place the order right away and I'll have everything to work on in 2-3 weeks.
- Pascal

okevin54, 2022-05-31:
Pascal, what country are you in? I don't mind sending you the heli and you can keep it. I really want the protocol for the C186 (BO-105) it's a scale heli that is realistic looking, so I'm not attached to the K127, it just has the protocol I need for the C186.
Let me know where you are and I can see if I can ship it for less than half the cost of a new one.
Thanks,
Kevin

pascallanger, 2022-05-31:
I'm in France. Contact me on rcgroups (user: hpnuts) for more information.

okevin54, 2022-05-31:
ok, going to check shipping

okevin54, 2022-05-31:
Pascal, I get a 500 error when I search your name in RCgroups. I think sending half is the way to go. Do you have PayPal?

okevin54, 2022-05-31:
yes it is Kevin959, I get this error from paypal when I put your email in to send money
We can't find pascal_langer @ yahoo . fr. Try entering their mobile number or email.
Not sure if that's because I'm in the US or not. I was able to send you 40 through your donate button on Github.
Transaction ID: 21048283398103806
Please let me know if it doesn't come through.
Thanks Again
Kevin

pascallanger, 2022-05-31:
I've received it and will place the order now.

okevin54, 2022-05-31:
I really appreciate it

pascallanger, 2022-06-22:
The bind and normal packets structure looks similar to the E129 but with 3 extra bytes.

pascallanger, 2022-06-26:
I can fly the C186 ;-)
I need to cleanup the code and do some more testing like changing the ID before pushing it to github.

diego0815, 2022-06-26:
Wow! Great job Pascal!

pascallanger, 2022-06-27:
The new builds having this protocol are available here: https://downloads.multi-module.org/latest-test/
Documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#e129---83
Please test and report.

closed 2022-06-30 by pascallanger