pascallanger/DIY-Multiprotocol-TX-Module #694: JP4IN1 not detected by Windows 10

port513, 2022-06-08:
Bought my unit some years ago and now I tried to update to latest. I have the latest OpenTX on my FrSky.
Tried almost everything, 5 different computers with Windows, 2 with Linux and the unit still doesn't get detected.
Are there enything I miss in my flashing procedure?

Not really an issue buying a new one but if it works I would like to use it...

pascallanger, 2022-06-08:
Have you tried to update it from the radio itself?

port513, 2022-06-08:
> Have you tried to update it from the radio itself?

Yes, I have to old bootloader for that to work

port513, 2022-06-09:
> Which version of the jp4in1 do you have? Pictures of both sides of the PCB would help.

Then I would have to take it down, it's a Jumper JP4IN1
Bought a new unit just in case this is damaged.

LED says it's the old bootloader, hope the new one is better.

Could not see the firmware version either from my radio even when a model was on external TX.

closed 2022-06-16 by pascallanger