pascallanger/DIY-Multiprotocol-TX-Module #701: R81 radiomaster sync issue

scooper1, 2022-07-04:
I fly with another person at same time
With both are using radiomaster tx16 with the lastest stable version of edgetx

We are using the  radiomaster r81 receiver in d8 mode

At random times on startup we seam to get a miss sync were the radio reports telemetry lost  and the packet led on r81 flash at a 10th of normal speed

At this point we removed the Rx battery and reconnect and every thing is ok 

The real issue that when in this bad signal on one radio system. The other radio set starts to report low signal levels or telemetry lost. 

This issue continues untill the Rx on the first is powered down from  bad sync issue 

This happens on about 1 in every 100 receiver power ups

Just to confirm above one radio is affect a complete different active radio link

This problem can occur both ways  either tx1 affects tx2 or   tx2 affecting tx1

It could be a problem In the r81 Rx telemetry tx path

But I seen this using other makes of the Rx in d8 mode.  It could be that it's a bad common chip design

Has anyone else got ideas on this?

My problem is that I using the r81 as small sbus Rx with telemetry to enable model finding on a Mico heli 

And don't know of any sbus Rx with telemetry that are small to be able to replace this setup. Again ideas welcome

pascallanger, 2022-07-04:
That looks like a bug in the RX starting to send packets all over the RF channels and jamming things around it.
What you could do when it happens is to run one of the TX16 in RF scanner and observe what the impact is before and after disconnecting the RX.
Pascal

pascallanger, 2022-07-04:
Also, have you done the frequency tuning of the Multi module against the RX? and this on both radios?

scooper1, 2022-07-04:
fully tuned and rebound afterwards

closed 2023-10-30 by pascallanger