pascallanger/DIY-Multiprotocol-TX-Module #704: FEI XIONG FX620 Protocol

gke, 2022-07-08:
I managed to get the debug version running and these are two logs I managed to drop from my Fei Xiong SU35 FX620 . I am not having a lot of luck with matching things up with the FX816_nrf24l01.ino fields. I also get what seems to be a few different different hopping sequences. The packet trace shows the throttle and aileron data moving. Hints on how to proceed appreciated or even better someone with the skills to help out.

Thanks
Greg

[FX620 1.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9075211/FX620.1.zip)
[FX620 2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9075212/FX620.2.zip)
[FX620 3.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9075213/FX620.3.zip)
.

pascallanger, 2022-07-08:
I need 3 more dumps before trying to understand something:
 - bind: leave the TX in bind mode and do an auto search
 - normal: move only throttle really slowly from low to high to low. Stop and save the log calling it THR.log
 - normal: move only aileron really slowly from center to left to center to right to center. Stop and save the log calling it AIL.log
Pascal

pascallanger, 2022-07-08:
**Bind**
Unknown until I have the bind dump requested above

**Normal**
Bitrate: 1M
Scrambled: Y
Address: FE F8 34 A9 32
Payload: 5 bytes
0|1|2|3|4
-|-|-|-|-
THR/AIL|00|00|AB|Check

THR/AIL: unknown until I have the 2 normal dumps requested above
Check= sum of payload[0..4]+0x9E

Different hopping sequences: relaunch the dump at different distances/orientations between the multi module and the FX620 TX. Right now you are catching ghost frequencies...
Channel 8 for example does not seem to be valid. So if you have dumps not showing it you might be at the right spot.

gke, 2022-07-09:
Thanks Pascal. I think I have stable hopping at 24,34,44,54 and binding at 18? The throttle is definitely on Ch0 and the Aileron is Ch4. Changes to Ch4 are not logged unless Ch0 is changing. The F35 is differential thrust so there is mixer coupling between Throttle and Aileron. Not sure what Ch3 is but it does not change so perhaps some sort of switching. Log files attached along with my first cut on the code.
[FX620.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9075809/FX620.zip)
[FX620_nrf24l01.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9075813/FX620_nrf24l01.zip)
?

gke, 2022-07-09:
Please ignore last mail. I will spend a little more time checking the log files to see what is happening to the Aileron values as I just re-read your post and saw that I am seeing the CRC on Ch4 changing not an Aileron value.

pascallanger, 2022-07-09:
Sorry but I don't understand what you say... What do you mean by CH0 and CH4? Are you talking about the payload bytes 0 and 4. If that's the case then B0 changes on your dumps when you move throttle and B4 is a check (calculation given above) to see if the content received is correct. What you call Ch3 is the byte 3 I suppose which I agree is constant on all your dumps = 0xAB.

What I think is that the address length is wrong. Try to redo the dumps with an address length of 4 and even 3. An address length of 3 would give a bind address of AA BB CC which looks not bad. I think then you'll see the aileron value appearing on the first or second byte.

pascallanger, 2022-07-09:
More I look at it more I think that the address length is 3.
Bind address: AA BB CC
I anticipate that the bind payload will be 34 A9 32 18 00 00 27 -> Check=B6=27 is the sum of bytes 0..5 which is usual
Normal address: 34 A9 32
I'm not going to guess the payload, but I'll say that B6=Check will be the sum of bytes 0..5 too.

pascallanger, 2022-07-09:
For the implementation, I'll add a subprotocol to the current FX(816) protocol => Protocol FX, subprotocol 0=816 and subprotocol 1=620.
I'll take care of this as soon as I have the correct dumps to work on.

pascallanger, 2022-07-09:
Please read all my posts from the beginning since I'm updating them as I go. Like this you have a full picture.

gke, 2022-07-09:
Thanks Pascal. I will get the dumps done today but some flying first.

pascallanger, 2022-07-10:
I've written the protocol with your ID for now.
Are you able to test quickly? If yes I can publish it, you do the test and report, I do some modifications and you test again. We will need like this a couple of iterations to understand if Aileron could be a proportional value or not and to hopefully have a couple of working IDs instead of just yours. As you understand we need to be online together to try and test without loosing too much time.

gke, 2022-07-10:
Good Morning Pascal.

I agree we should make B1 (Aileron) linear and B2 (Throttle) max at 63, With the Tx as delivered the bang/bang aileron control makes it strange to fly - very jerky.

It is interesting that there seems to be no protocol consistency even within one company for its different products, Just having a single protocol and only change an ID field only or just use the Tx module ID for binding the different aircraft - too logical ;). BTW John Julian JJ604 is a good friend.

gke, 2022-07-10:
Yes I can test - where do I grab the code from?

gke, 2022-07-10:
I think it is mid day for you now. Tell what time is best for you to do the testing.

gke, 2022-07-11:
I am back home.

pascallanger, 2022-07-11:
The code is available on Master.

gke, 2022-07-11:
Thanks Pascal

Binding not working yet so I used the original Tx to bind. Throttle not working initially but is now after a change. I am trying not to make changes that stop the 816 variant working  but I can borrow a P38 that uses 816 to check. I tried the ailerons with proportional bu the Rx is using bang/bang.  I will have more of a play tomorrow and get back to you. Good progress.

pascallanger, 2022-07-11:
For the bind, I don't really see what could be the issue. The only thing is the packet period of bind is 4500µs and the packet period of normal is 3250µs. But usually they don't check that, the RX receives one bind packet and it is good with it. May be it's not the case here, we can try to update the timing while binding.
What change have you made to the throttle to make it work? I'm surprised that you need something different...

gke, 2022-07-12:
Hi Pascal

I am recovering from COVID and have a little brain fog so I cannot see what the binding issue is either. The aircraft binds immediately with the original Tx. Once bound it responds fine with the MM code. It will be simple I am sure and probably is no processing of any ack from the aircraft just a simple binding packet count I saw the longer period for the binding and had extended that to 4500uS.

You will barely recognise the current code as I have a habit of rewriting to get an understanding of what is happening :(. 
[FX_nrf24l01.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9089372/FX_nrf24l01.zip)

pascallanger, 2022-07-12:
Can you try the latest code on Master?

gke, 2022-07-12:
Still not binding. Everything else OK if I bind with original Tx. I am going to re-run the bind dump tomorrow.

pascallanger, 2022-07-12:
If it's like other small remotes when you turn it on it will send bind packets. But it can also send different bind packets when you move the throttle up and down to finish the bind process. So make sure to run a scan at each step of the process: turn on, throttle up, throttle down.

pascallanger, 2022-07-12:
"Setting it to 1mB results in no packets being logged." This is normal, when you force the bitrate you are exiting the "Auto search" feature. You therefore need to select manually the RF channel to find out what you are looking for.

Let's see what the bind dump will give. I can't find what's wrong with the bind since I'm doing exactly what is in your dumps...

gke, 2022-07-13:
I have two MP modules and have now set one up for dump monitoring and the other for testing the code. The dumps from MP look the same as those for the original Tx. It still does not bind but otherwise remains OK. I have tested a P38 which uses 816 and it binds fine. The rebinds in the MP menu trigger a rebind as well. I have not checked aileron and throttle fully yet for 816 bit it should remain OK.

I am going to take a break to get some other things done and recover fully from COVID but now I have the test it will be easier to check things out.

Cheers
Greg

pascallanger, 2022-07-13:
While the TX is in bind mode, have you scanned all the frequencies using a bitrate of 1Mb?

gke, 2022-07-13:
No I have not it seems we have just been capturing the final packets in the bind sequence.  Carefully timing when I start the bind shows a start frequency as low as 8 in one log with  x78 in the last byte of the packet but this could still be ghosting.  The other log also shows the x78 value being transmitted in the expected bind channel.

[20220714 Bind Ghosting.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9106558/20220714.Bind.Ghosting.zip)

gke, 2022-07-13:
OK this seems to bind happily. Still not sure how an ID is used  but is may be the LSBs of the last address byte. 
[FX_nrf24l01.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9106608/FX_nrf24l01.zip)

gke, 2022-07-14:
I will get the full dump after flying this morning.  The bind_counter counts down so I am emitting 0x78 first then 0x00. One of the previous dumps suggests that multiple frequencies are being used but the other shows only 20. I suspect ghosting but the dump today at 1mB should show us.

gke, 2022-07-15:
OK I am using the NRF dump and forcing the address length to 3. The frequency channel is stuck on the default of 40. What do I need to do to get the dump? Does it involve me editing the dump code and if so what do I need to do to compile the debug version. Sorry for being a little slow ;). 

Flying today was OK with 620 and 816.

gke, 2022-07-15:
Binds OK and controls working. I am happy for the issue to be closed.

pascallanger, 2022-07-15:
Before closing, can you try with different ex numbers to see if it works fine?

gke, 2022-07-16:
OK will do ASAP

pascallanger, 2022-07-16:
That's kind of strange. When changing the RX number the protocol is restarted with a different ID, starting with a bind then normal. The plane should loose connectivity and not recover unless you rebind it...
When you say up to 14, does it mean that after 14 it does not work anymore?

gke, 2022-07-16:
I am sure I commented the define out unless there are two defines.  I will recheck again tomorrow morning. The bind process when triggered is really quick . So if as you say "the protocol is restarted with a different ID, starting with a bind then normal" does that mean a rebind is being forced each time the RX number changes?

I ran RX number up from 0 to 14 and ran out of time at that point. Wedding Anniversary Dinner took precedence ;).

gke, 2022-07-16:
OK. Test again this morning. The define was commented out. Changing the channel number the bound light is still lit on the aircraft  (of course) but no longer acting on controls - duh. Power cycle aircraft and rebind and control is restored. I cycled up through 1..5 of the Rx numbers than tried higher numbers e.g. 32. Attached is the file as tested and this time with the define out completely.

[FX_nrf24l01.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/9126539/FX_nrf24l01.zip)
.

pascallanger, 2022-07-17:
Ok so we are good. Closing this thread.

closed 2022-07-17 by pascallanger

gke, 2022-07-17:
Thanks Pascal
Greg