pascallanger/DIY-Multiprotocol-TX-Module #707: Support for RH807 mini drone

FabianInostroza, 2022-07-20:
Please add support for the protocol used by this drone. Looks similar to JJRC H36. It uses the RF2518 chip on the TX and RF25155 on the RX. I've tried different protocols used on similar chips with no luck.
See https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/discussions/706 for some photos.

Regards.

pascallanger, 2022-08-01:
I've received the mini drone but I fear it's the wrong one. I've used the link you gave me but mine has a XN chip. This has nothing to do with the RF that you have.
I'm sorry but I could look at implementing this drone but that won't help you...

pascallanger, 2022-08-01:
Where are you located? I'm in France but I'll travel to the US in September. May be there is an option to ship the remote/drone and send it back.

FabianInostroza, 2022-08-02:
Hi, that's unfortunate. I don't know if the protocol in that drone is implemented yet, if it isn't and you have some spare time to reverse engineer it and implement it may help other people, you decide.
I live in Chile, I doubt shipping to/from here would make sense.

Anyway, thanks for this project.

pascallanger, 2022-08-02:
I've already started to look at the protocol.

### Bind
RF channel: 20
Send: approx 16150µs
RF settings: 1Mb, Scrambled
Address: D2 B5 99 B3 4A
Payload size: 10
4C|4B|3A|6F|00|00|D2|00|AA|AA
--|--|--|--|--|--|--|--|--|--
F0|F1|??|??|??|??|A0|??|EOF|EOF

F0, F1 could be related to the frequencies used in normal mode
A0: first byte of the normal address?
EOF: End of frame? same as normal mode

### Normal
RF channels: 73=0x49, 75=0x4B=F1?, 76=0x4C=F0?
Hop: approx 8025µs
RF settings: 1Mb, Scrambled
Address: D2 B5 99 B3 4A  ?? Where is this address coming from ??
Payload size: 12
00|00|FF|00|20|20|20|00|40|00|AA|AA
--|--|--|--|--|--|--|--|--|--|--|--
??|??|Flag change?|THR|AIL|ELE|RUD|??|Flags|??|EOF|EOF

EOF: End of frame? same as bind mode
Flags: 40=beg, 80=med, C0=exp
Flag change: every time you press a button the value change for a few packets then go back to FF

I could implement the protocol like this for now and see the flags later.

bla671, 2023-07-13:
is there an update to r807 binding? and if not is theres a manual brute force way to bind to rh807 drone to multiprotocol module?