pascallanger/DIY-Multiprotocol-TX-Module #710: power limitation for jumper t18

amysticool, 2022-07-21:
Hi, 
I notice that the rf power for 868/91MHz has change from latest update for jumper t18 and r9 frsky long rang receiver. This is limited to 50mW but the radio can go up to 300mW so why this is restricted ?? can it be corrected for future release ?

Thanks

Ostay, 2022-10-08:
The SX1276 transmitter module built into the Jumper T18's 5IN1 Radio can power up to 100mw, but the latest update to the 5IN1's firmware is limited to 17dbm(50mw). I want to know why? Look forward to adding 20dbm(100mw) option in later versions, thanks!

pascallanger, 2022-10-08:
It's simply how Jumper have designed the RF board. They've used the output where no power adjustment can be done and added a fixed gain of 50mW at the output stage.

closed 2022-10-08 by pascallanger

Ostay, 2022-10-08:
Thank you for your reply. I have understood the problem. Thank you very much!