pascallanger/DIY-Multiprotocol-TX-Module #717: Which files need to be changed to add a new protocol?

diego0815, 2022-07-29:
Hi Pascal

I tried to create a copy of Kyosho Hype sub-protocol (without FHSS) as a clear and safe basis for further experiments.
However whatever I try I never managed to make the RX bind under the new name, if I stub-out FHSS it all still works under the original name.

Thanks for any kind of hint.

Diego

pascallanger, 2022-08-01:
I'm not sure what you are looking for.
But to add a new protocol, you have to declare it in:
 - protocol number in multiprotocol.h
 - protocol and sub protocol in multi_protos.ino
 - config.h to enable the protocol

diego0815, 2022-08-01:
Thanks for your reply.

I cloned your repo to here: https://github.com/diego0815/DIY-Multiprotocol-TX-Module/tree/SX1280/Multiprotocol with the purpose to try playing with the SX1280 to emulate other RF components, not sure if it's feasible but tried to start playing with it.

To have a clean start for experiments and to rule out any interference from other bits of the code, I copied the hype protocol separate from FHSS and renamed it to shengteng, as my RX is built by that company...

However I can't seem to get it to work, as soon as it is no more under the original kyosho ino file.

I'm not sure but it seems bind might work, but then the rx blinks like lost RF. 

Kind of lost, so I wanted to know if I forgot to register the protocol somewhere, but seems not...

However, I guess I have to dig deeper, but currently ran out of ideas...

pascallanger, 2022-08-01:
I think you've forgotten to modify A7105_SPI.ino: https://github.com/diego0815/DIY-Multiprotocol-TX-Module/blob/71f7412570b5401a20fb06f81fa2d8eadd4ce222/Multiprotocol/A7105_SPI.ino#L174
It's related to the A7105 only. So you might not have started with the simplest protocol.
Adding the SX1280 is not an issue. Just look at what has been done for the SX1276 and implement it the same way.
Depending of your module, you have a SPI port which is available to solder an additional RF board (which is how the SX1276 is implemented).

diego0815, 2022-08-02:
Added the forced tuning, but no luck...

interestingly, same as before the mod, it seems to bind:
power RX with bind plug --> RX blinks green
bind on Kyosho hype --> RX bound (LED turns off)
power cycle RX without bindplug --> servo is controlled, RX LED off
change TX to STModel --> RX green blinking
power cycle RX with bindplug --> RX green blinking
start bind on TX --> RX blinks green (bind or no TX signal / can't tell if it changes the speed of blinking while binding)
power cycle RX without bindplug --> RX LED blinks, no control
change TX to hype --> RX needs binding again

so something happens, but not what I expected :)

pascallanger, 2022-08-02:
Ok I'll check further, push your changes into your branch so I can review the latest.

diego0815, 2022-08-02:
Thanks, it should be there.
I tried to override the ID with the one from your SPI traces and also tried to set the hopping table to all 0, not sure if that is anything clever todo, but well, no better ideas.

I hope I'm not wasting too much of your time :)

diego0815, 2022-08-02:
could it be somehow connected to the Arduino IDE used?
im on 1.8.19 on win for now - but then why would Hype protocol work?

diego0815, 2022-08-09:
I was thinking about the diferences between the stubbed out original file and the copy and the only difference I could imagine right now is timing differences.

Could that be a possibility that some timings are so much borderline, that even marginal changes affect operation? O just was thinking because in the original issue about addition of hype protocol you mention some irregularities in timing.

Also while testing I had quite often servo jitter without touching the TX.