pascallanger/DIY-Multiprotocol-TX-Module #718: Hardware compatibility CC2500

atharvbadkas, 2022-07-30:
Hello
Please anyone let me know about the compatibility of this Rx with project...I couldnt find original CC2500 Module so this is my option.
Link https://www.dnatechindia.com/CC2500-RF-Transreceiver-Module.html

[image]

Thanks in advance

pascallanger, 2022-07-30:
Nothing to do with this project...

atharvbadkas, 2022-07-30:
Well I want to connect my Flysky Tx with Receiver: 8CH Frsky protocol SBUS output D8 Mode

I already built Multiprotocol module with NRF for Bayang protocol
[image]

I want to add CC2500 module for frsky protocol
I couldn't found "banggood CC2500 Module" which is recommanded but I was able to found this module.
 Please help

pascallanger, 2022-07-30:
What about this one: https://m.fr.aliexpress.com/item/32945151938.html?spm=a2g0n.detail.1000014.12.51456c29PQoers&gps-id=platformRecommendH5&scm=1007.40000.267768.0&scm_id=1007.40000.267768.0&scm-url=1007.40000.267768.0&pvid=a8e25a6c-f6d0-430c-b56f-84abbb9b2588&_t=gps-id:platformRecommendH5,scm-url:1007.40000.267768.0,pvid:a8e25a6c-f6d0-430c-b56f-84abbb9b2588,tpp_buckets:668%232846%238111%23498&pdp_npi=2%40dis%21EUR%21%217.8%21%21%21%21%21%40211b5e1e16591799742016296eadde%2166253260547%21rec

pascallanger, 2022-07-30:
It has a PA which is important for range. There are plenty on AliExpress.

closed 2022-07-30 by pascallanger

atharvbadkas, 2022-07-30:
Banggood and aliexpress shipments are banned in India....so I cant order from it...
I only needed it for indoor use, so low range will not matter.
It has standard UART interface.
What do you think?

pascallanger, 2022-07-30:
Remove and use the small RF board. This is the only thing you need on your link.

atharvbadkas, 2022-07-30:
Okay..thanks
Can you share pin connection diagram with STM/Arduino ?

pascallanger, 2022-07-30:
Arduino: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/PCB%20v2.3d/Schematic_v2.3d.jpg
STM: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/STM32%20PCB/Schematic_Multiprotocol_STM32_MB_v1.0_t.jpg

atharvbadkas, 2022-08-04:
Hi
I did remove the RF board. Replaced the NRF with CC2500 RF board...made proper connections of MISO MOSI SCK and CSN.

Previous NRF protocol circuit
[image]

Made changes in config.h file(selected only frskyD protocol) and pin.h file(I had to change PPM MISO MOSI CSN Pins so it can match previous circuit).Code compiled successfully so Uploaded the code in Arduino Nano.
But its not binding with the quad.(tried AUTOBIND and later Channelbind also)

FC :BEECORE FRSKY F3_EVO_Brushed ACRO Flight Control Board
RX:8CH Frsky protocol SBUS output D8 Mode

Is there anything I'm missing here?

atharvbadkas, 2022-08-04:
Pins I changed in pins.h
PPM D2
CSN A1
MISO A0
MOSI D3
SCK D4

pascallanger, 2022-08-04:
What voltage are you applying on the arduino board?
I'm asking since the NRF24L01 has its pins 5V compatible so you can plug it to a 3.3V or a 5V micro controller it will work just fine. But that's not the case of the cc2500, it only accepts 3.3V on its pins. If you have used it with 5V then I'm afraid that you've burnt it...

atharvbadkas, 2022-08-05:
Yeah I have placed 3.3v regulator for RF board.

atharvbadkas, 2022-08-05:
Is there anyway to know on serial monitor, whats going on? As troubleshooter

pascallanger, 2022-08-05:
> Yeah I have placed 3.3v regulator for RF board.

I tried to explain you that a 3.3v regulator for the RF board is not enough. All the signals reaching the CC2500 must be 3.3V

atharvbadkas, 2022-08-05:
Ohh snap...I didnt looked into datasheet and assumed it will work...will add level shifter and new RF board

schwabe, 2022-08-05:
Or run the Arduino at 3,3V

pascallanger, 2022-08-05:
As Arne said, this is what we do on the schematic I've shared with you.

atharvbadkas, 2022-08-07:
Hello,
I have ordered new RF board with logic level converter.
In the mean time can you check my Config.h if there's anything wrongly configured?

FC : BEECORE FRSKY F3_EVO_Brushed ACRO Flight Control Board
RX: 8CH Frsky protocol SBUS output D8 Mode

Which sub protocol to select from FRSKYD, DCLONE for this Rx ? or PPM_SBUS ?

config-
`
//#define USE_MY_CONFIG
//#define CHECK_FOR_BOOTLOADER
#define AETR 
#define ENABLE_BIND_CH
//Set the channel number used for bind. Default is 16.
#define BIND_CH	5
#define WAIT_FOR_BIND
#define CC2500_INSTALLED
//#define CC2500_ENABLE_LOW_POWER
#define	FRSKYD_CC2500_INO
#define ENABLE_PPM
#define MIN_PPM_CHANNELS 4
#define MAX_PPM_CHANNELS 8
#define NBR_BANKS 1
const PPM_Parameters PPM_prot[14*NBR_BANKS]=	{
#if NBR_BANKS > 0
//******************************       BANK 1       ******************************
//	Switch	Protocol 		Sub protocol	RX_Num	Power		Auto Bind		Option	Chan Order
/*	1	*/	{PROTO_FLYSKY,	Flysky		,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	2	*/	{PROTO_AFHDS2A,	PWM_IBUS	,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },	// RX number 0
/*	3	*/	{PROTO_AFHDS2A,	PWM_IBUS	,	1	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },	// RX number 1
/*	4	*/	{PROTO_AFHDS2A,	PWM_IBUS	,	2	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },	// RX number 2
/*	5	*/	{PROTO_AFHDS2A,	PWM_IBUS	,	3	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },	// RX number 3
/*	6	*/	{PROTO_AFHDS2A,	PWM_IBUS	,	2	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },	// RX number 4
/*	7	*/	{PROTO_AFHDS2A,	PWM_IBUS	,	3	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },	// RX number 5
/*	8	*/	{PROTO_FUTABA,	NONE		,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },	// option=fine freq tuning
/*	9	*/	{PROTO_FRSKYV,	NONE		,	0	,	P_HIGH	,	NO_AUTOBIND	,	40	,	0x00000000 },	// option=fine freq tuning
/*	10	*/	{PROTO_FRSKYD,	PPM_SBUS		,	0	,	P_HIGH	,	AUTOBIND	,	40	,	0x00000000 },	// option=fine freq tuning
/*	11	*/	{PROTO_FRSKYX,	CH_16		,	0	,	P_HIGH	,	NO_AUTOBIND	,	40	,	0x00000000 },	// option=fine freq tuning
/*	12	*/	{PROTO_FRSKYX,	EU_16		,	0	,	P_HIGH	,	NO_AUTOBIND	,	40	,	0x00000000 },	// option=fine freq tuning
/*	13	*/	{PROTO_DEVO	,	NONE		,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
/*	14	*/	{PROTO_WK2x01,	WK2801		,	0	,	P_HIGH	,	NO_AUTOBIND	,	0	,	0x00000000 },
#endif
`
Thanks