pascallanger/DIY-Multiprotocol-TX-Module #724: Can't install Bootloader on Jumper T12 Pro

SteveHindmarsh, 2022-08-20:
I have tried to upgrade from V1.3.1.9 RETA to mm-stm-serial-reta-v1.3.3.14.bin to no avail.
The update via the transmitter SD file fails with the FW update error NoSynch message.
I have tried using Flash Multi to update the bootloader but it fails to communicate with my TX via any USB mode even after asking it to install drivers and following the driver download instructions from your documentation. The CP210x_Universal_Windows_Driver is installed but I think Windows provides this by default now anyway?
The T12 Pro has an internal Jumper4in1 module which therefore must be connected to via the Radios USB, it is not a stand alone multi module.

OpenTx, BetFlight companions both communicate with my radio no problem in various USB modes so I fail to see how this can be a driver issue? What am I missing here?

benlye, 2022-08-20:
Flash Multi doesn't work with internal modules - the internal module isn't connected to the radio's USB port in any way.

benlye, 2022-08-20:
Flash Multi [documentation](https://github.com/benlye/flash-multi/blob/master/README.md#unsupported-modules) [updated](https://github.com/benlye/flash-multi/commit/5baf34d055c91859dfa03f79b8408d7714492f8b).

SteveHindmarsh, 2022-08-20:
I'm not seeing how any of your documentation update helps me? How exactly can I update my internal module?

benlye, 2022-08-20:
It doesn't help you. I just clarified the doc to explicitly state that you can't use my tool in the way you are trying.

If you can't update from the radio then you're stuck unless you want to solder a USB-to-serial adapter to your internal module. 

You should probably ask your radio reseller or manufacturer for support.

benlye, 2022-08-20:
If you're invested in the soldering route, the process is described [here](https://www.rcgroups.com/forums/showthread.php?3428887-Jumper-T16-Internal-Multiprotocol-Module-Firmware-Update).

It was written due the T16 but should apply to the T12 Pro.

SteveHindmarsh, 2022-08-21:
You say, ask your radio reseller, thing is its OpenTX and others (yourself) making the changes not them, my last internal multi module update (V1.3.1.9 RETA early 2020 I believe) installed no problem from the SD without requiring Flash Multi? I don't understand what has changed and why between then and now? I may have to take the soldering route but can't understand why this drastic change has ruined what was a perfectly functioning updateable internal multi module?

pascallanger, 2022-08-21:
What has changed is most likely your opentx or edgetx version.
There is nothing we can do to help you. The upload to the module is done by opentx or edgetx. If it doesn't work anymore you need to contact the dev. The bootloader of the internal module has not been changed.

The only thing we can help you with is to connect to the serial port of the module and use flash multi since we are in full control of that environment.

My opinion is that you should update the radio software to the latest and/or look at the issues for someone having the same issue.
Since you can reproduce the issue, they will be able to help you quickly but you need to contact them not us.

closed 2022-08-21 by pascallanger