pascallanger/DIY-Multiprotocol-TX-Module #728: DSM Forward Programming - AS3X/SAFE Off-On values reversed

yakulis, 2022-08-31:
I am working with a TX16S radio and AR631 receiver. When attempting to program the AS3X and SAFE settings, I noticed that the values being reported back on the screen were the opposite.  

Gyro Settings -> SAFE Select ->    The menu items AS3X:, SAFE:, and SAFE Select all have on and off values that appear to be reversed.

I changed my local copy of the script, reversing lines 555 and 556 which contain "Off" and "On", and now the appropriate values match.

I am unable to change the SAFE: value however. I will be getting a receiver programmer in the mail tomorrow to see how programming the receiver might change the values on the forward programming. More to report later....

yakulis, 2022-09-01:
Verified again today that the values are indeed reversed. I was able to reset the AR631 receiver with an SPMA3065 programmer and then tested all of the Forward Programming. I was able to setup Safe Select, SAFE, and AS3X modes using the script after I switched the Off and On values on lines 555 and 556. 

I recommend this change be implemented to the script.

qsiguy, 2022-09-11:
I had the same issue with the AR631 with on/off values swapped.  Was about to change the on/off script lines but came looking to verify.  FYI, I found the same issue also with the wheel not able to swap to OFF but if you long click on the field it would change the value with out scrolling to OFF.

yakulis, 2022-09-11:
Agree that the receiver has to be "unlocked" first if it was a BNF model by performing a firmware update. I used the SPMA3065 programmer to load an older version of firmware. Then I was able to reload the firmware set to the model of my plane with it having the additional menu items and functionality. That way all of the gains and throws are loaded from the model (which can't be adjusted through the script), but you can still set the other functions. 

This is all great information, but I don't to lose the fact that the On and Off values are incorrect in the script and need to be switched.

yakulis, 2022-09-11:
Change has been incorporated with #733

closed 2022-09-11 by yakulis

frankiearzu, 2022-11-29:
Yakulis... you confirm my findings that on/off was reversed in 1.3.3.14. thank you.