pascallanger/DIY-Multiprotocol-TX-Module #730: Unable to bind FS2A-4CH AFHDS2A Receiver

HinkePank, 2022-09-08:
Hello, I just bought 4 of these very small recevers on Ali.
[image]
Unfortunately I am not able to bind it to my IRangeX 4 in 1. I t has the latest stable sw 1.3.3.14 and I tried all th FlSky2A variants and also all the FlSky variants as well.
When I power-on the receiver with the bind-button pressed, it flashes quick but never changes the blink-frequency to lower.
Tried very near distance (5 cm)  and also more away (1m).
I don't own an original FlSky transmitter, so I can't test if the revceivers will ever work or if I just bought fakes.
Does anybody here has expierience whith those receivers ?
What else could I try ?

Thanks a lot for any response
Harald

pfeerick, 2022-09-09:
Just to cross-reference this and give a little more info, one of our users as also just commented on this, and indicated there appears to be some difference in the design of the boards (pictures in the issue). https://github.com/EdgeTX/edgetx/issues/2291

pfeerick, 2022-09-21:
Just adding a little more trivial info to the mix ... I now have four of these with the square STMF031 chip - two of each from the two sellers mentioned in this issue. A test specimen from each seller refused to bind to the MPM, whereas my FS-iA6B and FS-iA10B Flysky AFHDS2A RXs were fine. However, I do have an NV14 so could try genuine AFHDS2A and they bound straight away. Ok, that is technically not true... I did the bind process on these wrong in putting the RX into bind mode before the TX, and had to bind them twice before it stuck, so I will try with the other two the right way around, but I'm pretty sure that was just PBKAC error 😆. It looks like they are mid May and late June production batches, and have the exact same markings on them as the other examples of misbehaving-working RXs. So the "bad" RXs may be becoming more common... 😢 

[image]

pascallanger, 2022-09-21:
Unless someone sends me one of the troublesome RX, there is nothing I can do.

pfeerick, 2022-09-22:
I believe [cmlarsen](https://github.com/cmlarsen) was [arranging to send one](https://github.com/EdgeTX/edgetx/issues/2291#issuecomment-1241490568) about two weeks ago. If that fell through or goes astray, I'm happy to donate one or two to the cause also ;) I mainly got these because they seemed a handy small size, and was hoping to get two good ones and two dodgy so the game of "spot the difference" could begin. :grin:

pascallanger, 2022-09-22:
No he hasn't sent me anything...

pascallanger, 2022-09-22:
I need either one to be sent to me in France or donate some money so I can place a purchase of the known not working RX but in this case we must be sure that I will get the right one.

pfeerick, 2022-09-22:
Fair enough. You have a PM waiting for you on RCG now ;)

pmumby, 2022-10-26:
Just wanted to chime in, as it turns out I have a number of these just arrived from one of my suppliers, they suffer from the same issue. Using a Jumper T-Pro transmitter. I have tons of them in stock, and they are all the "larger chip" version, and won't bind using any of the FlySky protocols on the T-Pro internal MPM.

I'm happy to provide one for troubleshooting if that's helpful, or provide any assistance as needed, as this will quickly turn into a customer service nightmare for me as the radios I sell can't bind to the receivers I sell ;)

@pascallanger please feel free to reach out to me at admin@combatrobotics.co.nz if you want me to provide a receiver, (thought it will ship from New Zealand, I'll gladly ship it at my expense). And let me know anything I can do to assist in troubleshooting.

Cheers!

HinkePank, 2022-11-04:
I wanted to have a try with the receivers with the longer (rectangle) chip on the back and ordered one from ALIexpress some weeks ago. It had just arrived and I immediately tried to bind it - it worked !
I am using a Taranis X9D+ with newest OpenTX-sw, a IRange 4 in 1 with the latest Multiprotocoll-Firmware.
As protocoll I selected FlSky2A PWM+SBUS.

I can see that on the working receiver the rectangular chip on the back (no letters on it) had shortend pin 5 and 6.
If this is also a STM32F031C6T6 (like the square), it would mean that there is a short from VDDA to PA0 ??

But also the front of the PCB differs slightly.

I hope this info helps.
I know that Pascal bought a receiver from Ali and is waiting for it. I hope he can soon spend some time on it.

I also hope that he gets a bit more financial support as he is doing a real good job for us !

Regards!

MarkR42, 2022-11-17:
Ok I hope the problem can be identified. If I am able to, I shall try to compare the SPI sequences between the old / new FS2A receivers (although I only have the old model just yet)

pascallanger, 2022-11-17:
I've received this RX model :
[image]
[image]

I can bind it by touching the antennas.
The RX control is working fine with no packet drops reported, I've only tested a few meters away but it looks good.
This RX is sending telemetry but it barely works at near range (losing a lot of telemetry frames) and doesn't work at all a few meters away... Does anyone know if an original FlySky TX receives the telemetry fine far away?

At a first look, Multi has troubles to receive data from this RX for whatever reason.

pascallanger, 2022-11-17:
Test Multi v1.3.3.23 from: https://downloads.multi-module.org/latest-test/
You should be able to bind this RX more easily now, it binds at 1m or closer for me (you need to wait a couple of secs for the bind to happen).
Fix: during bind, I'm now turning on/off the LNA time to time. Since this RX seems to have really low transmission range Multi was not able to receive any data without the LNA enabled.

If someone can do some regression tests with different RXs.
Please report here.

I don't know if the telemetry range issue is normal or not. But the only thing left would be to play with the frequency tuning.

HinkePank, 2022-11-18:
I could bind it also now - RX antenna directly in contact with the multimodule-antenna.
I compiled the v1.2.3.20 with : 

#define USE_A7105_CH15_TUNING

and then played around with the frequency - i could not see any effect to make telemetry better.
It is always changing rapidly.

pascallanger, 2022-11-18:
Why don't you test with the latest version 1.3.3.23 ?

pascallanger, 2022-11-18:
Master is always latest

HinkePank, 2022-11-18:
Ok - just learned it the hard way ;-)

HinkePank, 2022-11-18:
I compared now version 1.3.3.20 with 1.3.3.23 :
Binding works very good in .23 (50-60cm distance)
No difference in telemetry-behavior.
Playing with tuning is also equal in both versions - no big influnce to the telemetry

Even when I got no telemetry-info back, the attached servo was always responding well and the blue led was still stable on.
Also trying a range-test was not to bad - sorry, just in the house, no comparison outside pssible.

pfeerick, 2022-11-19:
Binding with MPM 1.3.3.23 (internal MPM on TX16S) is now working great... and looks like it's about the same range as the telemetry... i.e. between 2-3 metres. After that, the telemetry link drops. I compared this with the Flysky NV14, and there was negligble difference between them. So this seems to be a problem with the Rx, not the MPM/TX (however, this is with a sample of 1, so I will try another Rx and will edit this with the result). I've also got the Flysky iA6B and iA10B so will verify they still bind and work normally. 

I did a brief open-air walk test with the transmitter running a servo sweep, and at approx 50metres at full power, there was no issue with servo movement, even when fully blocked by body and hand over the antenna. At low power and range test on TX16S /w internal MPM, I could get it to start stuttering at around 30m, and fully drop out at 50m with a body block. The Flysky NV14 was the same, or perhaps a smidgen worse in range when in range test. In all cases, the antenna were not in ideal orientation, so it's quite possible they could be improved on.

HinkePank, 2022-11-19:
Just sgort info about the directly working receiver ( with the rectangle chip at the back ) :
This does not provide telemetry.
Range test with reduced power brings equal results for both types.

pascallanger, 2022-11-19:
Ok so I think we can say that this issue is fixed.

HinkePank, 2022-11-19:
Thank you very much !

closed 2022-11-19 by pascallanger

pmumby, 2022-11-20:
Not intending to re-open this, just wanted to confirm the comments above, did we essentially find that the older "rectangular processor" FS2A devices simply did not have telemetry, and the problem is they added telemetry in the ones with the newer chip, but obviously without some level of testing (ie: the telemetry transmission isn't working correctly, and has nearly no signal strength). The result of which is the binding process was broken due to the extra telemetry being present but non-functional? Or was it something entirely different? (I haven't taken the time to review the actual code changes yet). Anyway, thanks heaps for your help on this Pascal! Greatly appreciated!

pascallanger, 2022-11-20:
They have added some telemetry compared to the previous version. But they have forgotten to set the RF power output of the cc2500 to its maximum.
This is resulting in bind and telemetry issues. Bind is affected since Multi must receive some data from the RX but due to the low RF power it was not possible. The fix was too increase the sensitivity of multi so it can receive the data from the RX.

MarkR42, 2022-11-23:
FYI, I was able to bind to Multiprotocol 1.3.3.0 (without the update above) on my new-style FS2A using my Jumper T-Pro internal multiprotocol module.

So maybe the builders of the FS2A have fixed it in newer firmware, or perhaps it only happens on some units.

josephduchesne, 2022-12-21:
I've got a radio that uses the "large" chip version. It wouldn't bind on  1.1.x.x (some old version I had installed) or the latest stable (1.3.3.20), but the latest dev build (1.3.3.24) bound first try. I think this is fixed.

wv1124, 2024-05-21:
i cant bind with new-style in 1m，but lost connect in 3~5m  with flysky i6x

TomTech2004, 2024-07-03:
Hello
Can someone tell me what pins of the processor the channels are connected to?
Thanks