pascallanger/DIY-Multiprotocol-TX-Module #734: can't discover more than 2 temp sensors in Devo protocol

khmewes, 2022-09-13:
Hi, I am trying to add more than two temperature sensors (4 in total) in my 4in1 Devo protocol telemetry using an OpenTX Radiomaster TX16S 4in1. So far no luck it will only discover two sensors, T1 and T2. I am sure I have 4 sensors T1-T4 using the Walkera CTL01-D telemetry module with a MX400 quad. It works with the original Walkera Devo F12E and with the Deviation Jumper T8SG 4in1. My question is, does OpenTX have a limitation of two temperature sensors and is there a way to get around this, i.e.add more temp sensors? Is there a fix?
Best Regards

pascallanger, 2022-09-13:
This is not a bug but per design and inline with the documentation: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Protocols_Details.md#devo---7
I'm using frsky d8 protocol to send the devo telemetry data back to the radio. The issue is that the frsky d8 telemetry protocol supports only 2 temp sensors. You can modify the code if you want to send T3 and T4 in other unused telemetry fields.

khmewes, 2022-09-13:
That's a little confusing. are you saying you are using frsky D8 together with the Walkera Devo protocol but for telemetry only? Or is the Devo protocol actually frsky D8 protocol? Nevertheless how would I modify the source code and where can I even get it? If modified wouldn't a new version of the Multi firmware have to be created to incoporate the modification? I think this goes beyond my capabilities. Can somebody do this for me (you? :-)) please?

pascallanger, 2022-09-13:
Yes this is the code sending the temperature data back to the radio. It only looks for 2 working temp sensors and ignore the others.
You can't just change the 2 by a 4 since as I told you the issue is with the frskyd8 telemetry, so you need to send the 2 temp sensors somewhere else since it only supports 2:  frsky_send_user_frame(0x02 + i*3, val, val>>8);

pascallanger, 2022-09-13:
You could use Acc-x (0x24) and Acc-y (0x25) to store the additional temp values as an example:
`			case 0x31:	// Temperature packet
				//memcpy(&packet[1],"\x29\x2A\x2B\x00\x00\x00\x00\x00\x00\x00\x00\x00",12);							// 21°, 22°, 23°
				for(uint8_t i=0; i<4;i++)
					if(packet[i+1]!=0xff)
					{
						val = packet[i+1];
						val -= 20;
						frsky_send_user_frame( (i<2 ? 0x02+i*3 : 0x24-2+i ), val, val>>8);												// temp 1 & 2 and acc-x/acc-y
					}
				break;`

khmewes, 2022-09-13:
Will i be able to discover/see all 4 temp sensors then in the otx telemetry section/widget on my radiomaster TX16S?
Of course I have no clue how to open a project and compile it and then get it in the firmware on my radio. Are you interested in creating a modified version of the Devo protocol? I think this is of interest not only for me with my old Walkera MX400 drone and installed CTL01-D telemetry module, but the CTL01-D or similar module could also be installed in other Walkera Quads like the QRX350, Scout X4, Tali etc. Since the temp sensors are for the motor temps it makes sense to have 4 and the Tali even has six motors. These 'relatively newer' quads curiously don't transmit motor temps data for some reason. Perhaps Walkera wasn't any longer concerned about this or it was a production cost issue. Who knows. Anyway if you are not interested in modifying the code, please send me detailed instructions on how to do it Arduino OSD software installed on my computer and did some programming  in the past. I am just a bit 'rusty' with it.
So whatever you are willing to do on this is highly appreciated.
Cheers,

Klaus

pascallanger, 2022-09-13:
You'll see 2 temp sensors as of today and the temperature of the 2 other sensors will be displayed in acc-x and acc-y.
How to compile is well explained on this github: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/docs/Compiling_STM32.md
I can do a push of a new version including the above change but it would be great to test it before which I can't do this week.

khmewes, 2022-09-13:
Thank you that would be great. Just tosatisfy my curiosity i will visit the link and attempt to modify and compile it. I am not sure about the outcome so I will rely on your work for this. But it might be an interesting experience. So whenever you can get it done that is fine by me, no rush, it's a hobby afterall, at least for me (-:
Much obliged
Klaus

khmewes, 2022-09-14:
It occurred to me that you might not have a Walkera CTL01-D to test the mod so I could do the beta testing for you, if you agree.

khmewes, 2022-09-15:
Hi Pascal,
why was Frsky protocol used for telemetry? Which protocol is Walkera using and can/could that be used instead? Is there a patent or copy right issue?
btw I managed to modify and compile the firmware. But since I have little experience with this project/sketch I am very hesitant to install it on my Radiomaster TX16S. I will wait for your official release.
Salut
Klaus

khmewes, 2022-09-15:
Thanks schwabe for the info. Let's see how the work-around Pascal is implementing works. It uses the 2 accelerometer channels acc-x and acc-y which are usually used for pitch , yaw and roll telemetry and I don't use these in these drones we are working on. So we'll see how it works out.
I will keep you in the loop/mind for additional work-arounds.
Thanks for the reply and info
Klaus

khmewes, 2022-09-15:
Hi Pascal,

I modified the section in the Devo code by replacing the case 0x31 section with the code you wrote. I compiled the Multiprotocol sketch for the STM32F103CB board without errors or warnings. I did not change the original config.h code. Do you think that is ok? I exported the compiled sketch to a .bin file ‘multi-stm-1.3.3.19.bin’. Will that work on the Radiomaster TX16S?

Thanks,

 

Klaus

khmewes, 2022-09-16:
well I installed to vewly compiled firmware on the Radiomaster TX16S. I could discover and edit the  acc-x and acc-y  sensor readings at  ID 0024 and 0025. The temp readings were 0 deg initially so I adjusted the ratio setting first, leaving the offsets alone for now. So OpenTX is handling the acceleration sensor differently than the temp sensors (gain etc) as expected. I ended up with values of 245.4 for accx and 228 for accy when I matched them to temp reading of tmp1 and tmp2 at 25 deg C. I also noticed that the tmp1 reading was 1 deg higher at 26 deg C than the tmp2 so I will go through a ccalibration step with my Multimeter temp probe just for kicks. I don't believe the offsets need to be adjusted unless there is a huge discrepancy at higher temperatures like at 50-60 deg C, which is about as hot as the motors get during flight in summer.
I hope my compilation effort didn't screw up any of the other features and prorocols. I will carefully test the other Devo and Walkera 24xx, 26xx and 28xx models I have as well as the OMP protocol.
I will give an update as soon as I am confident that everything works as it should.
Thanks again for the code and the help in general.
Cheers, and happy and safe flying
Klaus

khmewes, 2022-11-21:
Hi Pascal,

Not sure if I can contact you this way. But if so I have a question. Would
it be possible to write a Lua script to display the Pitch channel output as
the blade angle instead of a %-age and display it as a screen widget, like
in the channel output widget? Or as a  separate screen display? I understand
that the %-age would have to be calibrated, either manually or from the
throttle/pitch curves. Let me know if you get this message if you don't
mind.

 

Best Regards,

 

Klaus Mewes

closed 2023-10-30 by pascallanger