pascallanger/DIY-Multiprotocol-TX-Module #738: Eachine E008 Protocol

rally-dimi, 2022-10-01:
Hello everyone,

Is there any indication on what protocol might be used in `Eachine 008` drone ? 
Apparently there's a [E010](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/457) and a lot of instructions on how to pair that, but nothing on E008.

Appreciate any help, hints where to start. (I tried protocols that are used for E010 and few other ones)

pascallanger, 2022-10-01:
Most of the time the protocol is different since eachines is mainly a reseller of different manufacturers. Try all the protocols and sub protocols. If none work, you need to send me your TX+quad to France or do a donation to purchase it.
Pascal

rally-dimi, 2022-10-03:
Is there a lua script that would allow me to do that automatically ? 
I'll try that, if not - then yes, i'll buy you a drone from bangood eu for a good cause.

rally-dimi, 2022-10-06:
One more clarification - the chip inside the trasmitter says `2124ct` and maybe one more 1/2 size font letter.

[image]

[image]

rally-dimi, 2022-10-06:
Drone IC's:

[image]

rally-dimi, 2022-10-25:
@pascallanger what is the address where I can send it ?

pascallanger, 2022-10-27:
@rally-dimi You can contact me on pascal _ langer @ yahoo . fr