pascallanger/DIY-Multiprotocol-TX-Module #747: OMP protocol no longer need freq tune?

richardclli, 2022-11-10:
Just spot that OMP protocol is no longer need freq tune.  It seems this change is after the change in XN297 emulation layer.

Just want to know why some C2500 based protocols do not have the freq tune options enabled?  No need or just a mistake?

pascallanger, 2022-11-10:
That's strange... Let me check...
You have a cc2500, correct ?

richardclli, 2022-11-10:
Yes, and seems it is related to this commit:
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/commit/4a626eaf144b4b7834d104717e7430fd5b154883#diff-411d4cbbc40a0214c0cf019de71e0de1edbf82c4413a20a5bd80fddbb79d3f81

richardclli, 2022-11-10:
Many CCNRF related protocols have their definition in Multi_Protos.ino uses OPTION_NONE, instead of OPTION_RFTUNE, is this the cause?

pascallanger, 2022-11-10:
Are you using EdgeTX or OpenTX ?

pascallanger, 2022-11-10:
The OPTION type is decided on the fly depending on which component is the best suited for the job. So I'm wondering if EdgeTX which is pulling the full protocols list is updating the type afterwards if I send it...

richardclli, 2022-11-10:
EdgeTX

pascallanger, 2022-11-10:
I'm pretty sure that this is an EdgeTX issue. @raphaelcoeffic can you have a look into this issue?

richardclli, 2022-11-11:
I tried to change a line in Multi_Protos.ino and the Freq. Tune options appeared:

-		{PROTO_OMP,        STR_OMP,       NO_SUBTYPE,            0, OPTION_NONE,    0, 0, SW_NRF,    OMP_init,        OMP_callback        },
+		{PROTO_OMP,        STR_OMP,       NO_SUBTYPE,            0, OPTION_RFTUNE,    0, 0, SW_NRF,    OMP_init,        OMP_callback        },

richardclli, 2022-11-11:
So the protocol to EdgeTX is to send the definitions in Multi_Protos.ino to EdgeTX, right?  Please point out if I was wrong and give me some direction to look into.

pascallanger, 2022-11-11:
The option field and other parameters are set depending on which RF component is the best for a given work based on the bitrate and the presence of RF components: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/2d469d074ebea1f6a48220a2c3fd416f9e703aef/Multiprotocol/XN297_EMU.ino#L56

EdgeTX is pulling at startup all the protocols and their features. But at that time and until the protocol is actually launched there are some unknowns.
Once the protocol is launched, multi sends in each status frame to the radio what exactly should be displayed. -> This is where the issue is. EdgeTX seems to ignore the changes being sent.

pascallanger, 2022-11-11:
Yes, the list you are pulling at the beginning of for the different downs that you needed.
But once a protocol is launched, you should display what's requested in the status frame.

jogybaer0815, 2023-05-20:
Is there any movement? 
Freq. fine tune still not available at OMP protocoll

richardclli, 2023-05-22:
> Is there any movement? Freq. fine tune still not available at OMP protocoll

EdgeTX 2.9.0 will have this fixed.  There is a PR already, need some more tests to finalize.

richardclli, 2023-06-11:
It is already there in current nightly, so this can be closed.

closed 2023-06-11 by richardclli