pascallanger/DIY-Multiprotocol-TX-Module #750: in DSM FwdPrg.lua, value displayed for flight mode does not match DX8e value

Dweinbach, 2022-11-14:
in Gyro Settings/Flight Mode Setup on a Spektrum RX, the values I get on my AR637T are 1,2,3 as channel goes -100,0,100.

On my RadioMaster TX16s with EdgeTX 2.8.0RC4, the values displayed are 0,1,2. This can be confusing, since 1 and 2 are actually showing the settings for 2 and 3 (confirmed)

frankiearzu, 2022-11-16:
Check my branch if you want yo test it.
Look for issue #751 for more info.

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/751

frankiearzu, 2022-11-17:
Added code in my new version to handle this case..  instead of showing  "Flight Mode:   1",  it properly concatenate them and showing them as a SubHeader(centered) as "Flight Mode 1"  (no ":")

closed 2023-10-30 by pascallanger