pascallanger/DIY-Multiprotocol-TX-Module #751: DSM Forward Programming Enhancements (New GUI, etc)

frankiearzu, 2022-11-16:
Already done, but this is the work:

1. Separate GUI from DSM Fwd programming protocol logic and menu
2. Create DSM simulator to help develop GUIs
3. Abilility to log the conversation between RX/TX to be able to understand better the behaviour and fix problems
4. Comment better the code

frankiearzu, 2022-11-16:
@pascallanger 
I think i have no permissions to create a pull request from your repository.
I worked my changes on a copy:

https://github.com/frankiearzu/DIY-Multiprotocol-TX-Module/tree/frankie-dsm-fwrd-prg-enhancements

right now in the main /SCRIPTS/TOOLS directory, there are two files,
one for EdgeTX (Color/Touch) and another for OpenTx (Back&White, simple text).  I am using in the color version a few calls who are not supported in OpenTx, but i think i can make it work for EdgeTX/OpenTX after a few more work.  The idea of the Black and White is to port the GUI interface to other RX with smaller screens.  All share the same DSM library.

The code can run locally on Companion 2.8 using the simulator, as well as TX (RM T16S for me).

In the beginning of the file, you control if using a simulator or a real thing:

local SIMULATION_ON = false  -- FALSE: use real communication to DSM RX (DEFAULT), TRUE: use a simulated version of RX 
local DEBUG_ON = 1           -- 0=NO DEBUG , 1=HIGH LEVEL 2=LOW LEVEL   (Debug logged into the /LOGS/dsm.log)

if  SIMULATION_ON is true, it will start in simulation mode, but there is an option to go to the real RX at the end
(make it easier to test on Companion as well as the TX without changing the SIMULATION_ON back and forth.

[image]
[image]
[image]

**LOG_FILE:**  
The /LOGS/dsm_log.txt has the conversation with RX/TX.. that can help debug problems easilly, and was of greate
help to create the simulation by just copying the menu/lines values.

[image]

**DIRECTORY STRUCTURE**
Currently, i put helper files in a subdirectory:  /SCRIPT/TOOLS/DSMLIB to hide the lua file to show in the main tools of the radio.
one of my future plans is to add images of the different orientation of the RX when doing initial calibration

**WHERE TO CUSTUMIZE MESSAGES**
The file /SCRIPT/TOOLS/DSMLIB/DmsFwPrgLib.lua has all the DSM protocol specific code and messages
added a few formating options for the messages to control if you want the text CENTER/RIGHT align, and BOLD
information in comments of how to do it.  (for example, "Test\c\b"  \c for center, \b for bold) 
  
This allow better look of the screens:  (Center/Bold) for subHeaders and Flight-Mode
[image]

OpenTX/Black and White version: 
[image]

pfeerick, 2022-11-16:
You should be able to make a PR... have you tried clicking on the "Contribute" and then "Open Pull Request" buttons when on your github branch? That should take you to the page where you can submit the PR. 

[image]

This looks great btw, I'll have to give it a try with the 637 RX I have that supports forward programming :)

frankiearzu, 2022-11-16:
Thank you Peter,  when test it with AR637, the script will record the menus/etc in the /LOGS directory. With that log, we can debug more scenarios and see if there is any difference with AR631.. From what i understand, they share the same Fwrd programming firmware code.  On my AR631, everything works, except the sub-menus inside the "System Setup".. they get stuck trying to get the data.

frankiearzu, 2022-11-17:
@pfeerick thank you for the Github hint..  been after becoming a software manager for many years, haven't use github in some time :-)
code is checked in. feel free to provide feedback.

@pascallanger  feel free to assign bugs/improvement requests to me. I can definitly do GUI part.
Don't have a protocol analyser to do protocol reverse engineering. So far, there are three parts that still haven't found how they work, let me know if you want me to move them into a separate ticket of more (protocol reverse engineering)

1. AR631/AR637:  Gyro Setting -> System Setup:  All that submenu pages, the menus start reporting 0x05 (unknownmenu  lines) and cannot get out if it. stays stuck on the line 0... not the same behavious as 0x05 lines at the start of the main menu of the RX.
2. AR631/AR637: (with new RX)  Gyro-Settings->First Time Sertup: Was able to get this to work, when it reach the point of detecting the orientation of the RX,  it does not detect it correcly (return the current value regardless).  i guess besides navigating to the page, we have to send a command to the RX to detect orientation.
3. AR631/AR637: (with new RX)  Gyro-Settings->First Time Sertup: towards the end, it needs to reboot the receiver, it havigates to a special menu (menuId 0x0001) but the RX does not reset/restart, so i think there is a message to send there.
4. for some type of  List options  (especially line type MENU1 (0x0C) ) the range for the list of options seems incorrect (usually 0->244).  The pattern that i have seen is that is an  "ihnhibit" + range of options..   I cannot see in the data any pattern to get the proper value range.. maybe there is an extra message to send to RX get the proper value range or options.  Right now, for some i can do it via extra metadata on the definition of the message text.  The good think is that the RX validates the value and correct it to the closest valid option.

frankiearzu, 2022-11-19:
Just a few more improvements that i worked Friday night. Will try to solve some remaining problems before check-in the code.

Have the "First Time Setup" and "First Time SAFE setup" working. something that was not working in the original RX.

### RX orientation:
Added the image showing capabilities.. 
[image]

@pascallanger :  Currently i am putting all the RX images inside /SCRIPT/TOOLS/DSMLIB/img,  was tempted to put it in directory /SCRIPT/TOOLS/DSMLIB/; flatter structure helps deployment if people are manually copying things around.   Any sugestion??  having the "img" makes it more cleaner..  

### First Time SAFE Setup (angle page)
This page has a few new thing:
1. New line_type "0xE0" for data in degress. Haven't see it before. I think i start to see a pattern of some bits describing data type (8 bits, 16 bits, signed) and some bits for formatting (percent and degress).. will analyze that later.
2. Same MessageID (Message #)  to show two different text in the same receiver.. one message value when used as a line header and a different one for a Line List Value.  Now if we split the Text  array for the list of values (List_Text) it totally makes sence. Also solves previously reported contradictions  (for example, Label "Roll", "Yaw" was colliding with Channel names List, in this screenshow, the top label "Level model..." as colliding with List_Value for orientation of the RX).  Just thatby separating the text of menu headers (Text) and list values (List_Text) cleans up a few menu list invalid ranges. Will leave it in a way that it look first in one, then in the other (List will prioritize looking at List_Text first, the Text, headers the other way) to play it safe.

[image]

### Remaining screens with problems
Everything works right now in AR631 and Heli RX. The only think that it does not work, and can't pass it is the set of menus
in "Gyro Setting"->"System Setup".  All of them starts giving "Unknown 0x5" weird lines.  i was able to bypass similar problem in "Gyro->Setting"->"initial setup"  by sending 0 instead of the previously selected line when retriving the new menu, but does not work for this set of screens.  Will try other values to see if it works. Besides that, now you you can do the full initial setup that was not working in the original code. 

[image]