pascallanger/DIY-Multiprotocol-TX-Module #761: 1.3.3.20 loss of connections with V2 frsky receivers?

Luznatural, 2022-11-20:
Since I have upgraded the internal 4 in 1 module of my Radiomaster Tx16S Mk1 to firmware 1.3.3.20 (EdgeTX 2.7.1 firmware in the radio) I have experienced two catastrophic mishaps in glow planes that were fully reliable before.
Both planes were flying stable and straight ahead when suddenly started uncommanded movement without me being able to perform any input until the crash. Once recovered the remainings, the receivers, batteries or servos didn't show any damage and behaved correctly.
Fine tuning had been performed in both planes, and no RCCS warning nor low voltage of the battery was announced by the radio at the moment of the issue.
I discarded interference, as we were flying only 3 people in the field, and our club is well separated from any interference source.
Initially, I guessed there might be a problem with Edgetx firmware, so I downgraded it to the previous firmware, but the issue repeated with another plane. This time I was lucky as the plane simply lost command but remained straight ahead in a shallow dive, touching the ground very softly (only the landing gear affected).
Since then, I have performed several tests in the ground with different receivers and servos, and from time to time the phenomenon repeats without performing any special or remarkable command. Simply, it happens. I mean, I do not perform any brusque movement nor move the servos extremely, the battery is not going down on voltage, etc. You just have to wait with the radio on, and suddenly one or various servo move by themselves.
It,s like the receivers loosing connection with the radio (red light in the receiver for 1 to 3 seconds, again without any RCCS low or critical warning, nor low battery warning).
All the receivers I use are with the latest Frsky V2 FCC firmware (X8R) since 2021.
I downgraded my 4 in 1 module to the previous firmware, and I was unable to repeat the problem, so at this moment and with many doubts in mind, I think it is better to report it to you, just in case.
I have seen some similar reports on EdgetX issues, but in my experience (and apparently) the problem is not related to it.
You are doing a great job. I HOPE I am wrong!

benlye, 2022-11-20:
I've got a TX16S running EdgeTX 2.71 and a MULTI-Module running the released v1.3.3.20 build, with the module bound to an X8R running EU firmware v2.1.  I've set a failsafe on the RX and am running a servo sweep test on the TX to constantly move the servo.  If/when there is a glitch I'd expect the servo to stop moving and/or jump to its failsafe position, both of which I should hear/see.

I've run the test continuously for over 20 minutes and have not observed any glitches - no loss of servo control and no loss of signal shown by the RX LED.  How long does it take for you to observe the loss of signal / loss of control on your RX?

Luznatural, 2022-11-20:
Good afternoon benlye,
Thank you so much for your response.
My apologies to answer so late.
In my tests, sometimes takes 5 minutes to have the problem, but some others you stay for one hour and you don't get any error. Mi failsafe is set to "hold". I do not know if that,s a factor for the problem.
Today, in my club the same happened to a friend also with TX16S Mk1 ETX 2.7.1, the firmware of the module 1.3.3.20 and X8R with firmware 2.1.0 v2, with a gas plane that was separated only 300 meters. But this time the radio voiced "telemetry lost" before loosing control.
I believe it's a random occurrence. But I am afraid it happens
Again, I hope I am wrong!!!

pascallanger, 2022-11-20:
Which fw version are you going back to ?

Luznatural, 2022-11-20:
Good afternoon Pascallanger. I go back to 1.3.3.9

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

pascallanger, 2022-11-20:
1.3.3.9 is not a release... Have you compiled it yourself ?

Luznatural, 2022-11-20:
My apologies, I went to the radio to check. It,s 1.3.3.7.
I didn't compile it.

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

Luznatural, 2022-11-20:
Yesterday, I read in the list of new issues for ETX someone telling something smililar but mentioning it was caused by ETX 2.7.1.
I do not know if it's related or not.
But now I cannot find there that issue...

benlye, 2022-11-20:
@pascallanger should we create a 1.3.3.20 test build using the old compiler?

I still haven't been able to reproduce the issue.

pascallanger, 2022-11-20:
@benlye Yes let's try to rule it out.
I'm not able to reproduce it as well using a S6R but I'm still on opentx.

benlye, 2022-11-20:
@Luznatural can you try to reproduce the issue with each of the firmware builds in this zip file, please?  They are both from the v1.3.3.20 release code but are created with different compilers.

In your radio they will both show up as v1.3.3.20 so you will need to keep track of which one you flash and test.

[multi-stm-1.3.3.20.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10050665/multi-stm-1.3.3.20.zip)

Luznatural, 2022-11-20:
I will do that tomorrow morning. I am out of home now. Do not worry. I will Keep track and report back

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

Luznatural, 2022-11-22:
Good morning to all,

I have been doing extensive tests with the three firmwares that you sent me, using the same receiver installed in one plne without wings to be able to see the receiver, reduced power on and taking the radio to a distance of 50 meters, making a fine tuning, and watching RCCS. Special care taken to make sure which version of firmware was installed. All firmware updates to the internal 4 in 1 module were made from the SD firmware folder, using the radiomaster TX16S Mk1.

Firmwares:

  *   1.3.3.20 Released, I will call it released
  *   1.3.3.20a, I will call it a
  *   1.3.3.20b, I will call it b

Results:

  *   Fine tuning in all three firmware verion resulted in: released +4; a +5; b +4.
  *   I have been able to reproduce the failure in released 3 times in 2 hours of checking. The symptoms were green light of the receiver going off, total loss of signal during 4 to 5 seconds, then green light again. RCCS went to cero without any warning and back to normal after green light reappeared.
  *   Unable to reproduce the failure in releases a and b in two hours testing each, but detected from time to time a blinking for less than a secondin of the green light of the receiver, without loss of control and no RCCS warning, just a fluctuation of the RCCS from 82 to 72 and back.
  *   RCCS results using low power or normal power: released 83 and 89; a 78 and 82; b 78 and 82
  *   I had to repeat firmware update to released version, as rthe first time it caused a warning of firmware not loaded. wrong operation. Second intent was correct.

My initial conclusions are:

  *   Released version has normally a higher RCCS number than versions a and b; this is theorically better. Could the problem be a possible saturation of signal to the receiver due to the higher RCCS?
  *   Versions a and b are theoretically more stable than released, according to the results.

If you want me to do any further tests, I will do them.

I hope it helps....

-------- Mensaje original --------

Fecha: 20/11/22 21:52 (GMT+01:00)
Para: pascallanger/DIY-Multiprotocol-TX-Module ***@***.***>
Cc: Luznatural ***@***.***>, Mention ***@***.***>

benlye, 2022-11-22:
If both the 'a' and 'b' firmware performed the same, then it seems like we can tentatively rule out the compiler change - if the compiler was the problem then I would have expected one of those firmware build to show the same behavior as the 'released' build.

I have a slightly more scientific test running on my bench since last night - I'm using an Arduino to capture the PPM output of my X8R receiver.  I'm, outputting the PPM values on all 8 channels to serial and capturing that data to a log file.  The radio is outputting servo sweeps with various timing on four channels, constant +/- 50% values on two channels, and 0% on two more channels.  My failsafe is set (via the TX) to set all channels to -100%.

I ran it for four hours this morning (EdgeTX 2.7.1 on the TX16, released FW v1.3.3.20 on the MULTI-Module, EU v2.1 FW on the RX) and did not have any detectable outages on any channel.  I'm running it again now and will let it run until the TX batteries go flat.

I'm not sure what else I can change to test.  I'm wondering if we should try to duplicate the module ID and RX number?

pascallanger, 2022-11-22:
@benlye what is the difference between a, b and release?
I would have thought that either a or b was the same as release. But it looks like only release is impacted so I'm a bit confused.
What about your X8R, is it running the exact same code as @Luznatural?
How do you identify a glitch in your test? Would it be any channel reaching -100%? I don't know how long the RX needs to go to failsafe, you most likely need a few secs without packets to reach this state. You might not want to do it but I think a good input to have would be to solder a wire to the X8R LED so you see in your log when the green LED goes off or red goes on (gnd being common only 1 wire is needed before the resistor).

Having the same ID and RX number would remove one more element but I would be surprised if it has an impact because I assume the tests on the 3 fws have been done using the same RX, same ID and same RX number, @Luznatural can you confirm?

benlye, 2022-11-22:
a = 1.3.3.20 code compiled in my environment with the old board/compiler
b = 1.3.3.20 code compiled in my environment with the new board/compiler
release = 1.3.3.20 code built in Git by the CI/CD process

My build environment is Ubuntu using the Arduino CLI, so should be the same as the Git build environment.  My X8R is running the latest v2 EU firmware, which is what  @Luznatural said they are running.

I would identify a glitch as any channel deviating measurably from the expected output.  I'm sampling the channels around 60Hz, so I think I should be able to see a 1-3sec outage. I've been parsing the output using a script as well as plotting it to analyze visually.  If I force a disconnect (e.g. by changing the RX Num on my model) the RX drops to the F/S values immediately.

The X8R is _supposed_ to output the RSSI as PWM on the RSSI pins, but I couldn't get a signal there (this seems to be 'normal').  I can open it up and see if I can solder a wire to the red LED.

benlye, 2022-11-22:
I now have an input coming from the green LED and I'm logging its state along with the PWM data.

I had to connect to the side of the LED which is connected to the STM3 (not the resistor side), and even then it was tricky - the pin is constantly cycling, so I had to de-bounce it and detect the state over a few cycles.  If it changes state I'm detecting and logging it within ~0.05s, so I think that's OK.

(If you watch it very closely you can actually see the green LED flicker every now and then with the naked eye. I verified that this also happens with older MULTI firmware, so I don't think it's a symptom.)

benlye, 2022-11-23:
My RX is running v2.1.0 LBT firmware, dated 2020-03-24 on the FrSky website.  The MD5 hash of the file is `F7350366F76F2AC6E0EF4AC4EDF8AA2D`.  It's the one that can still be downloaded from [their website](https://www.frsky-rc.com/x8r/).

@Luznatural can you confirm that this is the exact same firmware you are using on your X8R RX?

I'm happy to try changing the TX ID and RX.  At this point running the tests is very easy and I can leave it running in the background for any length of time.

Luznatural, 2022-11-23:
Just to reminded you, I am using FCC firmware...

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

benlye, 2022-11-23:
> Just to reminded you, I am using FCC firmware... 

Ugh, I don't know why I had it in my head that you were using EU/LBT.  Setting up to test with that now...

pascallanger, 2022-11-23:
Let's see... FCC or lbt are pretty much the same...
@Luznatural can you share your ID using the multi config lua script and your RX number?

benlye, 2022-11-23:
My test is running again. I flashed the RX using `X8R_ACCST_2.1.0_FCC.frk` with MD5 hash `174DC4E071917E93802C622A9C7ADB881`.

If you can let me know your TX ID and RX num I can test those on my radio.  You can get the TX ID using the MultiConfig script in the Tools menu.  It is the 'Global ID' value.

benlye, 2022-11-24:
I ran the test with FCC firmware on the X8R for 12hrs - no glitches.

pascallanger, 2022-11-24:
12 hours wow. I would have expected that result since like I said FCC and LBT are pretty much the same at least from a Multi perspective.

@Luznatural We can't reproduce the issue... Could it be your own setup?
-  X8R fw version, can you confirm that you are running the same as Ben
-  Could it be that you have a telemetry screen or a lua running in the background which could affect the radio behavior?
     - Try to create a blank model and just run your test
-  Your ID & RX num is affecting either multi or the RX (known issue on SxR, I haven't heard the issue on other models but who knows)
     - Give us your ID and RX num
-  Your plane setup:
     - onboard power issue?
     - one device like the ESC or servo creating interferences
     - ...
     - Try to do a setup on your bench using a trusted power source, receiver and a trusted servo

Luznatural, 2022-11-24:
Sorry for the delay. I was on workout trip out. My Tx ID is 65 57 FE 42
And the receiver Number is 17
Sorry again

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

Luznatural, 2022-11-24:
My X8R Firmware is FCC 2.1.0
The plane is a glow .46 Thunder Tiger engine. I do not have an ESC, and the servos are brand new futaba that run smoothly without extranjero noises.
I will do the tests again with brand new RX, servos and fresh battery, and report again.

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

Luznatural, 2022-11-24:
I have been doing a recap in the TX configuration, and I have the following:
Module firmware multimodule 1.3.3.20 AETR
External module OFF

VERSION
FW: 2.7.1 (a55aff02)
DATE:2022-05-30
TIME:23:15:22
CFGV: 221
OPTS: bluetooth, crossfire, ghost, afhds3, internalgps, internalmulti, internal access, multimodule, luac

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

benlye, 2022-11-25:
I ran the same test again today for several hours using the TX ID and RX number above - same result, no dropouts.  

I can't easily get 50m between my TX and RX but I put ~10m and a few solid walls in the way and I had a steady 80db RSSI.  Every now and then my Arduino detects the RX LED going out but it's only for 1-2 frames, or about 15-30ms.  Not long enough to visually turn the green LED off, and certainly not long enough to turn the red LED on and trigger failsafe.  

Every configuration I've tested behaves pretty much exactly the same and I can't find any evidence that anything is any different with v1.3.3.20.

Luznatural, 2022-11-25:
I have been repeating my humble tests yesterday, and this time I was unable to get a single failure.

Perhaps is a problemas of my área of flight rather than related to the software or firmware.

I think both of you have been very kind, patience with this issue, and invested a lot of time both three of us, to make sure that the system is safe.

On the other hand, I made a forensic inspection of my friend,s weekend crash, and he had several things to fix in his plane, like having a single lipo for both the gas engine and the receiver. So, the possibility of interference is exponential, especially with an X8R receiver. So I lectured him for raising the bells of alarm.

Then, we will have to let it go.

Thanks a lot for taking care of us and being so rigorous in your firmware deliveries.

If anything else arises I will let you know, and I hope you take me in account if you have to perform any future test, that I will gladly assume.

Thanks again

Best regards
Jaime

Obtener Outlook para Android<https://aka.ms/AAb9ysg>

closed 2023-10-30 by pascallanger