pascallanger/DIY-Multiprotocol-TX-Module #764: Channel value in HoTT?

TheIsotopes, 2022-11-21:
@pascallanger
Can you tell me why you set the channel travel in HoTT protocol to -125% <-> +125% and not to defaults -100% <-> +100%?

I noticed it when i wanted to use a Brain2 with my TX16S and a GR-12L.
I had to reduce the servo travel in the TX16S ( -77% <-> +77%) in order to receive correct values in the Brain2.

With an MX-16 and the GR-12L i didn't have to change the servo travel much to get the correct values in the Brain2.

pascallanger, 2022-11-21:
I've matched whatever was the output of the original radio or module I've been sent to work on this protocol. So my guess is that not all HoTT radios have the same ouput...

closed 2022-11-21 by pascallanger

TheIsotopes, 2022-11-21:
@pascallanger As far as I know, all hott transmitters have the same output value. 
Which transmitter and receiver did you use to compare the protocol.?

pascallanger, 2022-11-21:
It's so long ago now that I can't even remember.

TheIsotopes, 2022-11-21:
I consider this more of a bug and would appreciate it if the servo travel was set to default ( -100% <-> +100%).

pascallanger, 2022-11-21:
This protocol is out for a really long time. There are so many people using it that touching something like the servo travel could have a huge impact.
So this is not something I'll address at this point.

TheIsotopes, 2022-11-21:
it is a big effort to change the standard servo travel. I would like to test it.

pascallanger, 2022-11-21:
It's a one line change in the code: https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/70527512611df61058cf09044247707a0940ea3f/Multiprotocol/HOTT_cc2500.ino#L208

TheIsotopes, 2022-11-21:
ok i see ... in the two lines i found change the 860 to 1100 and it should work.