pascallanger/DIY-Multiprotocol-TX-Module #766: won't run on OTX Radiomaster TX16S

khmewes, 2022-11-23:
I am getting these two error messages. This is with my OTX RM TX16S. On my ETX RM TX16S it works fine. So it must be an OTX issue.
[image]
[image]

BatesKG, 2022-11-26:
Getting same errors here on latest OpenTx on my TX16S.  In the documentation it says the B+W version is for OpenTx, but that version also gives an fstat error.

frankiearzu, 2022-11-26:
You are right, fstat is not supported in OTX. let me see what other way can i do to validate that the extra files are there.

frankiearzu, 2022-11-26:
Just checked in a fix, but the pull request still needs to be merged to main.

after removing the "dirExist" lines, added an "asset" to give an error if the files does not exist. the loading of the extra files now looks like:

```
local dsmLib
if (SIMULATION_ON) then
  -- library with SIMILATION VERSION.  Works really well in Companion for GUI development
  dsmLib = assert(loadScript(DSMLIB_PATH.."DsmFwPrgSIMLib.lua"), "Not-Found: DSMLIB/DsmFwPrgSIMLib.lua")(DEBUG_ON)
else
  dsmLib = assert(loadScript(DSMLIB_PATH.."DsmFwPrgLib.lua"),"Not-Found: DSMLIB/DsmFwPrgLib.lua")(DEBUG_ON)
end
```

BatesKG, 2022-11-26:
Thanks for looking into this Frankie.  I tried the above change, and the script now runs.   Unfortunately, I'm getting a weird strobing effect on the text.  The selection highlight is always on two rows and is strobing between them.  When you first navigate onto a new page, all of the text is strobing until you cycle over it once.  Then it's back to just the highlighted row strobing between the last text you were on and the current focus.

frankiearzu, 2022-11-26:
Could be the combination of color selection/attributes that im using for OTX.  Tried to use the default theme there, since color management is a bit different between OTX and ETX.   The Black&White version should work for you.
Will research the color to try to force them to a known settings instead of using the Theme. On my OTX blue schema seems to work fine.
[image]

BatesKG, 2022-11-26:
I was using the blue schema. Here is the same screen as above, but on mine, the selection highlight is bouncing between 'FM Channel' and 'Act' of the AS3X line.
 
[image]

BatesKG, 2022-11-26:
The formatting/layout between our two screenshots is different.   Your title has a grey background and the field holding "FM Channel >" is much wider on yours.  Your screenshot is from a RM TX16S running OTX 2.3.15?   I also don't have the boxes around the setting values, just on the Back button.    Where would those differences be coming from?

frankiearzu, 2022-11-26:
@BatesKG 
Found the problem with the flickering/strobbing.. OTX needs the GUI to be refreshed on evey cycle of DSM_Run() to avoid that behavoir.  I check in a fix for that, but is not merged into main.  
A quick fix for you is to comment the lines who tries to optimize (adding the **--** at the beginning of the lines)
not refreshing the screen on evey DMS_Run call (towards the end of the file), so it calls GUI_Display on every cycle.

```
-- if (ctx.Refresh_Display and (getTime()-lastRefresh) > refreshInterval) then --300ms from last refresh 
    GUI_Display()
    ctx.Refresh_Display=false
    lastRefresh=getTime()
-- end
```

khmewes, 2022-11-26:
I have replaced the lines (actually comented them out) with the lines you posted. Strange action on the screen, too fast to decern and crash, back to tools menu. I am also running OTX 2.3.15 but was bound to 230S Smart receiver RXFC6250HX v5.6.255. Since I am not a Lua programmer I'll wait until a stable version is posted.
Thanks for looking into this and all the effort.

frankiearzu, 2022-11-26:
@khmewes,
Just test it with my 230s, seems to work. Also with a AR631.  OTX 2.3.14.    Will install OTX 2.3.15 to see if there is something different. What editor do you use??  Notepad++? looks like normal notepad tried to cut some long lines.

khmewes, 2022-11-26:
I am using Notepad++
Meanwhile here is the error message I am getting.
[image]
It might also think it's in ETX and is looking for the touch panel.

BatesKG, 2022-11-27:
khmewes, on OpenTX, you can only run the B+W version.   I get the same error if I run the Color-Touch version on OTX.

I use Notepad++ as well.    I tried the B+W file from your posted zip file Frankie.   Still strobing on the highlight/focus row and on various text.  When you first run the script the first page strobes, but then settles down after a couple seconds.  Going into the gyro menu, the behavior is unchanged.    Sorry!

frankiearzu, 2022-11-27:
@BatesKG @khmewes,

The latest problem was the way i was detecting if it was OpenTX or EdgeTX.. was using OTX 2.3.14, and it did not work on 2.3.15.   I did a fix so now it properly detects EdgeTX, OTX 2.3.14 and OTX 2.3.15.   In OTX 2.3.15 they change a LUA function to make it closer to Edge, and tha was how i was detecting ETX or OTX. 

Zip attached.    Thank you for your testing, i hope that this solves the problem.  

[DSM FwdPrg_05.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10097264/DSM.FwdPrg_05.zip)

BatesKG, 2022-11-27:
Tried this latest zip.   I've got some good news......and I've got some bad news.   On the plus side, NO more flickering!  That looks much better.   On the downside, everything is now read only.  IE: if you go into the AS3X settings page, or any of the other pages, your only option is the Back button.   You can no longer scroll onto the gain values, etc. to edit them.

BatesKG, 2022-11-27:
Oh and yes, the Color version now does run for me also.   Both versions no longer allow editing though.

khmewes, 2022-11-27:
Same for me. No editing possible.

khmewes, 2022-11-27:
The numbers disappear when you click on them.

BatesKG, 2022-11-27:
khmewes, how did you click on the numbers in OTX?   No matter which way I turn the wheel, I'm stuck on the Back button and can't click on any of the values.  OTX has no support for touch, right?

frankiearzu, 2022-11-27:
Most of the Gyro Gains are not editable in the AR631/637.  Try the "Gyro-Setting"->"F-Mode setup", more editable fields there.

In the color version, if the value has a box around it, is editable.  in the black/white version, if you can navigate to it, is editable. 
Using the wheel, scroll to move around fields, press wheel to edit, wheel to change values while editing. press to confirm to exit edit.
OTX has no touch capabilities, only EdgeTX.

Allowing to edit a field or not, is really part of information of the menu coming from the RX.

On my AR631,
The "Gyro Seetings"->"AS3X Setting" the only editable field is the "Gain Sensitivity: " values 1X,2X,4X
The "Gyro Seetings"->"AS3X Setting->AS3X Gains" nothing is editable, show only values for flight modes. Usually values 40,50,60. the flight mode will change with the F-Mode switch.
The "Gyro Seetings"->"AS3X Setting->Priority" nothing is editable usually 160 for all.
The "Gyro Seetings"->"AS3X Setting->Heading" nothing is editable
The "Gyro Seetings"->"AS3X Setting->Fixed/Adjustable Gains" you can edit if you want the gains fixed or adjustable.
The "Gyro Seetings"->"AS3X Setting->System Setup" most of the sub-menus has editable options (RX orientation, etc)
The "Gyro Seetings"->"F-Mode setup" usually is editable to configure the channel for f-mode and settings (auto-level, etc.
The "Other Settings"->"FailSafe->Failsafe" many editable options.
The "Other Settings"->"Frame Rate" many editable options.

Some BNF will not show all the menus to protect the plane configuration.

Data comming from the RX is written in english to the log  /LOGS/dsm_log.txt
there are some things in the data of the menu that say if it is editable or not, 
Each line has the type of value, for example "T=M" is a line to navigate to a menu, "T=V_NC" is a value No Change (read only),
"T=L_m0" list menu, "T=V_p" value percent, etc.

for example, in my Gyro Seetings->AS3X settings->AS3X Gains show in the log like this:

12.400 MENU_TITLE: SEND DSM_getMenu(MenuId=0x1012 LastSelectedLine=0)
12.450 MENU_TITLE: RESPONSE Menu: M[Id=0x1012 P=0x0 N=0x0 B=0x1011 Text="AS3X Gains"[0x1DE]]
12.500 MENU_LINES: SEND DSM_getFirstMenuLine(MenuId=0x1012)
12.600 MENU_LINES: RESPONSE MenuLine: L[#0 **T=V_NC** VId=0x1000 Text="Flight Mode"[0x8001] Val=nil [0->10,0] MId=0x1012 ]
12.650 MENU_LINES: SEND DSM_getNextLine(MenuId=0x1012,LastLine=0)
12.750 MENU_LINES: RESPONSE MenuLine: L[#2 **T=M** VId=0x1012 Text="Rate Gains"[0x1E0]   MId=0x1012 ]
12.800 MENU_LINES: SEND DSM_getNextLine(MenuId=0x1012,LastLine=2)
12.900 MENU_LINES: RESPONSE MenuLine: L[#3 **T=V_NC** VId=0x1004 Text="Roll"[0x40] Val=nil [0->100,40] MId=0x1012 ]
12.950 MENU_LINES: SEND DSM_getNextLine(MenuId=0x1012,LastLine=3)
13.050 MENU_LINES: RESPONSE MenuLine: L[#4 **T=V_NC** VId=0x1005 Text="Pitch"[0x41] Val=nil [0->100,50] MId=0x1012 ]
13.100 MENU_LINES: SEND DSM_getNextLine(MenuId=0x1012,LastLine=4)
13.200 MENU_LINES: RESPONSE MenuLine: L[#5 T=**V_NC** VId=0x1006 Text="Yaw"[0x42] Val=nil [0->100,60] MId=0x1012 ]
13.250 MENU_LINES: SEND DSM_getNextLine(MenuId=0x1012,LastLine=5)

With all with "T=V_NC", all fields are Read-Only/No-Change. Will play with my version to see if i can force edit those fields.

Frankie

frankiearzu, 2022-11-27:
Just research a bit more... they AS3X gyro gains are readonly.  What you do is that you assign a knob channel to the "Gain Channel", and by moving the knob you adjust the values with "Capture Gyro Gains" to update the values.

https://www.youtube.com/watch?v=bAO4EVQ3TTI

khmewes, 2022-11-27:
BatesKG, I could scroll down and was able to click on them.

khmewes, 2022-11-27:
Frankie, I was scrolling through the fields fine, but when I tried to click on the field/value the value went blank, i.e. disappeared, only background color, no blinking cursor or anything else. Just went blank, The box and the text remained.

khmewes, 2022-11-27:
Will the BW version run on the TX16S color display? If so I will try it. Could be a color combination/theme issue which I am using. As far as i know I am using the default blue theme.

khmewes, 2022-11-27:
Frankie, if there is a problem with the color theme I can zip the theme and the current code I am using and post it. The ETX TX16S I am using also works fine, touch screen and all. I am using amymodified color scheme on it and can zip it and post it also.

BatesKG, 2022-11-27:
The AS3X Gains, and Priority as well as others should be editable.   They are on a Spektrum radio.   I can scroll into menus down to the level where the AS3X Gains are, and then the scroll wheel won't take me off of the Back button.    
On version 0.2, a lot of the text values are Unknown, but I could still navigate down to the AS3X Gain value and edit them.
I just tried the 0.5 version that is now online as of this morning and I still can't edit the AS3X Gains.   
Just a warning that with the currently posted version, if you go into the Relearn Servo option and click apply, it does something that locks all the servos hard over and you can't move them again until you do a Relearn with an actual Spectrum Tx.

BatesKG, 2022-11-27:
As soon as the ETX gets off of the RC candidates, I'm going to move all my TXs to ETX.   So if this is working fine there, I'll just wait for that.
Thanks for all the great work done on this!

frankiearzu, 2022-11-27:
@khmewes:
Yes, both color and B&W will work on TX16S color display.. both shares the same Forward Programing Library, what changes is the GUI Display portion. The idea of the B&W is to make it work to smaller screen radios later (Simple B&W+Text only).

The color script in OTX uses the default theme.  
During Edit: The Edit color is the same as the color as the header of the menus, and will "Gray out" all other parts of the screen.  RTN to escape the change (return to how it was). 

@BatesKG : Thanks for the info about the AS3X gains.. tried the V0.2 and same thing, not editable  (thats where i started the code from).   No longer have a Spektrum radio in hand with FwPr capabilities, but going to the field right now, and will test it with a friend one. If they are Edittable on spektrum, i will change that behaviour..  that type of line (T=V_NC: Value no change) is probably a bad understanding when it was reverse engineer it.    The think is that "Flight mode" comes the same way, and not editable.. Probably need to create an exception for them.  All this reverse engineering is try and error, since Spektrum don't document the internals of it.

Will add a warning about the " Relearn Servo" in the text of the menu.. to discourage people from using it until we can figure out how to make it work. That was one of the pages that i needed to hack.

Really apreciate your help! 
As for EdgeTX, have been using 2.6.0 and 2.7.1 for a while, and the 2.8.0 RC4 for about 1m (since came out). Works great.  Yesterday they release the official 2.8.0, and upgraded last night.

BatesKG, 2022-11-27:
It probably depends on what Rx you are connected to.   I'm testing on an AR637T.    If I was using an AR637TA that came in an Eflite plane and wasn't unlocked, then I believe the AS3X gains may be locked to factory settings in that case.   
Thanks for the head's up on the ETX 2.8.0 release.  I had checked yesterday but not this morning.

frankiearzu, 2022-11-28:
@BatesKG..  you are right.. they are editable..  was looking at my logs, and i think besides the line type, there is other way to know if it is readonly or not.   The flight mode line seems like is an exception to this, but can recognize it easilly.

Let me do some changes and will share a ZIP with you guys before i check in any code to github.
Thank you again,
Frankie

BatesKG, 2022-11-28:
For testing, the Spektrum radio I'm using is the ix20 and the Rx is an AR637T.
For #1 above, the "Level model and capture attitude" is a button.   When you click it, the roll and pitch values will change to the current offsets based on how level the Rx is.   The Roll/Pitch values aren't live as you move the Rx though.  You can also click on the individual Roll/Pitch values to manually edit them.  Note for this section to work properly, the Orientation of the Rx in the model needs to be set correctly first.

#2:   If you set the Source Flight Mode and Target Flight Mode both to FM1 and click Apply, it first goes to a confirmation page, and then if you click Yes, it goes to the Complete page.   If you click Complete, it goes back to the Utilities page.                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                      
[image]
[image]
[image]

Regarding another comment you made about the AS3X Gains adjust page and the Flight Mode..... The Gains are editable. The Flight Mode displayed changes (along with the gain values as you switch flight modes, but the Flight Mode field itself is not manually editable.

Thanks again Frankie for doing all this work!
Kevin

frankiearzu, 2022-11-28:
The new zip attached.. all the fields that previously was read only, now are editable

frankiearzu, 2022-11-28:
During development, i have the simulation turned on, so i can test first in Companion, and then the same file goes to the Real TX.
With companion, I can at least test quickly 2 receivers from my computer.
When i check the code into GitHub, i make sure the simulation is OFF.  
Will fix a few messages, and contribute the latest changes  to GitHub.

frankiearzu, 2022-11-28:
It really depends what you are testing... if it is a GUI only thing, Companio helps a lot to do it quickly.  if is something related to the low level protocol (that Companion cannot do),  testing on the REAL RX is the only way.

khmewes, 2022-11-28:
Agree.

frankiearzu, 2022-11-28:
Latest changes.
Gyro Settings -> System Setup -> Safe/Panic Mode Setup -> Attitude Trim works.
Formatting of nubers, Right aligh in the box (Color only) and %/o when known.

[TOOLS.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10106661/TOOLS.zip)

frankiearzu, 2022-11-29:
@BatesKG,
One last screen that i don't know if the text is correct:
The "Gyro Seetings"->"AS3X Setting->System Setup"->"Orientation".

After setting the orientation manually, and press "Continue", It goes to another page to confirm Resetting the RX and somethig else. the first text  i guess it since it will reset the RX.. but the 2nd, no idea.   Can you check in your Spektrum please.
After that, i think is in good shape.. a friend is testing with AR630 (that i think is the same as the AR631), and an AR10360T

[image]

frankiearzu, 2022-11-29:
@khmewes 
Found the discrepancy with v0.2 and 0.5 with On/Off on the Blade 230s.  My values came from an older v0.2 and was corrected later.  Flew today an stability ON is correct.  The latest file is Tools-Nov-29.zip ..   you need to replace all the files from the zip, including DSMLIB. Tested in OTX 2.3.15 and ETX 2.7.1/2.8.0

BatesKG, 2022-11-29:
Hi Frankie.
  The Unknown on your screenshot is just a text label.    Here is the screen sequence......
[image]
[image]
[image]
[image]

khmewes, 2022-11-29:
@frankiearzu I am somewhat confused now which modes you are calling Normal and stunt and how they correspond to the ones Spektrum is calling FM1, FM2 and FM3.
There is also a FM0 and FM4 showing when you go throught the range of -100% to +100% on Channel 5.
I used to assume that +75% which is FM3 was the stability mode, according to what people have been using and published. You have FM1, stability ON as your stability mode and FM2 as stunt 1 and FM3 as stunt 2.
Which is correct according to Spektrum and what are the output levels to channel 5 (gyro) on the actual Spektrum radio (I don't have one, can't check)
Thanks.

khmewes, 2022-11-29:
@frankiearzu Followup. Could you check with one of your Spektrum buddies what their convention is, i.e. what the Spektrum transmitter puts out? I guess it depends on their choice of switch position but switch away from user which is considered 'down' or ^| (looks like up always puzzled me why it's called down since it looks 'up' but I guess it's in relation to the radio 'top' face). So down puts out +100%. Also 100% in standard  protocol corresponds to 75% in Spektrum, but the multi protocol is supposed to take care of this, is it?
Cheers and Salut and Ciao

frankiearzu, 2022-11-29:
@khmewes  for the Heli version receivers FC6250HX,  shows Normal, Strunt-1, Stunt-2 (not Flightmode 1, 2, 3,4)
Here is a video:

https://www.youtube.com/watch?v=aCmbCvQ5vUo

khmewes, 2022-11-29:
@frankiearzu Hi, I installed your latest release on my OTX 2.3.15 TX16S and it is not crashing anymore. Great. Seems to be working and looks good. 
To add to my previous posts and private messages, I found two additional modes that show when I use values less than -50% (FM 0) and greater than 85% (FM4). I couldn't find any reference to them in Spektrum manuals so far. what they do is tbd. Could be related to recovery mode.

frankiearzu, 2022-11-29:
@BatesKG Thanks for the screen.. updated the text to match.

@khmewes.. Will look into the extra flight modes on the Blade.. i know that i have use FM1/Normal. FM2/Stunt, FM3-Stunt2. and FM4/Hold... Will look the FM0.. have never used it.. but will configure my TX to expose it.

Question for both:  many people that only use TX16 (not Spektrum users) has asked me why not rename the channels Gear, Aux1, etc to  CH5. CH6, etc that makes more sence for OTX/ETX.  I was thinking  to show both:  Gear (Ch5), Aux1 (Ch6),  or  reversed Ch5 (Gear), Ch6 (Aux1).

what do you think??  

Updated Zip from today's changes:
[TOOLS-Nov-29-r2.zip](https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/files/10117829/TOOLS-Nov-29-r2.zip)

khmewes, 2022-11-29:
@frankiearzu I watched the video you posted the link for. This is quite different from the original settings on my 230S smart. My initial settings were I suppose adjusted to the DXS radio that came with my heli. It seemed to use channel 5, GEAR, for the flight mode and gain was inhibited. At least thats what v.0.5 told me under Setup. The modes were showing as Normal, Stunt1 and Stunt2. So gain would not be communicated to the receiver then because the Gain channel was inhibited?
So I followed the example in the video and set the FMode channel to AUX2 and the gain channel to GEAR (channel5). I put the values of -50%, 0 and 50%(I assume it is +50% not -50% as he said/misspoke). I also adjusted the stability to ON, ON (wanted to try if it worked also in stunt) and OFF for Stunt2. I have no entries in the Mixer settings anymore, they are under channel7, AUX2 now. I do not have an entry of -100% under AUX2 linked to the SF switch. That is done with the special function Override CH3 -100. What confused me was the 3rd choice under settings named "Output Channel 6" which was not shown in the video. Channel 6 is usually the Pitch channel and it showed Inhibit? in the select field. Shouldn't it be set to AUX1 which is channel 6? I did that for now to try it out, I hope not with bad consequences.
So what do you think of all this? Does this sound reasonable to you?
I am guessing the 'Gain Channel' communicates the individual gain setting selectable for the flight modes? As part of the flight envelope?
Very confusing. It seems to me Spektrum opened up a can of worms with all the programmability. Trying to appeal to the technobuddies and making their products look technologically sofisticated.
This is turning into a blog so I better stop here.
Let me know what you think.
Thanks

khmewes, 2022-11-30:
@frankiearzu I think in the context of FwdPrg which involves Spektrum technology keeping both nomenclatures Channel5(Gear) or Gear(CH5) is reasonble. You can give a nodd to the OTX/ETX people by using CH5 (Gear) etc. instead of Gear (CH5) etc. 
Just imho.

khmewes, 2022-11-30:
@frankiearzu Hi Frankie, it seems the original issue on this topic has been solved and v.0.5 runs fine on my TX16S with OTX. Any remaining discussion seems somehow  linked though. So do you want to keep this issue open or should another one be opened more specifically to configuration issues or remaining design issues? I still have some unanswered questions about the workings of programming the FC6250HX or more specifically the SPMAR6250MHXA version that is in the 230S Smart. There seems to be some differences between the two. But I have managed to make progress using the video about FwdPrg the FC6250HX and applying it to the SPMAR6250MHXA.
One of the questions remaining is the extra line that shows for the SPMAR6250MHXA in v.0.5, 'Output Channel 6'. That was not shown in the video. Is this unique to the SPMAR6250MHXA or did you put that in? I selected 'AUX1' as its  assignment since for Spektrum receivers that's what channel 6 is called and it's used for the Pitch servo output. I also assigned CH7, i.e. AUX2 as the Flight Mode communication channel per the video and gain channel is CH5, i.e. GYRO. I was surprised to see that it was called that and not GEAR. I also was surprised that the settings for the GYRO values were 75% for all flight modes. Is this correct?
I tested the 230S Smart today with these settings and it flew fine Normal and Stunt1, although I had stability ON for stunt1 as a precaution. There was a slight tilt  to the right in stunt1 before takeoff, but it flew fine, stable of course. Didn't fly in stunt2 with stability OFF although it seemed doing fine during spool-up, no tilt.
So what is your take on this?

frankiearzu, 2022-11-30:
@khmewes:   Everything that the tool shows in the screen comes from the RX. How Forward programming works is that we request menu information from the RX, the RX send it to us, and we are render the menu based on that info.  The menu in the RX don't have the text, but they give us a message# to display.. the tranlation to english of that message# is in the script.
(you can see the /LOGS/dsm_log.txt has that conversation with the RX).

That "Output Channel 6" could be correct message (since i think ch6 is the pitch management), or can be that we have the wrong translation for that message#. Will check.

As for the conversation about setting the FC6250MH receiver, thats is really a topic outside the tool.
I think the only thing that i will do next is change the messages for "Gear" to "Ch5 (Gear)", "Ch6 (Aux1), etc. so that is closer to how OTX/ETX uses.  

We can open specific tickets for specefic problems, and call this thread done, so that we can close it.

khmewes, 2022-11-30:
@frankiearzu Thanks Frankie for all your work. Hope we can keep working together in the future.
I will close the issue then.

closed 2022-11-30 by khmewes