pascallanger/DIY-Multiprotocol-TX-Module #777: failsafe on FLYSKY AFHDS2 "NONE" Option doesn't work

arhifisyant, 2022-12-06:
Im on :
radiomaster zorro internal JP4in1. 
EDGE TX 2.8.0 (final release)
multiprotocol 1.3.3.20
receiver ia10B
---------------
problem : 
I manage to set failsfe to "NONE". It doesn't work as it should.  
Receiver should output no signal pulse in all the channel. Instead, receiver behave same with "HOLD" option selected. I can prove it by moving the servo arm during failsafe. The servo still try to counter my force. 

I also tried custom failsafe, with each output channel configured one by one. it is working good, except the "NONE", option. Same with my explanation above, "NONE" option behave the same with "HOLD". 

Please fix this, I really need to avoid burning my servo when failsafe happens. At least when you guys release fix for this, I am the guy that really helped. Thank you in advance.

mha1, 2022-12-11:
In general NONE  can only work if the receiver supports it. Can you verify that NONE is working with a genuine Flysky transmitter?

arhifisyant, 2022-12-13:
lets agree that this is failsafe behavior should be :
NONE : receiver output no signal 
HOLD : receiver output last received signal 
ON with some value : receiver output signal according to the setting

I can confirm that in the original flysky transmitter (FLYSKY i6), with 6 channels, ia6b receiver. the failsafe behave like this. 

When we bind the receiver for the first time, and leave the failsafe settings to its default, untouch it, and turn off the transmitter, the receiver enter failsafe state and output no signal in all the 6 channels. I can confirm that by force the servo to move, all of them is free to move, including the ESC outputing tone continuously (meaning it receive no signal from the receiver). When we check failsafe menu, it is appear "OFF" in all 6 channels . 

[image]

When I set some of the failsafe channel to "ON", for example, channel 1 to "ON" with 25% value, channel 2 to 6 remain "OFF", the result is channel 1 in the receiver outputing signal as expected (25%) during failsafe state, channel 2 to 6 also outputing last signal "HOLD" (btw this is weird in the first place, against our common sense, instead, ch 2 to 6 it should output NONE). Of course when I set all the 6 failsafe channels to "ON" and every channel to put some value, it works as expected. 

This is the important part. When I turn all the 6 failsafe channel to OFF, the receiver, behave "HOLD" (this is also weird). Even if I pick option "restore failsafe settings to their factory default", the receiver still behave "HOLD". I withdraw my request above since the original transmitter failsafe behave like this. 

[image]

In conclusion, the failsafe behavior in flysky afhds2 is weird in the first place, but I believe there is some way to make at least flysky receiver behave "NONE" in all the channels, because in the original flysky transmitter when we bind the transmitter and receiver, untouch the failsafe setting, it behave NONE. 

I am fully responsible for this test report I did the test on my own. Hope it helps you figure out the solution and the community get the benefit

mha1, 2022-12-13:
Thanks for your feedback. For now it seems neither EdgeTX nor MPM can solve this as the receiver firmware needs to capable of turning off the outputs in a f/s situation. If the original Flsky transmitter can't command the receiver to do it there is no way MPM (and EdgeTX) could do it. It most likely means Flysky just didn't implement NONE.

arhifisyant, 2022-12-13:
Understandable . Maybe, just maybe, failsafe behavior in the current MPM for AFHDS2, already behave the same with original flysky tx. I mean, when we re-bind AFHDS2 receiver to MPM, and leave the failsafe untouched, failsafe state will output "NONE". This is my only asumption btw, I didnt re-check .

closed 2023-10-30 by pascallanger